zephyrproject-rtos/zephyr #91919: SPI driver issue

Pateldinkar, 2025-06-19:
### Describe the bug

Hello all,
I am using a peripheral which have an internal FIFO. From that peripheral, I receive the interrupt when fifo is full. I received the interrupt every 160ms and read 360 bytes of data. It uses a fixed register read/write cycle every interrupt for fifo read. This process is successful over the longtime. After some random point of time SPI driver sent the wrong value sequence and due to that, my peripheral internal configuration changed, and it stopped working as per expectations.

- A successful send sequence of 4 bytes is 0xFF,0xFF,0xFF,0xFF.[image]

- Found wrong sequence of 4 bytes is 0x00,0xFF,0xFF,0xFF. After this sequence peripheral goes into failure.[image]

In both failure and success cases, it is sent from the same API with a fixed register value. I also verify that the spi_transceive_dt is successful every time.

Why does this register sequence change at some random time?

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

pdgendt, 2025-06-19:
It might be useful to clarify which platform/driver/board you're using.

Pateldinkar, 2025-06-19:
I am using the nrf52840 MCU and have made a custom board.

Pateldinkar, 2025-06-19:
@JarmouniA No Specific step to reproduce. I will reproduce at a very random interval, but it is less than 8 hours.

danieldegrasse, 2025-06-24:
> I am using Zephyr "v3.5.99-ncs1-2" version.

@Pateldinkar, please open issues with NCS in the NCS repository- if you can reproduce this against an upstream version, then the issue can be reopened