zephyrproject-rtos/zephyr #91961: STM32F1: adc_channel_setup() fails with "Invalid channel reference" on valid DTS configuration

eiten, 2025-06-20:
### Describe the bug

The adc_channel_setup() function for the st,stm32f1-adc driver fails at runtime with the error message adc_stm32: Invalid channel reference. This happens even when the Device Tree configuration for the ADC channel is syntactically correct and complies with the driver's bindings (e.g., zephyr,reference = "ADC_REF_VDD_1";).

### Regression

- [ ] This is a regression.

### Steps to reproduce

Use a board with an STM32F103xB MCU.
Configure an ADC channel as a sub-node of &adc1 in the board's DTS file.
Call adc_channel_setup() for this channel in the application code.
The setup fails with error code -22 (EINVAL).

### Relevant log output

```shell
*** Booting Zephyr OS build v4.1.0-5574-gb1ad7877f238 ***
[00:00:00.001,000] <inf> main2: Starting final ADC test...
[00:00:00.001,000] <err> adc_stm32: Invalid channel reference
[00:00:00.001,000] <err> main2: Failed to setup motor ADC channel 9 (-22)
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

OS: Debian Bookworm
Toolchain: Zephyr SDK 0.17.1
Zephyr version: v4.1.0-5574-gb1ad7877f238

### Additional Context

- The issue was debugged extensively. We confirmed that the ADC clock prescaler and acquisition time settings are correct.

- The problem can be worked around by completely skipping the adc_channel_setup() call and manually setting the required hardware registers (ADC_CR2->TSVREFE for internal channels and ADC_SMPR1/2 for acquisition time). This proves that the underlying adc_read() function and hardware are working correctly and the bug is isolated to the setup/validation logic within adc_channel_setup().

Workaround is attached.

[workaround.zip](https://github.com/user-attachments/files/20840215/workaround.zip)

eiten, 2025-06-20:
I attached the DT file in the ZIP along with the main file.
How do I attach it as a git branch? Fork the project, clone it, init a new west workspace and add my files in which directory?

gautierg-st, 2025-06-30:
Hello @eitan,

The `zephyr,reference` property is unused in the STM32 ADC driver, but the driver returns an error if it is set to something else than `ADC_REF_INTERNAL`. If you use that, and correctly set the `vref_mv` to the real value of your reference, it should work.

closed 2025-09-22 by github-actions[bot]