zephyrproject-rtos/zephyr #91968: Drivers:video:GC2145 - Wish there was more control over image resolution and windowing.

KurtE, 2025-06-21:
### Is your feature request related to a problem? Please describe.

Suppose, I am using a display that is not one of the three defined resolutions: 320x240, 640x480 or 1600x1200
Such as the Arduino GIGA display 480x800 or ILI9488/ST7796: 320x480 and I would like the image to fill the full screen, what should I do?

For example with the GIGA display shield, my current only choice is to read it in as 1600x1200 and crop the image myself. 
However this approach requires a lot of resources, including memory and DCMI band width.  That is you throw away about 80% of the image returned.

Note: This is not totally specific to the GC2145

### Describe the solution you'd like

I wish that the zephyr code had some of the functionality that is contained in the Arduino library 
https://github.com/ArduCAM/Arducam_dvp
Or better yet from my updated version of this library, that @mjs513 an myself extended in particular for the Teensy boards.
https://github.com/mjs513/Teensy_Camera

In particular: extend the table of resolutions:  We currently have:
```
const int resolution[][2] = {
    {640, 480},   /* VGA       */
    {160, 120},   /* QQVGA     */
    {320, 240},   /* QVGA      */
    {480, 320},   /* ILI9488   */
    {320, 320},   /* 320x320   */
    {320, 240},   /* QVGA      */
    {176, 144},   /* QCIF      */
    {352, 288},   /* CIF       */
    {800, 600},   /* SVGA      */
    {1600, 1200}, /* UXGA      */
    {0, 0},
};
```
Also implement some form of zoom/clipping: The Arducam library has: GC2145::setResolutionWithZoom(int32_t resolution, int32_t zoom_resolution, uint32_t zoom_x, uint32_t zoom_y)

https://github.com/ArduCAM/Arducam_dvp/blob/master/src/GC2145/gc2145.cpp#L795-L891

Our version of it we have:   GC2145::setWindow(uint16_t reg, uint16_t x, uint16_t y, uint16_t w,
                      uint16_t h) {
https://github.com/mjs513/Teensy_Camera/blob/main/src/Teensy_GC2145/GC2145.cpp#L1103-L1125

It appears like the current file zephyr\drivers\video\gc2145.c has most of the capabilities built in to it like the function:
gc2145_set_window, however I don't see any way to easily access it.

Potentially, maybe one or more new methods should be added the the Camera class?
That would then probably require changes to the video object, for each of the cameras.

ngphibang, 2025-06-23:
> Now again suppose it takes a second to display the image. During that time the camera may have captured several images.
> Currently limited by it failing to grab a frame buffer. But when it comes time to draw the next frame, It grabs the oldest image
> in the queue, so typically the image is about as old as the time it takes to draw the image.
> Wish is: maybe there is only a need for 3 buffers. The one I am drawing, the last complete one the camera Read, and the current one the camera is reading into... Is there a prescribed way to do something like this?

I think you can use a LIFO queue and put the camera and the display in separate threads.

Another convenient way is to use a gstreamer-like framework and insert a queue element between the camera and the display elements.

KurtE, 2025-06-23:
Thanks @ngphibang ,  I probably would need to see what @iabdalkader did with the shared cache code to see if
that is currently doable and/or how hard it would be to implement.

@mjs513 also suggested maybe if we simply started the stream, grabbed one frame, and stopped it.
Currently in the arduino Camera wrapper library, I don't see anything that allows us to call the stop, nor not sure if it would work 
or not...

There appears like there might be some in the video.h:
```
static inline int video_stream_start(const struct device *dev, enum video_buf_type type)
static inline int video_stream_stop(const struct device *dev, enum video_buf_type type)
```
Maybe time for some hacking

josuah, 2025-06-26:
> EDIT: Or to relegate some of the windowing stuff down to the cameras that support it.

I should have precised what I had in mind: this is what I referred to: the DCMI is not able to do any video editing (crop/compose) on its own AFAIU.

But the `video_set_selection()` API can be used with image sensors directly to be able to crop them to any size.
For instance, interfaced with `gc2145_set_window()`.

Now I re-read the title again, it is not about having an API enabling to do this, but about having it implemented in the GC2145 driver itself...

This means the 1st part is now done: enable an API to do this.
The 2nd part remains to be done, absolutely.
If you wish to implement it, do feel free to ask for help and further explanations.
Otherwise the alternative is to wait that this gets implemented by maintainers and other developers. I know I will try to implement it on all supported sensors at some point but this is medium future.

avolmat-st, 2025-06-26:
DCMI and DCMIPP cannot be compared. Those are 2 differents IPs with very different capabilities, especially regarding to crop / compose.
The DCMI has some crop / compose capabilities, but very limited, basically will be able to do crop and /2 horizontaly / vertically only by skipping lines and/or pixels.

I agree with @josuah, a first easy step would be to map the selection API to the GC2145 driver, hence being able to control the windows of frame produced by the GC2145.
I was hoping to also add some limited DCMI selection features but those were not in my priority list hence I haven't done it yet.

KurtE, 2025-06-26:
Thanks @josuah @avolmat-st ( @mjs513 )

First quick question: I noticed that the video capture sample was updated as part of the PR.  Wondering are there any existing
boards that support this?   

I did not notice any ST boards listed for it under:
https://docs.zephyrproject.org/latest/samples/drivers/video/capture/README.html

Except the Arduino Nicla Vision.   

Now suppose I was going to experiment with this and with current stuff if I would like an image of 480x800 to fill
the GIGA display, the closest resolution is 1600x1200.  

Probably needs to add something like 
```
	.set_selection = gc2145_set_selection,
	.get_selection = gc2145_get_selection,

```
To the: static DEVICE_API(video, gc2145_driver_api) ,,,

But If I am running on a processor with the Pixel Pipe Line that can

do the windows down and likewise it has a camera that can do so?  Which one should do it?
My gut says the camera, as this still reduces the time it would take for the camera to return the data...
But that is just my gut...

Suggestions?

Thanks again

avolmat-st, 2025-07-05:
@KurtE , the selection API is pretty much done with the V4L2 - SET_SELECTION API in mind:
https://www.linuxtv.org/downloads/v4l-dvb-apis-new/userspace-api/v4l/selection-api.html

So, roughtly, what I had in mind considering the DCMIPP (which is the only one that currently implement this set_selection) is:
    1. there is a stream coming to the interface
    2. it will apply a CROP (aka kind of open a window) on this incoming stream, based on the rectangle defined
    3. at this stage, if nothing else would be done, what will come out of the dcmipp (in term of resolution) would be size of what is cropped.
    4. if a compose is as well set, it will apply a transformation on the resulting frame AFTER the crop. In case of DCMIPP, this can be only downscale.

The stm32mp135_dk board is using the crop only and currently the stm32n6 is using the compose only (just because I haven't put the crop in the configuration of the stm32n6 actually).

The specificity of those two platforms is that they both use the DCMIPP which is currently himself setting the sensor resolution to values given via the Kconfig.

In your case (the case of trying to crop a resolution which is higher than the default one), currently there is nothing specific done for this case so indeed as is the sample won't work in this case. You'd have for example to first set a first set_fmt on the sensor to set the "maximum" or "base" resolution you want, before setting a crop for example.

KurtE, 2025-07-05:
Side note to self and others:

For my testing, I was adding some more entries into the fmts array:
In particular, ones for the first several dividers of the 1600x1200:
```
	GC2145_VIDEO_FORMAT_CAP(800, 600, VIDEO_PIX_FMT_RGB565),	/* div 2 */
	GC2145_VIDEO_FORMAT_CAP(533, 400, VIDEO_PIX_FMT_RGB565),	/* div 3 */
	GC2145_VIDEO_FORMAT_CAP(400, 300, VIDEO_PIX_FMT_RGB565),	/* div 4 */
	GC2145_VIDEO_FORMAT_CAP(320, 240, VIDEO_PIX_FMT_RGB565),	/* div 5 */
```
But this makes me wonder about the code in gc2145_set_resolution:
```
	switch (w) {
	case RESOLUTION_QVGA_W:
		c_ratio = 3;
		r_ratio = 3;
		break;

```
Where 1600/320 = 5 and 1200/240=5, So why is choosing 3

closed 2026-09-16 by github-actions[bot]