zephyrproject-rtos/zephyr #91969: MAX32690: filesystem won't mount initially unless a delay is added

dkouba-atym, 2025-06-21:
### Describe the bug

I have a MAX32690 dev kit from ADI and have it configured with a LittleFS filesystem located in the chip's flash. Here's the relevant excerpts from my overlay:

```
	fstab {
        compatible = "zephyr,fstab";

        lfs1: lfs1 {
            compatible = "zephyr,fstab,littlefs";
            partition = <&user_data_partition>;
            mount-point = "/lfs";
            
            read-size = <256>;      // Match 32690 flash controller
            prog-size = <256>;      // Match 32690 flash controller
            cache-size = <256>;     // Not sure what the max is
            lookahead-size = <32>;  // Default
            block-cycles = <512>;   // Default

            automount;
        };
    };
```
```
&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

...

        user_data_partition: partition@154000 {
			label = "user_data";
			reg = <0x154000 DT_SIZE_K(1024)>;
		};
        
	};
};
```

When the filesystem is fresh after a full chip erase, LittleFS cannot initialize:

```
D: fs register 1: 0
I: littlefs partition at /lfs
I: LittleFS version 2.10, disk version 2.1
D: FS area 0 at 0x154000 for 1048576 bytes
I: FS at flash_controller@40029000:0x154000 is 64 0x4000-byte blocks with 512 cycle
I: partition sizes: rd 256 ; pr 256 ; ca 256 ; la 32
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1389: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
E: format failed (LFS -5)
E: fs mount error (-5)
E: Automount /lfs failed: -5
I: /lfs automounted
```

I had a suspicion this was because the flash device is not ready yet when we start using it, so I added a delay in the LittleFS init:

```
static int littlefs_init(void)
{
	static struct fs_mount_t *partitions[] = {
		DT_INST_FOREACH_STATUS_OKAY(REFERENCE_MOUNT)
	};

	int rc = fs_register(FS_LITTLEFS, &littlefs_fs);

	if (rc == 0) {
		struct fs_mount_t **mpi = partitions;
		k_sleep(K_MSEC(50));
		while (mpi < (partitions + ARRAY_SIZE(partitions))) {
			mount_init(*mpi++);
		}
	}

	return rc;
}
```

And the filesystem boots fine:

```
D: fs register 1: 0
I: littlefs partition at /lfs
I: LittleFS version 2.10, disk version 2.1
D: FS area 0 at 0x154000 for 1048576 bytes
I: FS at flash_controller@40029000:0x154000 is 64 0x4000-byte blocks with 512 cycle
I: partition sizes: rd 256 ; pr 256 ; ca 256 ; la 32
D: fs mounted at /lfs
I: Automount /lfs succeeded
*** Booting Zephyr OS build v4.1.0-40-g34b526826adf ***
I: /lfs automounted
```

`CONFIG_FILE_SYSTEM_INIT_PRIORITY` and `CONFIG_FLASH_INIT_PRIORITY` are at their default values, 99 and 50 respectively.

Anyone have an idea of what could be going on?

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

ttmut, 2025-06-25:
I tested this on MAX32690FTHR. Will try to get a MAX32690EVKIT as well.

`$ west build -p -b max32690fthr/max32690/m4 .\samples\subsys\fs\littlefs`

This is my overlay for `max32690fthr_max32690_m4.overlay`.
```
/ {
	fstab {
		compatible = "zephyr,fstab";

		lfs1: lfs1 {
			compatible = "zephyr,fstab,littlefs";
			partition = <&user_data_partition>;
			mount-point = "/lfs";
			
			read-size = <256>;      // Match 32690 flash controller
			prog-size = <256>;      // Match 32690 flash controller
			cache-size = <256>;     // Not sure what the max is
			lookahead-size = <32>;  // Default
			block-cycles = <512>;   // Default

			automount;
		};
	};
};

&flash0 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		user_data_partition: partition@154000 {
				label = "user_data";
				reg = <0x154000 DT_SIZE_K(1024)>;
		};
		
	};
};
```

Output:
```
*** Booting Zephyr OS build v4.1.0-6569-gf4a0beb2b7b1 ***
Sample program to r/w files on littlefs
Area 0 at 0x154000 on flash_controller@40029000 for 1048576 bytes
/lfs automounted
/lfs: bsize = 256 ; frsize = 16384 ; blocks = 64 ; bfree = 61

Listing dir /lfs ...
[FILE] boot_count (size = 1)
[FILE] pattern.bin (size = 547)
/lfs/boot_count read count:11 (bytes: 1)
/lfs/boot_count write new boot count 12: [wr:1]
------ FILE: /lfs/pattern.bin ------
09 aa aa aa aa aa aa aa 0a aa aa aa aa aa aa aa
0b aa aa aa aa aa aa aa 0c aa aa aa aa aa aa aa
0d aa aa aa aa aa aa aa 0e aa aa aa aa aa aa aa
0f aa aa aa aa aa aa aa 10 aa aa aa aa aa aa aa
11 aa aa aa aa aa aa aa 12 aa aa aa aa aa aa aa
13 aa aa aa aa aa aa I: littlefs partition at /lfs
I: LittleFS version 2.11, disk version 2.1
I: FS at flash_controller@40029000:0x154000 is 64 0x4000-byte blocks with 512 cycle
I: partition sizeI: littlefs partition at /lfs
I: LittleFS version 2.11, disk version 2.1
I: FS at flash_controller@40029000:0x154000 is 64 0x4000-byte blocks with 512 cycle
I: partition sizes: rd 256 ; pr 256 ; ca 256 ; la 32
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1386: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
I: /lfs mounted
I: Automount /lfs succeeded
*** Booting Zephyr OS build v4.1.0-6569-gf4a0beb2b7b1 ***
Sample program to r/w files on littlefs
Area 0 at 0x154000 on flash_controller@40029000 for 1048576 bytes
/lfsI: littlefs partition at /lfs
I: LittleFS version 2.11, disk I: littlefs partition at /lfs
I: LittleFS version 2.11, disk version 2.1
I: FS at flash_controller@40029000:0x154000 is 64 0x4000-byte blocks with 512 cycle
I: partition sizes: rd 256 ; pr 256 ; ca 256 ; la 32
I: Automount /lfs succeeded
*** Booting Zephyr OS build v4.1.0-6569-gf4a0beb2b7b1 ***
Sample program to r/w files on littlefs
Area 0 at 0x154000 on flash_controller@40029000 for 1048576 bytes
/lfs automounted
/lfs: bsize = 256 ; frsize = 16384 ; blocks = 64 ; bfree = 62

Listing dir /lfs ...
/lfs/boot_count read count:0 (bytes: 0)
/lfs/boot_count write new boot count 1: [wr:1]
I: Test file: /lfs/pattern.bin not found, create one!
------ FILE: /lfs/pattern.bin ------
01 55 55 55 55 55 55 55 02 55 55 55 55 55 55 55
03 55 55 55 55 55 55 55 04 55 55 55 55 55 55 55
05 55 55 55 55 55 55 55 06 55 55 55 55 55 55 55
07 55 55 55 55 55 55 55 08 55 55 55 55 55 55 55
09 55 55 55 55 55 55 55 0a 55 55 55 55 55 55 55
0b 55 55 55 55 55 55 55 0c 55 55 55 55 55 55 55
0d 55 55 55 55 55 55 55 0e 55 55 55 55 55 55 55
0f 55 55 55 55 55 55 55 10 55 55 55 55 55 55 55
11 55 55 55 55 55 55 55 12 55 55 55 55 55 55 55
13 55 55 55 55 55 55 55 14 55 55 55 55 55 55 55
15 55 55 55 55 55 55 55 16 55 55 55 55 55 55 55
17 55 55 55 55 55 55 55 18 55 55 55 55 55 55 55
19 55 55 55 55 55 55 55 1a 55 55 55 55 55 55 55
1b 55 55 55 55 55 55 55 1c 55 55 55 55 55 55 55
1d 55 55 55 55 55 55 55 1e 55 55 55 55 55 55 55
1f 55 55 55 55 55 55 55 20 55 55 55 55 55 55 55
21 55 55 55 55 55 55 55 22 55 55 55 55 55 55 55
23 55 55 55 55 55 55 55 24 55 55 55 55 55 55 55
25 55 55 55 55 55 55 55 26 55 55 55 55 55 55 55
27 55 55 55 55 55 55 55 28 55 55 55 55 55 55 55
29 55 55 55 55 55 55 55 2a 55 55 55 55 55 55 55
2b 55 55 55 55 55 55 55 2c 55 55 55 55 55 55 55
2d 55 55 55 55 55 55 55 2e 55 55 55 55 55 55 55
2f 55 55 55 55 55 55 55 30 55 55 55 55 55 55 55
31 55 55 55 55 55 55 55 32 55 55 55 55 55 55 55
33 55 55 55 55 55 55 55 34 55 55 55 55 55 55 55
35 55 55 55 55 55 55 55 36 55 55 55 55 55 55 55
37 55 55 55 55 55 55 55 38 55 55 55 55 55 55 55
39 55 55 55 55 55 55 55 3a 55 55 55 55 55 55 55
3b 55 55 55 55 55 55 55 3c 55 55 55 55 55 55 55
3d 55 55 55 55 55 55 55 3e 55 55 55 55 55 55 55
3f 55 55 55 55 55 55 55 40 55 55 55 55 55 55 55

41 55 55 55 55 55 55 55 42 55 55 55 55 55 55 55
43 55 55 55 55 55 55 55 44 55 55 55 55 55 55 55
45 55 aa
I: /lfs unmounted
/lfs unmount: 0

```

dkouba-atym, 2025-06-25:
I can try to put together a reproducible example next week.

A comment on your testing: I usually experience this failure after a full chip erase (`west flash --erase`). Once the filesystem is initialized properly once, its either very rare or I haven't seen it fail.

closed 2025-09-09 by github-actions[bot]