zephyrproject-rtos/zephyr #92031: Python 3.12 needs extra information for Ubuntu 22.04 LTS

jukkar, 2025-06-23:
### Describe the bug

I have Ubuntu 22.04 LTS and update to latest Zephyr which requires python 3.12.
This version is not available for Ubuntu 22.04 but user is required to either compile python from scratch or enable a 3rd party repo called deadsnakes. There are several pages in the Internet for instructions how to install python 3.12 for this Ubuntu version.

The installation instructions for Ubuntu https://docs.zephyrproject.org/latest/develop/getting_started/index.html#select-and-update-os are not correct. The doc says that the instructions are for 20.04 but that is not correct.

Also the text in https://docs.zephyrproject.org/latest/develop/getting_started/index.html#install-dependencies tells to enable kitware-archive if using Ubuntu older than 22.04 but that does not be related to this python 3.12 problem.

I enabled the 3rd party deadsnakes repo to get python 3.12 and managed to install it, but there are now problems with pip which does not get installed automatically and requires manual configuration that I am trying to figure out myself. Currently I have a system where west does not work etc because of this pip issue.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Showstopper – Prevents release or major functionality; system unusable.

### Additional Context

For a seasoned developer / user it might be ok to compile python etc, but for a casual user that wants to try zephyr this is not good. The instructions for the python update should be better.

jasperjonker, 2025-06-23:
One way out of this is using [pyenv](https://github.com/pyenv/pyenv?tab=readme-ov-file#simple-python-version-management-pyenv) instead of system packages/other repos. It also has a plugin to have dedicated virtual environments with different python versions.

Syntax is a bit like this: `pyenv install 3.12` to install python version 3.12, and with [pyenv-virtualenv](https://github.com/pyenv/pyenv-virtualenv?tab=readme-ov-file#pyenv-virtualenv) plugin you can create virtualenvs based on the python version: `pyenv virtualenv 3.12 zephyr-env`. `pyenv activate zephyr-env` will activate it, like a normal virtualenv.

Might not be a proper fix, but at least can get your system running

pdgendt, 2025-06-23:
There are many ways to do this. Using `uv` for example:

```shell
$ uv venv --python 3.12
$ source .venv/bin/activate
```

However, the problem described in the issue is how users should get started, which is a valid one.

zhxt, 2025-06-28:
`zhxt@linux-y3ek:~/zephyrproject/zephyr$ source  ../.venv/bin/activate

(zephyrproject) zhxt@linux-y3ek:~/zephyrproject/zephyr$ west -v  build  -b bes2700yp_devkit/bes2700yp/cm55   samples/hello_world   
-- west build: generating a build system
cmake version 3.30.2 is OK; minimum version is 3.13.1
Running CMake: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/zhxt/zephyrproject/zephyr/build -GNinja -DBOARD=bes2700yp_devkit/bes2700yp/cm55 -S/home/zhxt/zephyrproject/zephyr/samples/hello_world
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/zhxt/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.30.2
CMake Error at /usr/share/cmake-3.30/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Python3: Found unsuitable version "3.10.12", but required is
  at least "3.12.1" (found /usr/bin/python3, found components: Interpreter)

  Hint: The project() command has not yet been called.  It sets up
  system-specific search paths.
Call Stack (most recent call first):
  /usr/share/cmake-3.30/Modules/FindPackageHandleStandardArgs.cmake:601 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.30/Modules/FindPython/Support.cmake:4002 (find_package_handle_standard_args)
  /usr/share/cmake-3.30/Modules/FindPython3.cmake:596 (include)
  /home/zhxt/zephyrproject/zephyr/cmake/modules/python.cmake:41 (find_package)
  /home/zhxt/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:131 (include)
  /home/zhxt/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/zhxt/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/zhxt/zephyrproject/zephyr/build -GNinja -DBOARD=bes2700yp_devkit/bes2700yp/cm55 -S/home/zhxt/zephyrproject/zephyr/samples/hello_world
`
@pdgendt I had tried  using uv to create a venv, but still have issues, WEST_PYTHON was defined with Python path from host env. Did I do something wrong ?

zhxt, 2025-06-28:
@pdgendt okay, fixed with `uv  pip install  pip` and reinstall west. Please ignore the previous message. Thanks!