zephyrproject-rtos/zephyr #92059: The West runner does not work after switching to Python version 3.12.0

tunguyen4585, 2025-06-24:
### Describe the bug

The West runner does not work after switching to Python version 3.12.0

**platform:** no working on multiple boards, I run on frdm_k64f, s32z2xxdc2@B/s32z270/rtu0, mr_canhubk3. This is seems a common issue, not board/vendor specific.

**Please note that this issue does not occur with other Python versions, such as 3.12.1**

### Regression

- [x] This is a regression.

### Steps to reproduce

**reproduce:**
install python 3.12.0
run command "west build -p -b frdm_k64f samples/hello_world"
run command "west debug" or "west flash"

### Relevant log output

```shell
(.venv) jenkins@CPP00194701E:~/workspace/zephyr$ west debug Traceback (most recent call last): File "/home/jenkins/workspace/.venv/bin/west", line 8, in <module> sys.exit(main()) ^^^^^^ File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/west/app/main.py", line 1199, in main app.run(argv or sys.argv[1:]) File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/west/app/main.py", line 278, in run self.run_command(argv, early_args) File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/west/app/main.py", line 584, in run_command self.run_extension(args.command, argv) File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/west/app/main.py", line 708, in run_extension self.cmd = self.extensions[name].factory() ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/west/commands.py", line 565, in __call__ mod = _commands_module_from_file(self.py_file) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/west/commands.py", line 718, in _commands_module_from_file spec.loader.exec_module(mod) File "<frozen importlib._bootstrap_external>", line 994, in exec_module File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed File "/home/jenkins/workspace/zephyr/scripts/west_commands/debug.py", line 11, in <module> from run_common import add_parser_common, do_run_common File "/home/jenkins/workspace/zephyr/scripts/west_commands/run_common.py", line 29, in <module> from runners.core import FileType File "/home/jenkins/workspace/zephyr/scripts/west_commands/runners/__init__.py", line 74, in <module> _import_runner_module(_name) File "/home/jenkins/workspace/zephyr/scripts/west_commands/runners/__init__.py", line 14, in _import_runner_module importlib.import_module(f'runners.{runner_name}') File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module return _bootstrap._gcd_import(name[level:], package, level) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/home/jenkins/workspace/zephyr/scripts/west_commands/runners/canopen_program.py", line 14, in <module> import canopen File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/canopen/__init__.py", line 1, in <module> from canopen.network import Network, NodeScanner File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/canopen/network.py", line 19, in <module> from canopen.node import RemoteNode, LocalNode File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/canopen/node/__init__.py", line 1, in <module> from canopen.node.remote import RemoteNode File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/canopen/node/remote.py", line 4, in <module> from canopen.sdo import SdoClient, SdoCommunicationError, SdoAbortedError File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/canopen/sdo/__init__.py", line 1, in <module> from canopen.sdo.base import SdoVariable, SdoRecord, SdoArray File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/canopen/sdo/base.py", line 7, in <module> from canopen import objectdictionary File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/canopen/objectdictionary/__init__.py", line 281, in <module> class ODVariable: File "/home/jenkins/workspace/.venv/lib/python3.12/site-packages/canopen/objectdictionary/__init__.py", line 290, in ODVariable INTEGER24: IntegerN(24), ^^^^^^^^^^^^ TypeError: Struct() argument 1 must be a str or bytes object, not int
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- OS: Ubuntu 20.04 LTS, Windows 11
- Toolchain: zephyr-sdk-0.17.0
- Commit:  af4e875ea5b79ec2bf0b046939e5d802cc40fa8c

tunguyen4585, 2025-06-24:
> [@tunguyen4585](https://github.com/tunguyen4585) from your comment here [#92019 (comment)](https://github.com/zephyrproject-rtos/zephyr/discussions/92019#discussioncomment-13549764) it seems that you traced back the observed behavior to a specific commit. Add the commit sha in the issue description and tick the "is a regression" box if you believe this is a regression.

No, I just ask him since he introduce to require to use python3.12 or higher version and i installed python3.12.0 and found the issue. I've checked before his PR the issue has also happened already. It may relate to the combability between python3.12.0 and canopen module

pdgendt, 2025-06-24:
Are you asking to make 3.12.1 the explicit minimum version? Why don't use the latest patch release? Python is currently at 3.12.11 with many bugfixes.

tunguyen4585, 2025-06-26:
I opened a [PR](https://github.com/zephyrproject-rtos/zephyr/pull/92207) to update the document

commit c68d0b96 (2025-06-27): doc: correct python minimum required

Fixes #92059

Update getting started guide, migration-guide and
python.cmake to use Python 3.12.1 as minimum supported
version
closed 2025-06-27 by dkalowsk