zephyrproject-rtos/zephyr #92063: Arduino Opta ethernet doesn't work immediately after firmware update

nathaniel-greaseboss, 2025-06-24:
### Describe the bug

Hey Zephyr team, I hope you doing well. I'm hoping for a little help in debugging a problem I've been experiencing. Immediately after doing a firmware update on an Arduino opta I am unable to get the ethernet device to initialise correctly. The simplest workaround I've found so far is to unplug the power and plug it back in again after which the ethernet works fine. As we plan on having these devices remotely deployed we aren't going to be able to do a complete cold power cycle. So as it currently stands we wouldn't be able to remotely firmware update these devices without ethernet breaking which is essential to our use case.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Build and flash the telnet sample project `west build --board arduino_opta/stm32h747xx/m7 --pristine always samples/net/telnet && west flash`
2. Double press reset button on the device to complete the upload.
3. Plugin in ethernet cable (In my case on a mac).
4. Test that the ethernet is working as expected. (In my case it wasn't).
5. Unplug the arduino opta USB-C wait 5-10s. Hitting the reset button doesn't recover the ethernet device.
6. Plug the USB-C back in again.
7. At this point my ethernet is works fine again.

Other variations of this problem I've experienced while attempting to debug;
- I've tried this with the stock arduino bootloader, and a more vanilla mcuboot bootloader paired with mcumgr in our own application. Both have the same problem. (Note: that I had to solder on a jtag header to override the bootloader).
- I've tried this whole process on two other opta's where I haven't made any modifications to the PCB or bootloader to the same effect.

NOTE: I tried some of the sample arduino apps and was able to get ethernet working fine directly after a reboot. So I think I can safely rule out a hardware problem.

### Relevant log output

```shell
Truncated output of `device list` on zephyr shell

- ethernet@40028000 (DISABLED)
  DT node labels: mac
- mdio (READY)
  DT node labels: mdio
- ethernet-phy@0 (READY)
uart:~$

Truncated output of logging from our own internal app;

E: HAL_ETH_Init failed: 1
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

- OS: Macos
- Zephyr Version: v4.1.0
- Toolchain: Zephyr SDK

### Additional Context

- Hardware: Opta Powered via USB-C and plugged into a mac via a usb->ethernet adapter. No external peripherals are attached to the OPTA.

nathaniel-greaseboss, 2025-06-24:
Hey @pillo79, thanks for being so quick to respond. I ran into a seperate build issue with your change that looks unrelated;

```
devicetree error: Label 'storage_partition' appears on /soc/flash-controller@52002000/flash@8000000/partitions/partition@1C0000 and on /soc/spi@52005000/qspi-nor-flash@0/partitions/partition@0
```

I just cherry-picked your commit on top of the zephyr mainline 4.1-branch release, which seems to build fine. I've just flashed that to the board. However it seems to be hanging on the `k_busy_wait` call i.e. it's still stuck on that call after a minute or so. e.g.
```
(gdb) bt
#0  sys_clock_cycle_get_32 () at /Users/nathanielbrough/projects/endpoint/zephyr/drivers/timer/cortex_m_systick.c:360
#1  0x0807197a in arch_k_cycle_get_32 () at /Users/nathanielbrough/projects/endpoint/zephyr/include/zephyr/arch/arm/misc.h:26
#2  k_cycle_get_32 () at /Users/nathanielbrough/projects/endpoint/zephyr/include/zephyr/kernel.h:1929
#3  z_impl_k_busy_wait (usec_to_wait=usec_to_wait@entry=200) at /Users/nathanielbrough/projects/endpoint/zephyr/kernel/busy_wait.c:27
#4  0x08041d86 in k_busy_wait (usec_to_wait=200) at /Users/nathanielbrough/projects/endpoint/build/app/zephyr/include/generated/zephyr/syscalls/kernel.h:168
#5  board_gpio_init () at /Users/nathanielbrough/projects/endpoint/zephyr/boards/arduino/opta/board_gpio_init.c:41
#6  0x0805d384 in z_sys_init_run_level (level=level@entry=INIT_LEVEL_PRE_KERNEL_1) at /Users/nathanielbrough/projects/endpoint/zephyr/kernel/init.c:371
#7  0x0805d686 in z_cstart () at /Users/nathanielbrough/projects/endpoint/zephyr/kernel/init.c:775
#8  0x08045618 in z_prep_c () at /Users/nathanielbrough/projects/endpoint/zephyr/arch/arm/core/cortex_m/prep_c.c:215
#9  0x080453fe in z_arm_reset () at /Users/nathanielbrough/projects/endpoint/zephyr/arch/arm/core/cortex_m/reset.S:174
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

I'm going to experiment with the init priorities as I'm guessing that the clocks are perhaps not initialised at this point so it's hanging?

Many thanks again for being so quick to respond it's very much appreciated!

pillo79, 2025-06-25:
Thanks for testing and investigation, I had a hunch it was too early for that function but gave it a try anyway. 🙂
Used a longer timeout to actually comply with datasheet specs.

PR #92145 (2025-06-26): boards: opta: shape the LAN controller reset pulse
Instead of assuming the reset pin is low at boot, create a reset pulse by setting the pin low long enough for the LAN controller to recognize it.

Fixes #92063.
closed 2025-06-26 by dkalowsk