zephyrproject-rtos/zephyr #92123: CPU Frequency Scaling Subsystem

seankyer, 2025-06-24:
### Problem Description

Zephyr currently lacks support for CPU frequency scaling. As a result, SoCs that offer this feature cannot leverage it through the Zephyr kernel. Developers wishing to use CPU frequency scaling must then resort to using bare-metal APIs and implement this power management feature manually at their application level.
This introduces issues:
-	Portability: The developers application becomes coupled to their SoC, a violation of Zephyr’s promise of portability
-	Complexity: The developer must now manage the complexity of performance and power management themselves

### Proposed Change (Summary)

Introducing a CPU Frequency Scaling (CPUFreq) subsystem to Zephyr will standardize performance management. This subsystem will shift the responsibility of managing CPU frequency from the application layer to the OS, improving portability and reducing complexity for developers wishing to leverage this feature of their SoC/MCU. This subsystem should behave similarly to the Power Management subsystem and transparently make performance decisions for the device running Zephyr, given a policy.

To support this, the following capabilities must be supported:
-	CPU Load Tracking: The kernel must track the CPU utilization over time -> Informing performance decisions
-	Frequency Scaling Mechanism: The CPUFreq subsystem should evaluate the CPU load, select an appropriate performance state (P-state) from a policy and adjust the CPU frequency/voltage configuration accordingly (logic implemented by a driver)

This addition will enable Zephyr to support dynamic performance scaling on capable SoCs, enabling users to transparently make use of those features.

### Proposed Change (Detailed)

**Background**
Many modern microcontrollers and system-on-chip’s are equipped with dynamic frequency and voltage scaling (DFVS) to adjust their configuration depending on workload. This enables a balance between performance and energy efficiency.

Since Zephyr lacks the capability to leverage this feature, it limits the ability of developers, using Zephyr, to take full advantage of their hardware, especially in power sensitive applicates such as:
- 	Battery-powered IoT devices, where battery life is critical
- 	Wearables and sensors, which may operate under temperature or energy constraints
- 	Edge computing nodes, where workloads may be bursty, but power consumption is still of concern

The proposed implementation of this subsystem is based on previous art of Linux and its CPUFreq subsystem ([CPU Performance Scaling — The Linux Kernel documentation](https://www.kernel.org/doc/html/latest/admin-guide/pm/cpufreq.html)). 

Definitions
P-state -> “Performance State”, a static CPU frequency/voltage configuration that is defined by the SoC

**Overview**
To support CPU Frequency Scaling, the following must be added:
- 	CPU Load Detection: The Zephyr kernel must track the idle time and load of the CPU
- 	CPU Frequency Scaling Subsystem: Scalable and portable framework for CPU Frequency Scaling
    - Method of Selecting Appropriate P-state: Policy algorithm(s) for determining appropriate P-state
    - Method of Switching into P-state: Infrastructure for implementing the SoC determined ‘CPUFreq’ drivers
    - Devicetree Definitions: An SoC must describe what P-states are available and what the characteristics of each is
 
The Linux kernel’s architecture for this feature contains 3 software components.
- 	Core: Common code infrastructure for user space interfaces for all platforms that support CPU performance scaling
- 	Scaling Governors: Algorithms to estimate the required CPU capacity. Each governor implements one scaling algorithm
- 	Scaling Drivers: Talk to hardware and provide scaling governors with information on the available P-states and access to platform-specific hardware interfaces to change CPU P-state

These software layers allow for any policy algorithm to work with any scaling driver, decoupling the software feature from the hardware specific details. This proposal will follow the same architectural design, ensuring portability of the feature between SoCs. In the proposed system-level diagram, shown later, I will map the layers suggested in this proposal to their Linux equivalents (they are strikingly similar).

This proposal will use the ‘on-demand’ governor policy from Linux as the initial implemented policy. This subsystem is designed such that additional policies can be added in a modular manner in the future. On-demand is chosen as the initial policy due to the simplicity of ‘first fit’ which should generally provide effective frequency scaling in response to computational load. 

**CPU load detection**
The Zephyr kernel currently supports CPU load calculation as part of the debug subsystem. CPU load is calculated by leveraging tracing hooks, which are called before and after a CPU goes into idle, like so:
 
[image]

CPU load also considers the time spent in the interrupt context as well. The counter by default uses k_cycle_get_32() however this can be replaced with another counter if needed in the device trees.

_Reference: [CPU load — Zephyr Project Documentation](https://docs.zephyrproject.org/latest/services/debugging/cpu_load.html)_

As part of this feature, CPU load detection should be promoted to its own subsystem and decoupled from the debug subsystem, enabling it to be used in a production context. This subsystem can run as a workqueue and have its frequency configured via Kconfig option by the user, as explained in the next section.

**Calculating CPU Load Periodically**
The calculation for CPU load cannot simply be calculated by  `(idle_time)/(k_uptime())`, as the CPU load will show as being disproportionately low as the lifetime of the device (denominator) increases. This will result in the device being unresponsive to bursts of activity. Instead, CPU load should be calculated on a periodic basis, considering the change in idle time since the last calculation occurred.

The current code for CPU load relies on the caller of the `cpu_load_get` function to reset the count. So, in the promotion of this logic to its own subsystem, we must add the behavior that the CPU load be evaluated at some set frequency. This can be achieved by registering a workqueue job to run `cpu_load_get`, resetting the load count at some set interval (tuned to the user’s system requirements). A new API to the CPU load subsystem will be `cpu_load_get_latest()` which will return the most recently calculated CPU load over the previous period. This value will be relied upon in the CPU frequency subsystem to make performance mode transitions.

Calculating the CPU load on a rolling basis should leverage a workqueue. The frequency of the job should run periodically as configured by a Kconfig. That means CPU load % can be interpreted as ‘CPU load over the past CONFIG_CPU_LOAD_PERIOD-ms’. This Kconfig can be modified by users, to tune the responsiveness of the CPU load subsystem.

Pseudo code for CPU load subsystem:
```
cpu_load = 0;
[... 112 lines cut ...]

[image]

Annotated is how each component of this system could be mapped to the Linux equivalent.

**Linux and Zephyr References**
[CPU Performance Scaling — The Linux Kernel documentation](https://www.kernel.org/doc/html/latest/admin-guide/pm/cpufreq.html)
[linux/drivers/cpufreq/cpufreq_governor.c at master · torvalds/linux](https://github.com/torvalds/linux/blob/master/drivers/cpufreq/cpufreq_governor.c)
[linux/drivers/cpufreq/cpufreq_ondemand.c at master · torvalds/linux](https://github.com/torvalds/linux/blob/master/drivers/cpufreq/cpufreq_ondemand.c)
[zephyr/soc/adi/max32/power.c at main · zephyrproject-rtos/zephyr](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/adi/max32/power.c)
[CPU load — Zephyr Project Documentation](https://docs.zephyrproject.org/latest/services/debugging/cpu_load.html)

**Initial Hardware for Implementation**
This feature can be developed initially for the MAX32655 which supports adjusting clock frequency at runtime.

### Dependencies

Since this is a new feature, I am not aware of other areas of the project it could impact. CPUFreq may be impacted by 
 [[RFC] Introduce Clock Management Subsystem (Clock Driver Based) by danieldegrasse · Pull Request #72102 · zephyrproject-rtos/zephyr](https://github.com/zephyrproject-rtos/zephyr/pull/72102), which may add support for dynamic clock frequency updates.

### Alternatives Considered

Since there exists previous art from the Linux kernel, it makes sense to base the Zephyr feature from there and modify it depending on what infrastructure already exists within the Zephyr tree.

Considered hooking this feature directly into the Zephyr power management subsystem and having it be part of the PM hooks that are ran in the idle thread. However, CPUFreq can be separated and used in isolation from power modes. Doing so makes the system more modular and the impacts of this PR more isolated.

cfriedt, 2025-07-08:
* minor note based on today's arch meeting
  * potential kconfig-controlled or machine-specific policy to take thermal limits into account for frequency scaling

ttmut, 2025-07-09:
Won't you need to readjust peripheral clocks once you

bjarki-andreasen, 2025-07-09:
> Just to clarify, if you scale the CPU frequency, wouldn’t that also affect the peripheral clock? On MAX32655, both are tied to the main system clock.

depends on the hardware, on the nrf54h20 it does not, on many other devices, it super does, like the sam4e :) Something like https://github.com/zephyrproject-rtos/zephyr/pull/72102 would probably be needed there (or some highly coupled code in the "p state driver" implementation)

nashif, 2025-07-09:
Hi @seankyer ,
Thank you for the discussion in the arch WG.

after the discussion, I looked into cpu_load again and there more things to address, beside those you mention above not being standalone etc:

- Right cpu_load  is only enabled for riscv and cortex-m, not sure why
- It does not support multi-core/SMP

We have runtime stats feature in the kernel tracking this type of information already (and a bit more), we probably should look into streamlining those and making them available to fre quency scaling instead of overhauling cpu_load and duplicating existing functionality.
This already available feature does support all architectures, can use different time sources and also support multi-core. See kernel/usage.c for some of the code involved.

andyross, 2025-07-16:
Driving by: note that the Linux architecture kinda/sorta failed here.  Real CPUs end up driving their own frequency and power state selection in many (most?) cases.  The idea of an OS governor had to be extended and revisited repeatedly.  The embedded world is if anything likely to be **less** consistent and **more** complicated than bigcore linux parts.

I guess my gut says that what we want here is not to come up with a big design with nouns and jargon, but to identify the absolute minimum set of OS features required to support this feature on the devices we know about.

Rather than proposing what we need to add, why don't we start with statements like "We cannot support frequency scaling because XXXX" and then start thinking about how to add XXXX.