zephyrproject-rtos/zephyr #92157: Crypto problems when using WiFi and BLE

cgbdevelco, 2025-06-25:
### Describe the bug

Target Platform: STM32H755 with shields: nRF7002EK (WiFi) and x_nucleo_wb05kn1_spi (BLE)
Bug: Build error with crypto "/zephyr-project-v4.1.0/modules/crypto/mbedtls/include/mbedtls/check_config.h:96:2: error: #error "Unsupported partial support for ECC curves acceleration, see docs/driver-only-builds.md"

I don't think the error is related to STM32H755 as the error also occurs on other targets such as STM32H753 and nRF5340

### Regression

- [ ] This is a regression.

### Steps to reproduce

The easiest way to reproduce the build error without too many modifications:
1. Use peripheral_hr as sample project
2. In CMakeList add set(SHIELD x_nucleo_wb05kn1_spi nrf7002ek) or use the shield argument in west
3. In prj.conf add CONFIG_NETWORKING=y and CONFIG_WIFI=y
4. Build for nucleo_h753zi/stm32h753xx

Observe that build fails with the ECC curve problem

### Relevant log output

```shell
In Zephyr 4.1 building to nRF5340DK (not using x_nucleo_wb05kn1_spi) fails with this setup, however if I build using NCS 3.0, it builds successfully.

I haven't found the difference of the setup beside the build log that when building using NCS has the following output

=========== Generating psa_crypto_config ===============
Backup: CONFIG_MBEDTLS_PSA_CRYPTO_SPM: False
Backup: CONFIG_MBEDTLS_PSA_CRYPTO_C: True
Backup: CONFIG_MBEDTLS_PSA_CRYPTO_KEY_ID_ENCODES_OWNER: False
Backup: CONFIG_MBEDTLS_PLATFORM_SETUP_TEARDOWN_ALT: False
Backup: CONFIG_MBEDTLS_THREADING: False
Backup: CONFIG_MBEDTLS_THREADING_ALT: True
=========== Checkpoint: backup ===============
Restore: CONFIG_MBEDTLS_PSA_CRYPTO_SPM: False
Restore: CONFIG_MBEDTLS_PSA_CRYPTO_C: True
Restore: CONFIG_MBEDTLS_PSA_CRYPTO_KEY_ID_ENCODES_OWNER: False
Restore: CONFIG_MBEDTLS_PLATFORM_SETUP_TEARDOWN_ALT: False
Restore: CONFIG_MBEDTLS_THREADING: False
Restore: CONFIG_MBEDTLS_THREADING_ALT: True
=========== End psa_crypto_config ===============
=========== Generating psa_crypto_library_config ===============
Backup: CONFIG_MBEDTLS_PSA_CRYPTO_C: True
Backup: CONFIG_MBEDTLS_PSA_CRYPTO_KEY_ID_ENCODES_OWNER: False
Backup: CONFIG_MBEDTLS_PSA_CRYPTO_SPM: False
Backup: CONFIG_MBEDTLS_USE_PSA_CRYPTO: False
Backup: CONFIG_MBEDTLS_PLATFORM_PRINTF_ALT: False
Backup: CONFIG_MBEDTLS_THREADING: False
Backup: CONFIG_MBEDTLS_THREADING_ALT: True
=========== Checkpoint: backup ===============
Restore: CONFIG_MBEDTLS_PSA_CRYPTO_C: True
Restore: CONFIG_MBEDTLS_PSA_CRYPTO_KEY_ID_ENCODES_OWNER: False
Restore: CONFIG_MBEDTLS_PSA_CRYPTO_SPM: False
Restore: CONFIG_MBEDTLS_USE_PSA_CRYPTO: False
Restore: CONFIG_MBEDTLS_PLATFORM_PRINTF_ALT: False
Restore: CONFIG_MBEDTLS_THREADING: False
Restore: CONFIG_MBEDTLS_THREADING_ALT: True
=========== End psa_crypto_library_config ===============
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

Zephyr SDK 4.1

danieldegrasse, 2025-07-01:
@cgbdevelco are you able to reproduce this issue with the v4.2.0-rc1 release of Zephyr (or the latest main)?

ceolin, 2025-07-02:
@danieldegrasse it is reproducible. I am looking it.

ceolin, 2025-07-11:
@cgbdevelco I don't think we have enough time to get it in 4.2

It is not clear for me the total scope of the problem yet, it may be just a matter of change configuration to not use hw accelerator or if it really needs that and it is not available in these build configurations.

That's said I believe is too risky hurrying a workaround / fix at this moment. We have to many platform specific patches in our TF-M fork (less than previous release though) and we are working to fix this technical debt.

ceolin, 2025-07-11:
> [@ceolin](https://github.com/ceolin) Do you have any suggestions on what configs could be changed to make it build? Tried playing around with CONFIG_WIFI_NM_WPA_SUPPLICANT_CRYPTO_ALT without any success. Is it easier to change crypto method for BLE?
> 
> I don't see this bug as platform specific. If you want to use BLE and Wi-Fi in Zephyr right now it simply does not work.

I don't know yet what is triggering the need for the HW crypto driver. As far as I could see there is only one driver in TF-M doing it.

mfjDevelco, 2025-07-28:
Hello, 

I am facing the same problem. Are there any updates regarding the issue?

kalpeshsthorat-dev, 2025-09-16:
Hi folks, I am also getting below error when porting to zephyr 4.2 from zephyr 4.1 
PS: This error is coming only when BLE is enabled.

96 | #error "Unsupported partial support for ECC curves acceleration, see docs/driver-only-builds.md"

can someone help in this regard ?

closed 2025-11-30 by github-actions[bot]