zephyrproject-rtos/zephyr #92163: Pushing breaking changes to main while referencing it from Getting Started

jmasek, 2025-06-25:
### Describe the bug

Hi, 
I am quite surprised that you are pushing breaking changes like https://github.com/zephyrproject-rtos/zephyr/commit/5a9a39caf3bf5195a38194bb844a428eaa7f2bb8 to the main branch while the default setting of west when proceeding through [Getting Started](https://docs.zephyrproject.org/latest/develop/getting_started/index.html)  is checkout main. 
I was not able to find any way to detect whether Zephyr is running a stable release (v4.1) or the main branch, kernel version is not updated. Therefore, as a module developer, I cannot react to these breaking changes. Is it correct?

It means that if a newcomer goes through your official Getting Started and adds the module (External project), they will see compilation errors because the main branch uses the latest stable tag version, right?

Am I correct, or do I miss something? Is such an experience intended?

Thanks

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

roberthusak, 2025-06-25:
> > The problem is that the version is still 4.1.0 even though the code already contains changes that will be published in 4.2.0.
> 
> Current Zephyr version in main is 4.1.99 and not 4.1.0
> 
> ```
> $ cat VERSION 
> VERSION_MAJOR = 4
> VERSION_MINOR = 1
> PATCHLEVEL = 99
> VERSION_TWEAK = 0
> EXTRAVERSION =
> ```

Thanks a lot, we didn't notice this file. According to the message `*** Booting Zephyr OS build v4.1.0-4735-gbcf7717e9744 ***` received during booting we assumed that the version is still 4.1.0.

Having a way to identify the breaking changes precisely (such as the presence of a specific macro for each such change) would still be more robust, but this should help as well 👍 Is there a convention that patch level 99 means that we're in an unstable Zephyr version?

kartben, 2025-07-02:
> However and I think this could be the bug, if you go to [docs.zephyrproject.org/4.1.0](https://docs.zephyrproject.org/4.1.0/)
> 
> The getting started docs should probably be about that specific release and checking out that release, not the tip of the main branch.
> 
> Maybe we need to changed this in released documentation.

#92564 :)

roberthusak, 2025-07-03:
@nashif @kartben Thank you both for addressing this issue in documentation.

> So, when you go to documentation at https://docs.zephyrproject.org/latest/
> 
> it clearly says: Welcome to the Zephyr Project’s documentation for the main tree under development .....
> 
> The documentation is about a code base under development and how to setup your environment target the tip of the tree and not a specific release.

You're right that it's written clearly in the docs homepage. However, many people might arrive directly to the Getting Started Guide (via Google search, https://www.zephyrproject.org, ...) and don't see this message. Furthermore, while the Getting Started Guide for 4.1.0 contains the note: "The [latest development version](https://docs.zephyrproject.org/latest/develop/getting_started/index.html) of this page may be more current than this released 4.1.0 version," there's no such note in the Getting Started Guide for the latest version. Wouldn't it be useful to mention something like: "The [latest release 4.1.0](https://docs.zephyrproject.org/4.1.0/develop/getting_started/index.html) might be more stable than the development version referenced in this page?"

In general, it seems to me that the documentation of the latest development version is the most preferred one. For example, https://docs.zephyrproject.org redirects to https://docs.zephyrproject.org/latest/ and the link to Getting Started Guide from https://www.zephyrproject.org points to https://docs.zephyrproject.org/latest/develop/getting_started/index.html.

Wouldn't it be better to redirect these (and maybe other popular links as well) to https://docs.zephyrproject.org/4.1.0/? Intuitively, it seems to me that it's better for newcomers to start with using a stable version of a project and move to the latest development version only after they gain some experience with it. Or is there an aspect I don't see?

> However and I think this could be the bug, if you go to https://docs.zephyrproject.org/4.1.0/
> 
> The getting started docs should probably be about that specific release and checking out that release, not the tip of the main branch.
> 
> Maybe we need to changed this in released documentation.

I think this will certainly help 👍 However, if the latest development version is still the preferred one in documentation, I'm afraid most people won't visit the Getting Started Guide of the release. Therefore, as we mentioned before, it would be hard for them to use external modules reliably, because their authors can't react to backward incompatible changes in between releases.

nashif, 2025-07-03:
we can always add a note about releases vs development at the spot where someone is asked to pull the tree using west, some kind of disclaimer that you are pulling code under development, expect things to be WIP etc...