zephyrproject-rtos/zephyr #92224: BT_LONG_WQ stack size to low

cgbdevelco, 2025-06-26:
### Describe the bug

Hi,
While trying to circumvent the bug in #92157 i disabled 
`CONFIG_BT_SMP=n`

And added the two configs below to make it build successfully.
```
CONFIG_PSA_WANT_KEY_TYPE_ECC_PUBLIC_KEY=y
CONFIG_PSA_WANT_KEY_TYPE_RSA_PUBLIC_KEY=y
```
However I found that stack overflow occurs for BT_LONG_WQ
Increasing BT_LONG_WQ_STACK_SIZE in Kconfig from 1300 (BT_GATT_CACHING is enabled) to 1400 removed the stack overflow.
When CONFIG_DEBUG_OPTIMIZATIONS=y is replaced with CONFIG_SIZE_OPTIMIZATIONS=y the default value of 1300 does not cause any stack overflow.

### Regression

- [ ] This is a regression.

### Steps to reproduce

# General config
CONFIG_LOG=y
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
CONFIG_MAIN_STACK_SIZE=5200
CONFIG_CRC=y
CONFIG_HWINFO=y

# Just for debugging
CONFIG_LOG_MODE_DEFERRED=y

# USB
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_NETWORK_ECM=n
CONFIG_USB_DEVICE_NETWORK_RNDIS=y
CONFIG_USB_DEVICE_OS_DESC=y
CONFIG_USB_COMPOSITE_DEVICE=y
CONFIG_USB_DRIVER_LOG_LEVEL_INF=n
CONFIG_USB_DEVICE_LOG_LEVEL_INF=n

# Ethernet
CONFIG_ETH_STM32_HAL=n

# Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_POSIX_API=y
CONFIG_NET_IF_MAX_IPV4_COUNT=2
CONFIG_NET_DHCPV4_SERVER_INSTANCES=2
CONFIG_NET_MAX_CONN=10
CONFIG_ZVFS_OPEN_MAX=32

# Network address config
CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_DHCPV4_SERVER=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.1.1"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.168.1.1"

# Bluetooth config
CONFIG_BT=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_DIS=y
CONFIG_BT_DIS_PNP=n
CONFIG_BT_AUTO_PHY_UPDATE=n
CONFIG_BT_SMP=n
CONFIG_PSA_WANT_KEY_TYPE_ECC_PUBLIC_KEY=y
CONFIG_PSA_WANT_KEY_TYPE_RSA_PUBLIC_KEY=y
CONFIG_BT_BONDABLE=n
CONFIG_BT_DIS_MODEL_NUMBER_STR="Zephyr"
CONFIG_BT_DIS_MANUF_NAME_STR="Zephyr"
CONFIG_BT_DIS_FW_REV=y
CONFIG_BT_DIS_FW_REV_STR="0.1.0"
CONFIG_BT_DEVICE_NAME_DYNAMIC=y

# STM32WB05KN config
CONFIG_SPI_STM32_INTERRUPT=y
CONFIG_BT_BLUENRG_ACI=y
CONFIG_BT_HCI_ACL_FLOW_CONTROL=n

# Allow for large Bluetooth data packets
CONFIG_BT_L2CAP_TX_MTU=252
CONFIG_BT_BUF_ACL_RX_SIZE=256

# SPI
CONFIG_SPI=y
CONFIG_SPI_STM32=y

# Core Wi-Fi stack and nRF7002
CONFIG_WIFI=y
CONFIG_WIFI_NM=y
CONFIG_WIFI_NRF70=y
CONFIG_NRF70_SYSTEM_MODE=y
CONFIG_NRF70_AP_MODE=y

# WPA supplicant support for AP mode
CONFIG_WIFI_NM_WPA_SUPPLICANT=y
CONFIG_WIFI_NM_WPA_SUPPLICANT_AP=y

CONFIG_WIFI_NM_WPA_SUPPLICANT_CRYPTO_ALT=y
CONFIG_WIFI_NM_WPA_SUPPLICANT_THREAD_STACK_SIZE=8192
CONFIG_WIFI_NM_WPA_SUPPLICANT_WQ_STACK_SIZE=8192

# Net
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_DHCPV4=y
CONFIG_NET_STATISTICS=n
CONFIG_NET_L2_ETHERNET=y

# Logging for NET
CONFIG_NET_LOG=n
CONFIG_LOG_BUFFER_SIZE=2048

### Relevant log output

```shell
*** Booting Zephyr OS build v4.1.0 ***
[00:00:00.432,000] <inf> main: Hello from nucleo_h755zi_q/stm32h755xx/m7

[00:00:00.432,000] <inf> wifi_supplicant: wpa_supplicant initialized
[00:00:00.484,000] <inf> bt_hci_core: Identity: 02:80:E1:00:00:00 (public)
[00:00:00.484,000] <inf> bt_hci_core: HCI: version 5.3 (0x0c) revision 0x2053, manufacturer 0x0030
[00:00:00.484,000] <inf> bt_hci_core: LMP: version 5.3 (0x0c) subver 0x0021
[00:00:00.484,000] <inf> bleHdlr: Bluetooth initialized
[00:00:00.484,000] <inf> SPI: SPI device is ready
[00:00:00.489,000] <inf> bleHdlr: Advertising successfully started
[00:00:00.494,000] <err> os: ***** MPU FAULT *****
[00:00:00.494,000] <err> os:   Stacking error (context area might be not valid)
[00:00:00.494,000] <err> os:   Data Access Violation
[00:00:00.494,000] <err> os:   MMFAR Address: 0x2400f418
[00:00:00.494,000] <err> os: r0/a1:  0x2400f418  r1/a2:  0x00000000  r2/a3:  0x00000000
[00:00:00.494,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x00000000
[00:00:00.494,000] <err> os:  xpsr:  0x0805a600
[00:00:00.494,000] <err> os: Faulting instruction address (r15/pc): 0x8e1834ca
[00:00:00.494,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:00.494,000] <err> os: Current thread: 0x24002790 (BT LW WQ)
[00:00:00.495,000] <err> os: Halting system
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

jhedberg, 2025-06-26:
Generally, when you change the default compiler optimization settings, it will have an impact on stack size usage, and the expectation is for the app developer to then fine-tune the app configuration since it's not really possible to automate this for every possible case. The Kconfig option for this also has a prompt specifically so that it can be fine tuned.

That said, I think it may be justifiable to update the default value in this case to 1400, if you want to propose a pull request for that?

jhedberg, 2025-06-26:
I'll also relabel this as an enhancement, since if things work with the default optimization options then that's all that we need to ensure works for sure.

cgbdevelco, 2025-06-26:
@jhedberg CONFIG_BT_LONG_WQ_STACK_SIZE is not user-configurable.

Regarding the default value, do you suggest that 1400 is the new default value or only when e.g. CONFIG_SIZE_OPTIMIZATIONS=y?

Strange thing though is that if I disable CONFIG_WIFI_NM_WPA_SUPPLICANT_AP=n, it runs fine without stack overflow when stack size is 1300

jhedberg, 2025-06-27:
> Do you still wish a PR on the merged conditions equal 1400?

This is part of a more wider topic of inspecting all of our stack sizes. We should try to simplify (ideally remove) their conditional defaults, and make the unconditional default such that they work on as many configurations as possible. Reducing memory footprint to the bare minimum can then be left as an application-specification optimization task. I'm not saying that you have to do this, but if you want to pick up this task feel free :)

As for this one, I'll close it, since the way it's presented it really shouldn't be an issue anymore (false assumptions about non-user configurability).