zephyrproject-rtos/zephyr #92265: Bluetooth: bond deleted before OOB security validation

JvanDooren, 2025-06-26:
### Describe the bug

Zephyr v4.1
Platform ESP32S3

project config:

CONFIG_BT=y
CONFIG_BT_SMP=y
CONFIG_BT_PERIPHERAL=y
CONFIG_BT_SMP_SC_PAIR_ONLY=y
CONFIG_BT_SMP_SC_ONLY=y
CONFIG_BT_FIXED_PASSKEY=y
CONFIG_BT_BONDING_REQUIRED=y
CONFIG_SETTINGS=y
CONFIG_BT_SETTINGS=y
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_NVS=y
CONFIG_NVS_DATA_CRC=y
CONFIG_BT_KEYS_OVERWRITE_OLDEST=y
CONFIG_BT_KEYS_SAVE_AGING_COUNTER_ON_PAIRING=y
CONFIG_BT_MAX_PAIRED=2

Actual behavior:

2 smartphones have bonded by a static passkey and keys (IPK?) are stored internally in the BLE subsystem.
Any connection from these known smartphones is accepted as keys are valid for those BLE central MAC.

When a 3rd smartphone is used to connect, the bt_conn_auth_info_cb.bond_deleted callback is called before the OOB checks are activated (asking for the passkey). This means that any smartphone can delete previously bonded smartphones, even if it doesn't pass OOB checks.

Expected behavior:

bt_conn_auth_info_cb.bond_deleted callback is called after the OOB checks have passed with success, but before bt_conn_cb.security_changed, bt_conn_cb.identity_resolved and bt_conn_cb.pairing_complete.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. 2 smartphones have bonded by a static passkey and keys (IPK?) are stored internally in the BLE subsystem.
2. Setup a connection using a 3rd smartphone

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

Ubuntu 24.04
Zephyr SDK 17.0
Zephyr 4.1

### Additional Context

[image]

jhedberg, 2025-07-01:
Also, if you're up for it, feel free to propose a pull request to fix it :)

JvanDooren, 2025-07-12:
I can check if if reproduces with a slightly modified samples/bluetooth/peripheral_sc_only.
And yeah, I known that a PR works fastest but I'm lacking the time to dig through the massive stack. I'll see what I can do.

closed 2025-11-26 by github-actions[bot]