zephyrproject-rtos/zephyr #92281: eth_nxp_enet_mac failed with -120

Stanislas-MARTIN, 2025-06-27:
### Describe the bug

I am using an overlay to set a set mac on the rt1170. It was working before commit 3d8196f41b38971c6dc837d908162da5f14e0ecf

### Regression

- [x] This is a regression.

### Steps to reproduce

1. Use the sample `hello_world` and replace the main by 
```c
#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(Main, LOG_LEVEL_INF);

#include <zephyr/kernel.h>
#include <stdio.h>

#include <zephyr/net/net_if.h>
#include <zephyr/net/net_core.h>
#include <zephyr/net/net_context.h>
#include <zephyr/net/net_mgmt.h>

#define DHCP_POLLING_MS (500)

static volatile bool ip_is_assigned = false;
static struct net_mgmt_event_callback mgmt_cb;
static struct net_dhcpv4_option_callback dhcp_cb;

static void start_dhcpv4_client(struct net_if *iface, void *user_data) {
        (void)(user_data);
        LOG_INF("Start interface %s: index=%d", net_if_get_device(iface)->name, net_if_get_by_iface(iface));
        net_dhcpv4_start(iface);
}

static void handler(struct net_mgmt_event_callback *cb, uint32_t mgmt_event, struct net_if *iface) {
        int i = 0;

        if (mgmt_event != NET_EVENT_IPV4_ADDR_ADD) {
                return;
        }

        for (i = 0; i < NET_IF_MAX_IPV4_ADDR; i++) {
                char buf[NET_IPV4_ADDR_LEN];

                if (iface->config.ip.ipv4->unicast[i].ipv4.addr_type != NET_ADDR_DHCP) {
                        continue;
                }

                LOG_INF("   Address[%d]: %s", net_if_get_by_iface(iface), net_addr_ntop(AF_INET, &iface->config.ip.ipv4->unicast[i].ipv4.address.in_addr, buf, sizeof(buf)));
                LOG_INF("    Subnet[%d]: %s", net_if_get_by_iface(iface), net_addr_ntop(AF_INET, &iface->config.ip.ipv4->unicast[i].netmask, buf, sizeof(buf)));
                LOG_INF("    Router[%d]: %s", net_if_get_by_iface(iface), net_addr_ntop(AF_INET, &iface->config.ip.ipv4->gw, buf, sizeof(buf)));
                LOG_INF("Lease time[%d]: %u seconds", net_if_get_by_iface(iface), iface->config.dhcpv4.lease_time);

                ip_is_assigned = true;
        }
}

int main(void)
{
        LOG_INF("NET initialization");

        // Initialize callbacks.
        net_mgmt_init_event_callback(&mgmt_cb, handler, NET_EVENT_IPV4_ADDR_ADD);
        net_mgmt_add_event_callback(&mgmt_cb);
        net_dhcpv4_add_option_callback(&dhcp_cb);
        // Start DHCP on all interfaces.
        net_if_foreach(start_dhcpv4_client, NULL);

        // Wait for IP assignment.
        while (!ip_is_assigned) {
                LOG_INF("Waiting for IP");
                k_sleep(K_MSEC(DHCP_POLLING_MS));
        }

        return 0;
}
```
2. Add the following to the prj.conf
```
CONFIG_NETWORKING=y
CONFIG_NET_LOG=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_IPV6=n
CONFIG_NET_IPV4=y
CONFIG_POSIX_API=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POLL_MAX=4
CONFIG_NET_CONNECTION_MANAGER=y
CONFIG_NET_DHCPV4=y
CONFIG_NET_DHCPV4_OPTION_CALLBACKS=y
CONFIG_NET_INTERFACE_NAME=y
CONFIG_NET_DEFAULT_IF_ETHERNET=y
CONFIG_NET_SOCKETS_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_NET_DRIVERS=y
CONFIG_NET_LOOPBACK=y
CONFIG_POSIX_TIMERS=y
CONFIG_XOPEN_STREAMS=y
```
3. Create an overlay with the following content
```
&enet_mac {
         /delete-property/ zephyr,random-mac-address;
        local-mac-address = [your-mac];
};
```
4. Build with the following command `west build -p -b mimxrt1170_evk/mimxrt1176/cm7 -- -DDTC_OVERLAY_FILE=app.overlay`
5. Flash the rt1170

### Relevant log output

```shell
Error log:
[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 1CC816
[00:00:00.053,000] <err> eth_nxp_enet_mac: phy_configure_link failed with error: -120
[00:00:00.054,000] <inf> eth_nxp_enet_mac: Link is down
[00:00:00.054,000] <err> eth_nxp_enet_mac: phy_configure_link failed with error: -120
*** Booting Zephyr OS build v4.1.0-7032-g3198db1b1229 ***
[00:00:00.055,000] <inf> Main: NET initialization
[00:00:00.055,000] <inf> Main: Start interface ethernet: index=1
[00:00:00.055,000] <inf> Main: Start interface lo: index=2
[00:00:00.055,000] <inf> Main: Waiting for IP
[00:00:00.555,000] <inf> Main: Waiting for IP
[00:00:01.055,000] <inf> Main: Waiting for IP
[00:00:01.555,000] <inf> Main: Waiting for IP
[00:00:02.055,000] <inf> Main: Waiting for IP
[00:00:02.555,000] <inf> Main: Waiting for IP
[00:00:03.056,000] <inf> Main: Waiting for IP
[00:00:03.155,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex
[00:00:03.155,000] <inf> eth_nxp_enet_mac: Link is up
[00:00:03.556,000] <inf> Main: Waiting for IP
[00:00:04.056,000] <inf> Main: Waiting for IP
[00:00:04.556,000] <inf> Main: Waiting for IP
[00:00:05.056,000] <inf> Main: Waiting for IP
[00:00:05.556,000] <inf> Main: Waiting for IP
[00:00:06.056,000] <inf> Main: Waiting for IP
[00:00:06.556,000] <inf> Main: Waiting for IP
[00:00:07.056,000] <inf> Main: Waiting for IP
[00:00:07.557,000] <inf> Main: Waiting for IP
[00:00:08.057,000] <inf> Main: Waiting for IP
--------------------------------------------------------------
Working log

[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 1CC816
[00:00:00.054,000] <inf> eth_nxp_enet_mac: Link is down
*** Booting Zephyr OS build v4.1.0-5129-g05f71f3874d4 ***
[00:00:00.056,000] <inf> Main: NET initialization
[00:00:00.056,000] <inf> Main: Start interface ethernet: index=1
[00:00:00.056,000] <inf> Main: Start interface lo: index=2
[00:00:00.057,000] <inf> Main: Waiting for IP
[00:00:00.557,000] <inf> Main: Waiting for IP
[00:00:01.057,000] <inf> Main: Waiting for IP
[00:00:01.557,000] <inf> Main: Waiting for IP
[00:00:02.058,000] <inf> Main: Waiting for IP
[00:00:02.558,000] <inf> Main: Waiting for IP
[00:00:03.058,000] <inf> Main: Waiting for IP
[00:00:03.558,000] <inf> Main: Waiting for IP
[00:00:04.058,000] <inf> Main: Waiting for IP
[0Hello World! mimxrt1170_evk@B/mimxrt1176/cm7
0:00:04.558,000] <inf> Main: Waiting for IP
[00:00:05.058,000] <inf> Main: Waiting for IP
[00:00:05.456,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex
[00:00:05.456,000] <inf> eth_nxp_enet_mac: Link is up
[00:00:05.459,000] <inf> net_dhcpv4: Received: 10.10.150.144
[00:00:05.459,000] <inf> Main:    Address[1]: 10.10.150.144
[00:00:05.459,000] <inf> Main:     Subnet[1]: 255.255.255.192
[00:00:05.459,000] <inf> Main:     Router[1]: 10.10.150.129
[00:00:05.459,000] <inf> Main: Lease time[1]: 36000 seconds
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

- OS: Windows, Wsl (Ubuntu 24.02)
- Toolchain: Zephyr SDK 0.17
- Commit: 3198db1b12292cf072b69f06000e9e258e529358

Stanislas-MARTIN, 2025-06-27:
@decsny

decsny, 2025-06-30:
Hi, I tested your provided code and configurations on this commit 3d8196f41b3 as you said was the one that broke it.

Specifically, I tested on the board target `mimxrt1170_evk@A//cm7`

Results was this:
```
uart:~$ [00:00:03.013,000] <inf> phy_mc_ksz8081: PHY 0 is up
uart:~$ [00:00:03.014,000] <inf> phy_mc_ksz8081: PHY (0) Link speed 100 Mb, full duplex

uart:~$ [00:00:03.015,000] <inf> eth_nxp_enet_mac: Link is up
uart:~$ *** Booting Zephyr OS build v4.1.0-6241-g3d8196f41b38 ***
uart:~$ [00:00:03.016,000] <inf> Main: NET initialization
uart:~$ [00:00:03.016,000] <inf> Main: Start interface ethernet: index=1
uart:~$ [00:00:03.016,000] <inf> Main: Start interface lo: index=2
uart:~$ [00:00:03.016,000] <inf> Main: Waiting for IP
uart:~$ uart:~$ [00:00:10.018,000] <inf> net_dhcpv4: Received: 192.0.2.5
uart:~$ [00:00:11.016,000] <inf> Main: Waiting for IP
uart:~$ uart:~$ net iface
Hostname: zephyr
Default interface: 1

Interface eth0 (0x80001060) (Ethernet) [1]
===================================
Link addr : 01:02:03:04:05:06
MTU       : 1500
Flags     : AUTO_START,IPv4
Device    : ethernet (0x300247ac)
Status    : oper=UP, admin=UP, carrier=ON
Ethernet capabilities supported:
	TX checksum offload
	RX checksum offload
	10 Mbits
	100 Mbits
	MAC address filtering
Ethernet PHY device: phy@0 (0x30024788)
Ethernet link speed: 100 Mbits full-duplex
IPv4 unicast addresses (max 2):
	192.0.2.5/255.255.255.240 DHCP preferred
IPv4 multicast addresses (max 2):
	224.0.0.1
IPv4 gateway : 192.0.2.2
DHCPv4 lease time : 464
DHCPv4 renew time : 232
DHCPv4 server     : 192.0.2.2
DHCPv4 requested  : 192.0.2.5
DHCPv4 state      : bound
DHCPv4 attempts   : 1
DHCPv4 state      : bound

Interface lo (0x800010f0) (Dummy) [2]
================================
Link addr : 00:00:5E:00:53:FF
MTU       : 576
Flags     : AUTO_START,IPv4
Device    : lo (0x30024740)
Status    : oper=UP, admin=UP, carrier=ON
IPv4 unicast addresses (max 2):
	127.0.0.1/255.0.0.0 autoconf preferred infinite
IPv4 multicast addresses (max 2):
	224.0.0.1
IPv4 gateway : 0.0.0.0
DHCPv4 lease time : 0
DHCPv4 renew time : 0
DHCPv4 server     : 0.0.0.0
DHCPv4 requested  : 0.0.0.0
DHCPv4 state      : selecting
DHCPv4 attempts   : 1
DHCPv4 state      : selecting
uart:~$
uart:~$ [00:00:11.016,000] <inf> Main: Waiting for IP
```

You can see I set the local-mac-address to `[01 02 03 04 05 06]`. The DHCP server I set up provided the board an address of 192.0.2.5 which it bound to no problem.

I can also ping the router of the network which is 192.0.2.2 in my case and it works:
```
uart:~$ net ping 192.0.2.2
PING 192.0.2.2
28 bytes from 192.0.2.2 to 192.0.2.5: icmp_seq=1 ttl=64 time=1 ms
28 bytes from 192.0.2.2 to 192.0.2.5: icmp_seq=2 ttl=64 time=0 ms
```

I'm not sure why your handler here is not getting called but didn't debug that as it didn't seem relevant to the core of the issue which claimed that there was an error in driver init and not getting an IP. This seems not to be the case on that commit. Maybe something else changed since then.  Also, are you using the EVKA or EVKB?

decsny, 2025-06-30:
I also cannot reproduce the issue on zephyr v4.2.0-rc1 tag

Stanislas-MARTIN, 2025-07-01:
Yes, I am using the EVKB

closed 2025-07-05 by dkalowsk

rgruberski, 2025-10-03:
@Stanislas-MARTIN Hey! I have exactly the same problem with a Teensy 4.1 board with an NXP MCU. Would you happen to have any suggestions on how to solve it? I have the latest Zephyr version. The hardware is connected correctly, and the same setup with a simple Arduino example works well. Thanks for any help!

```
uart:~$ [00:00:00.154,000] <inf> eth_nxp_enet_mac: Link is down
*** Booting Zephyr OS build v4.2.0-4907-gc77e5a60fdd1 ***
[00:00:00.155,000] <inf> Main: NET initialization
[00:00:00.155,000] <inf> Main: Start interface ethernet: index=1
[00:00:00.155,000] <inf> Main: Start interface lo: index=2
[00:00:00.155,000] <inf> Main: Waiting for IP
```

When I try to up the interface manually (from the console), I receive a -120 error. The link's LEDs are always off.

Stanislas-MARTIN, 2025-10-03:
Did you try to find a commit where it works on your board.
If yes, you could try to do a `git bisect` to find the commit that broke your setup.
Also are you sure that the function that returns the error is `eth_nxp_enet_mac`.

rgruberski, 2025-10-03:
> Did you try to find a commit where it works on your board.

Yes, I tried, but I have found nothing

> Also are you sure that the function that returns the error is eth_nxp_enet_mac.

I'm not sure. This is what I have in the logs. I tried many different configurations, but I still haven't had success.

rgruberski, 2025-10-03:
I also tried without `local-mac-address` and with `/delete-property/ nxp,unique-mac;`, the result is the same.

rgruberski, 2025-10-03:
It was an excellent suggestion. Thanks! I asked on Discord, and they advised using the latest tagged version, which works well. Simple solution, but I didn't think of it :) I tried other versions, but without package updates after checkout, my bad :)

PR #92498 (2025-07-05): drivers: ethernet: nxp: don't fail with -EALREADY
don't fail with -EALREADY being returned from
phy_configure_link.

Fixes: #92281
commit 19a3b217 (2025-07-05): drivers: ethernet: nxp: don't fail with -EALREADY

don't fail with -EALREADY being returned from
phy_configure_link.

Fixes: #92281