zephyrproject-rtos/zephyr #92299: drivers: video: smartdma: crash or not depending on log level

josuah, 2025-06-27:
### Describe the bug

Building the capture sample on `frdm_mcxn947` board, after the [rework to enable video](https://github.com/nxp-appcodehub/dm-multiple-face-detection-on-mcxn947?tab=readme-ov-file#31-step-1), the capture does work or does not work depending on the log mode.

Below the focus is on `samples/drivers/video/capture` only.

Thank you to @marinjurjevic for discovering and investigating this!

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. `west build -b frdm_mcxn947/mcxn947/cpu0 --shield dvp_20pin_ov7670 samples/drivers/video/capture -- -DCONFIG_LOG_MODE_DEFERRED=y` (and `west flash`) => no crash
2. `west build -b frdm_mcxn947/mcxn947/cpu0 --shield dvp_20pin_ov7670 samples/drivers/video/capture -- -DCONFIG_LOG_MODE_IMMEDIATE=y` (and `west flash`) => crash

### Relevant log output

```shell
With CONFIG_LOG_MODE_DEFERRED=y (default)

*** Booting Zephyr OS build v4.1.0-7085-gba905d045b97 ***
[00:00:00.045,000] <inf> main: Video device: video-sdma
[00:00:00.045,000] <inf> main: - Capabilities:
[00:00:00.045,000] <inf> main:   RGBP width [320; 320; 0] height [240; 240; 0]
[00:00:00.045,000] <inf> main: - Video format: RGBP 320x240
[00:00:00.048,000] <inf> main: - Supported frame intervals for the default format:
[00:00:00.048,000] <inf> main: - Supported controls:
[00:00:00.048,000] <inf> main:          device: ov7670@21
[00:00:00.048,000] <inf> video_ctrls:                  Horizontal Flip 0x00980914 (bool)   (flags=0x00) : min=0 max=1 step=1 default=0 value=0
[00:00:00.048,000] <inf> video_ctrls:                    Vertical Flip 0x00980915 (bool)   (flags=0x00) : min=0 max=1 step=1 default=0 value=0
[00:00:00.048,000] <inf> main: Capture started
uart:~$

With CONFIG_LOG_MODE_IMMEDIATE=y`

[ not a copy-paste error, some lines of log are absent ]

[00:00:00.048,000] <inf> main: - Supported frame intervals for the default format:
[00:00:00.058,000] <inf> main: - Supported controls:
[00:00:00.065,000] <inf> main:          device: ov7670@21
[00:00:00.073,000] <err> os: ***** USAGE FAULT *****
[00:00:00.081,000] <err> os:   Stack overflow (context area not valid)
[00:00:00.090,000] <err> os: r0/a1:  0xaaaaaaaa  r1/a2:  0xaaaaaaaa  r2/a3:  0x1000d3c8
[00:00:00.101,000] <err> os: r3/a4:  0x09000000 r12/ip:  0x100123ce r14/lr:  0x1000c8b5
[00:00:00.112,000] <err> os:  xpsr:  0x5b011600
[00:00:00.120,000] <err> os: Faulting instruction address (r15/pc): 0x0000005b
[00:00:00.130,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:00.140,000] <err> os: Current thread: 0x30000cb8 (main)
[00:00:00.148,000] <err> os: Halting system
uart:~$
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- OS: `Debian 13.0`
- Toolchain: `zephyr-sdk-0.17.0`
- Commit ba905d045b97a437fbd82f8c91204ef5caa2a96b (main)

### Additional Context

As a side-note, the video capture with UVC works (`west build -b frdm_mcxn947/mcxn947/cpu0 --shield dvp_20pin_ov7670 samples/subsys/usb/uvc/`), however as soon as the capture stop, an error message appears (`Bus fault on vector table read`).

teburd, 2025-06-27:
log mode immediate can use a lot more stack, I'd guess a stack overflow

ngphibang, 2025-06-30:
@hakehuang : Maybe because of the log issue ?

hakehuang, 2025-06-30:
> [@hakehuang](https://github.com/hakehuang) : Maybe because of the log issue ?

Thanks @ngphibang @josuah @marinjurjevic  @teburd . add the 
```
CONFIG_LOG_MODE_DEFERRED=y
```
works with below build command
```
west build -b frdm_mcxn947/mcxn947/cpu0 --shield=dvp_20pin_ov7670 --shield=lcd_par_s035_8080 -- -DCONFIG_FPU=y
```
but if you add the full build setting with CONFIG_TEST
```
west build -b frdm_mcxn947/mcxn947/cpu0 --shield=dvp_20pin_ov7670 --shield=lcd_par_s035_8080 -- -DCONFIG_FPU=y -DCONFIG_TEST=y
```
it hangs with arm_usage fault.
and I enlarge the stack with below
```
+CONFIG_TEST_EXTRA_STACK_SIZE=1024
+CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
+CONFIG_MAIN_STACK_SIZE=2048
+CONFIG_SHELL_STACK_SIZE=2048
+CONFIG_IDLE_STACK_SIZE=640
+CONFIG_ISR_STACK_SIZE=40960
``
I get below 

![Image](https://github.com/user-attachments/assets/820f5bb4-f13b-4dd5-bda4-4f024e8d7570)

the full log as below
```
Program received signal SIGSTOP, Stopped (signal).
z_arm_usage_fault () at /home/ubuntu/zephyrproject/zephyr/arch/arm/core/cortex_m/fault_s.S:80
80              mrs r0, MSP
(gdb) bt
#0  z_arm_usage_fault () at /home/ubuntu/zephyrproject/zephyr/arch/arm/core/cortex_m/fault_s.S:80
#1  <signal handler called>
#2  0x1000b83a in nxp_video_sdma_callback (dev=<optimized out>, user_data=0x30000000 <sdma_data_0>, channel=<optimized out>, status=<optimized out>) at /home/ubuntu/zephyrproject/zephyr/drivers/video/video_mcux_smartdma.c:89
#3  0x100085c6 in _isr_wrapper () at /home/ubuntu/zephyrproject/zephyr/arch/arm/core/cortex_m/isr_wrapper.c:80
#4  <signal handler called>
#5  0x1000096a in __divdf3 ()
#6  0x10001790 in powl ()
#7  0x10001000 in rgb888_to_lab (r=<optimized out>, g=<optimized out>, b=<optimized out>) at /home/ubuntu/zephyrproject/zephyr/samples/drivers/video/capture/src/check_test_pattern.h:45
#8  0x100029b0 in rgb565_to_lab (color=<optimized out>) at /home/ubuntu/zephyrproject/zephyr/samples/drivers/video/capture/src/check_test_pattern.h:87
#9  is_colorbar_ok (fmt=..., buf=0x1ea002 '\377' <repeats 200 times>...) at /home/ubuntu/zephyrproject/zephyr/samples/drivers/video/capture/src/check_test_pattern.h:135
#10 main () at /home/ubuntu/zephyrproject/zephyr/samples/drivers/video/capture/src/main.c:272
(gdb) info register
r0             0x1                 1
r1             0x0                 0
r2             0x0                 0
r3             0x3c                60
r4             0x30000000          805306368
r5             0x2d6ccff           47631615
r6             0x29047000          688156672
r7             0x33333333          858993459
r8             0x27c6b6f8          667334392
r9             0x3fc06bb8          1069575096
r10            0xffffffff          4294967295
r11            0x40033333          1073951539
r12            0x1000b7bd          268482493
sp             0x30021050          0x30021050 <z_interrupt_stacks+40904>
lr             0xfffffff1          0xfffffff1
pc             0x10008248          0x10008248 <z_arm_usage_fault>
xpsr           0x9000005           150994949
fpscr          0x0                 0
msp            0x30021050          805441616
psp            0x30021758          805443416
cycles         0x30d8b75           51219317
msp_ns         0x0                 0
psp_ns         0xfffffffc          4294967292
msplim_ns      0x0                 0
psplim_ns      0x0                 0
control_ns     0x0                 0
faultmask_ns   0x0                 0
basepri_ns     0x0                 0
primask_ns     0x0                 0
msp_s          0x30021050          805441616
psp_s          0x30021758          805443416
msplim_s       0x30017088          805400712
psplim_s       0x30021308          805442312
control_s      0x0                 0
faultmask_s    0x0                 0
basepri_s      0x0                 0
primask_s      0x0                 0
fpccr          0xc00004fd          3221226749
fpcar          0x30021778          805443448
fpdscr         0x0                 0
mvfr0          0x10110021          269549601
mvfr1          0x11000011          285212689
```

and the phenomena is the screen freezes and console can not operate either.

closed 2025-07-03 by danieldegrasse