zephyrproject-rtos/zephyr #92343: Track rgb565 / bgr565 interchange on boards and displays

josuah, 2025-06-28:
### Summary

After https://github.com/zephyrproject-rtos/zephyr/pull/79996 is merged, some displays will be fixed, others will be broken and require drivers or devicetree modification.

This issue is there to track the situation of affected drivers, boards, displays and samples.

### Describe the solution you'd like

Check which driver effectively needs a swap after the constant got fixed:

| Driver | Type | RGB565 | Status |
|---|---|---|---|
| dummy | Emulated | No | - |
| **gc9x01x** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_gc9x01x.c#L575) | ? |
| **hx8394** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/boards/shields/rk055hdmipi4ma0/rk055hdmipi4ma0.overlay#L87) | ? |
| **ili9806e_dsi** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_ili9806e_dsi.c#L291) | ? |
| **ili9xxx** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_ili9xxx.c#L394-L395) | ? |
| intel_multibootfb | Arch | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_intel_multibootfb.c#L58) | - |
| ist3931 | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_ist3931.c#L215) | - |
| led_strip_matrix | RGB | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_led_strip_matrix.c#L201) | - |
| ls0xx | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/ls0xx.c#L212) | - |
| max7219 | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_max7219.c#L251) | - |
| **mcux_dcnano_lcdif** | I/F | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_mcux_dcnano_lcdif.c#L136-L137) | ? |
| **mcux_elcdif** | I/F | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_mcux_elcdif.c#L33-L34) | ? |
| microbit | LED | No | - | 
| nrf_led_matrix | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_nrf_led_matrix.c#L221) | - |
| **nt35510** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_nt35510.c#L278) | ? |
| **otm8009a** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_otm8009a.c#L460) | ? |
| **renesas_lcdc** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_renesas_lcdc.c#L128-L129) | ? |
| **renesas_ra** | I/F | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_renesas_ra.c#L212) | ? |
| **rm67162** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_rm67162.c#L479-L480) | ? |
| **rm68200** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/boards/shields/rk055hdmipi4ma0/rk055hdmipi4ma0.overlay#L87) | ? |
| **sdl** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_sdl.c#L686-L692) | ? |
| sh1122 | L_8 | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_sh1122.c#L337) | - |
| ssd1306 | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/ssd1306.c#L378) | - |
| ssd1320 | L_8 | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_ssd1320.c#L372) | - |
| ssd1322 | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/ssd1322.c#L222) | - |
| ssd1327 | L_8 | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/ssd1327.c) | - |
| **ssd1331** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_ssd1331.c#L256) | ? |
| **ssd1351** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_ssd1351.c#L237) | ? |
| ssd1363 | L_8 | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_ssd1363.c#L350) | - |
| ssd16xx | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/ssd16xx.c#L565) | - |
| st75256 | L_8 | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_st75256.c#L324) | - |
| st7567 | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_st7567.c#L281) | - |
| **st7701** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/display/display_st7701.c#L236) | ? |
| **st7735r** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_st7735r.c#L264-L270) | ? |
| st7789v | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_st7789v.c#L225-L230) | ? |
| **st7796s** | RGB | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_st7796s.c#L108-L117) | ? |
| **stm32_ltdc** | I/F | [Yes](https://github.com/zephyrproject-rtos/zephyr/blob/ad4c3e3e9afe1385ab80adf3fa2d3309fad68c60/drivers/display/display_stm32_ltdc.c#L159-L161) | ? |
| uc81xx | Mono | [No](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/display/uc81xx.c#L475) | ? |

Then review the situation of samples to make sure there is no extra swap involved:

- The [samples/subsys/display](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/display) collection
- The [samples/drivers/display](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/display) collection
- The [samples/modules/lvgl](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/modules/lvgl) collection
- The [samples/drivers/video](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/video) collection
- Others?

Then review the situation of boards to make sure a swap did not happen on boards

- List to be generated with `grep -r 565`?

Then prepare a migration guide and release note with indications of what downstream users might need to do.

### Alternatives

Alternatives to keep maintaining the swap as a long-term solution were rejected on #79996.

Let users report problems, but better catch as much as possible upstream.

### Additional Context

Some drivers implemented workarounds to apply the reversal.

Display interface drivers might need to be checked to avoid double-inversion.

ngphibang, 2025-06-29:
Thanks @josuah for creating this tracking issue. I made some modifications in the issue description to clarify more on things:

- The PR fixes only the display sample (and 2 NXP display drivers) so it will not affect any dowstream application. The worse case is there will be some displays that will not work anymore **_with the display sample_** only.
- Looking at the lists of displays that are involved now, there are not many. Just about a half of them are involved RGB565 and 2 displays are already fixed by the PR, 1 (ST LTDC) is already working correctly with the sample after the fix. With @JarmouniA help, I think we can identify more (ST) displays that will not need any modification after the merge. Thank you.

avolmat-st, 2025-06-30:
Maybe we can add a new column to indicate which kind of platform those display are being used ?

zhang-wenchao, 2025-09-15:
`west build -b stm32h747i_disco/stm32h747xx/m7 --shield st_b_lcd40_dsi1_mb1166 samples/drivers/display`

The color and position of the latest version are abnormal. I am not sure if this is related.

ngphibang, 2025-09-15:
> west build -b stm32h747i_disco/stm32h747xx/m7 --shield st_b_lcd40_dsi1_mb1166 samples/drivers/display
> 
> The color and position of the latest version are abnormal. I am not sure if this is related.

Looking at the shield `st_b_lcd40_dsi1_mb1166` it seems the format is RGB888 which is unrelated with this change (RGB565/BGR565)

zhang-wenchao, 2025-09-15:
> > west build -b stm32h747i_disco/stm32h747xx/m7 --shield st_b_lcd40_dsi1_mb1166 samples/drivers/display
> > The color and position of the latest version are abnormal. I am not sure if this is related.
> 
> Looking at your photo and the shield `st_b_lcd40_dsi1_mb1166` it seems the format is RGB888 which is unrelated with this change (RGB565/BGR565)

otm8009a | RGB | Yes | STM32H747I-Disco

I see this in the table.

closed 2026-09-14 by github-actions[bot]