zephyrproject-rtos/zephyr #92354: drivers: video: video_stm32_dcmi: Detect and recover from: HAL_DCMI_ErrorCallback

KurtE, 2025-06-28:
### Summary

With several of the different sketches that do things like display a camera image to a display of some form.  Like Arduino Gigia Display shield, or some TFT display like an ST7796 or ILI9341, As I mentioned in the issue: #92352 I will receive several messages like:
```
[00:00:12.307,000] <dbg> video_stm32_dcmi: HAL_DCMI_FrameEventCallback: Failed to get buffer from fifo
```
Which in that issue I would hope to reuse the oldest frame...

But then I often times then get the message:
`[00:00:14.851,000] <wrn> video_stm32_dcmi: HAL_DCMI_ErrorCallback
`
And the camera completely stops at that point

### Describe the solution you'd like

I wish that the callback could correct the issue and recover.

### Alternatives

Or wish there was a way for the client code to detect this and call something to recover.

Detecting this might simply be, that I would expect an image to be ready by my timeout... If not then call the recovery...

### Additional Context

Most of my testing is done using the ArduinoCore-zephyr setup.  Currently running the latest github code.
The west update from their points us to a zephyr snaphshot from about a month ago.

josuah, 2025-06-29:
> `[00:00:12.307,000] <dbg> video_stm32_dcmi: HAL_DCMI_FrameEventCallback: Failed to get buffer from fifo`

This seems like something that could be avoided by continuously enqueueing frames to make sure the driver never starves.

> `[00:00:14.851,000] <wrn> video_stm32_dcmi: HAL_DCMI_ErrorCallback`

This looks like a bug on the DCMI HAL or the way it is used in the Zephyr driver?

If it turns out that it is an expected condition (i.e. DCMI was starved for too long), then maybe this needs some preventive measure to be implemented in `video_stm32_dcmi.c` so that the video API can be used without having to i.e. restart a driver that fails.

Also linking the Arduino-side issue where `HAL_DCMI_ErrorCallback` is also triggered:

- https://github.com/arduino/ArduinoCore-zephyr/issues/78#issuecomment-2708493630

Thank you for all the communication!

KurtE, 2025-06-29:
Thanks @josuah,

I totally agree that it would be better to avoid having to restart the stream!

Current setup:

[image]
Current sketch:
```
#include <ST77XX_zephyr.h>
#include "camera.h"

Camera cam;
#include <SPI.h>
#include <ST77XX_zephyr.h>
#include <ST77XX_zephyr_font_Arial.h>

#ifdef ARDUINO_PORTENTA_H7 
#define TFT_DC   4 
#define TFT_CS   2  
#define TFT_RST  3
#define TFT_SPI  SPI

#else
#define TFT_CS   10  // CS & DC can use pins 2, 6, 9, 10, 15, 20, 21, 22, 23
#define TFT_DC    8  //  but certain pairs must NOT be used: 2+10, 6+9, 20+23, 21+22
#define TFT_RST   9  // RST can use any pin
#define TFT_SPI  SPI1

#endif

#define CAMERA_WIDTH 480 //320
#define CAMERA_HEIGHT 320 // 240
// Use one or the other
//ST7735_zephyr tft = ST7735_zephyr(&SPI, TFT_CS, TFT_DC, TFT_RST);

// For 1.54" TFT with ST7789
//ST7789_zephyr tft = ST7789_zephyr(&SPI, TFT_CS, TFT_DC, TFT_RST);

// For 3.5" or 4.0" TFT with ST7796
ST7796_zephyr tft = ST7796_zephyr(&TFT_SPI, TFT_CS, TFT_DC, TFT_RST);

void fatal_error(const char *msg) {
  Serial.println(msg);
  pinMode(LED_BUILTIN, OUTPUT);
  while (1) {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(100);
    digitalWrite(LED_BUILTIN, LOW);
    delay(100);
  }
}

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  while (!Serial && millis() < 5000) {}

  Serial.println("\n*** start display camera image on ST77XX ***");
  // Use this initializer if you're using a 1.8" TFT 128x160 displays
  //tft.initR(INITR_BLACKTAB);

  // Or use this initializer (uncomment) if you're using a 1.44" TFT (128x128)
  //tft.initR(INITR_144GREENTAB);

  // Or use this initializer (uncomment) if you're using a .96" TFT(160x80)
  //tft.initR(INITR_MINI160x80);

  // Or use this initializer (uncomment) for Some 1.44" displays use different memory offsets
  // Try it if yours is not working properly
[... 94 lines cut ...]
InitVariant called
calling start_static_threads
static threads start_static_threads
[00:00:06.811,000] <dbg> video_gc2145: gc2145_set_resolution: set_resolution(480 320)
[00:00:06.821,000] <dbg> video_gc2145: gc2145_set_resolution: set resolution(480 320): ratio: 3 3
[00:00:14.815,000] <err> dma_stm32_v1: FiFo error.
[00:00:14.822,000] <inf> dma_stm32_v1: tc: 0, ht: 1, te: 0, dme: 0, fe: 1
[00:00:14.832,000] <err> dma_stm32: Unexpected irq happened.
[00:00:14.840,000] <err> video_stm32_dcmi: DMA callback error with channel 0.
[00:00:14.850,000] <err> dma_stm32_v1: FiFo error.
[00:00:14.858,000] <inf> dma_stm32_v1: tc: 1, ht: 1, te: 0, dme: 0, fe: 1
[00:00:14.867,000] <err> dma_stm32: Unexpected irq happened.
[00:00:14.876,000] <err> video_stm32_dcmi: DMA callback error with channel 0.
[00:00:14.886,000] <err> dma_stm32_v1: FiFo error.
[00:00:14.893,000] <inf> dma_stm32_v1: tc: 1, ht: 1, te: 0, dme: 0, fe: 1
[00:00:14.903,000] <err> dma_stm32: Unexpected irq happened.
[00:00:14.911,000] <err> video_stm32_dcmi: DMA callback error with channel 0.
[00:00:15.229,000] <wrn> video_stm32_dcmi: HAL_DCMI_ErrorCallback
```

Note: this is probably a torture test for SDRAM.  That is there are two DMA type operations going on,
one reading from the camera to SDRAM, and the other writing from SDRAM to display, plus each time
the camera reads a frame in, it then copies the 307200 bytes from on SDRAM buffer to another.

Will keep playing and trying stuff out.

avolmat-st, 2025-06-30:
> Note: this is probably a torture test for SDRAM. That is there are two DMA type operations going on, one reading from the camera to SDRAM, and the other writing from SDRAM to display, plus each time [...]

To check in this area, could you first try to only capture frames without display, with for example sleep between dequeue and enqueue to "simulate" the delay introduced by display and see if this still affect the capture ?

KurtE, 2025-07-01:
Sorry, I got distracted yesterday...  Other things like yard work...

I believe @mjs513 and I have tried a few different frequencies, like 8, 10, 12Mh, 
all acting slightly differently. With some we maybe don't receive any valid images, with some maybe one or two, the last
run I mentioned I received about 20, before it died.  It also depended on how large of an image I was retrieving. Best luck with
320x240 but would be a shame to restrict a 2MBit or 5MBit down to that resolution.

My gut also tells me that a lot of these issues have to do with SDRAM usage.  So wondering if at least for these
lower resolutions: 320x240*2=153600 bytes, if as in experiment have the camera use a buffer that is in main memory instead
of SDRAM, and see if that makes a difference.

The one distraction I am doing now, is to try to build a similar sketch/sample, that works similarly.... Sort of a cross to
the capture_to_lvgi sample, but instead of lvgi use me own ILI9341 display using SPI and my own code on Portenta H7...  Yesterday 
was able to get the display to work... (I know probably can use LVGI, using the Adafruit....)

Now trying to integrate in the camera stuff,  Currently have it compiling but link error: undefined reference to `__device_dts_ord_219'
Never quite sure how to deduce these, but...

Note: I am mainly doing this for two reasons.  Learning and as the Arduino snapshot of Zephyr does not currently have
the PR pulled merged in that added the new video APIS.  But potentially I could use a branch that @pillo79 has:
https://github.com/pillo79/zephyr/tree/zephyr-arduino-20250627 
To try it out on the Arduino side...

Now back to link error.

KurtE, 2025-07-01:
> Is caused by an incorrect nodelabel, device not enabled in DT, wrong node identifier...

Thanks,

I have been looking at the troubleshooting page:
https://docs.zephyrproject.org/latest/build/dts/troubleshooting.html

From the devicetree_generated.h I see:
` *   219 /soc/spi@40003800
`
Which from reference file address is: SPI2
```
&spi2 {
	status = "okay";
	pinctrl-0 = <&spi2_sck_pi1
		     &spi2_miso_pc2 &spi2_mosi_pc3>;
	pinctrl-names = "default";
	/* added for sketch */
	ili9341_spi_dev: ili9341_spi_dev@0 {
		status = "okay";
		compatible = "vnd,spi-device";
		reg = <0>;
		spi-max-frequency = <30000000>;
	};

};
```
So it is defined with status OK... I think now it is a .conf file issue, working through it.  I moved I few things around
between the main prj.conf file to the boards specific and I probably messed something up...

josuah, 2025-07-03:
I hit "close with comment" but meant to keep it open until solved ofc.

KurtE, 2025-07-03:
Quick update:

As a real hack, I setup to NOT use SDRAM for the camera buffer...
```
	/* Alloc video buffers and enqueue for capture */
	printk("Initialze video buffer list (%u)\n",  ARRAY_SIZE(buffers));
	for (i = 0; i < ARRAY_SIZE(buffers); i++) {
		buffers[i] = video_buffer_aligned_alloc(bsize, CONFIG_VIDEO_BUFFER_POOL_ALIGN,
							K_FOREVER);
		#ifdef CAMERA_USE_FIXED_BUFFER
		if ((i == 0) &&  (buffers[i] != NULL)) {
			// REAL hack change the buffer over to our fixed buffer...
			buffers[i]->buffer = (uint8_t *)frame_buffer;
		}
		#endif

		printk("  %d:%x\n", i, (uint32_t)buffers[i]);
		if (buffers[i] == NULL) {
			printk("ERROR: Unable to alloc video buffer\n") ;
			return 0;
		}
		buffers[i]->type = type;
		video_enqueue(video_dev, buffers[i]);
	}
```

Code is still outputting frames to the display...

Currently it has displayed: > 2750 frames without getting the error...

KurtE, 2025-07-03:
Thanks,

But I know for example that on the GIGA, with the GIGA display shield, they have sketch that can read in using 320x240 and
display it at 640x480 on the display, this is using the MBED version of Arduino with the arducam_dvp library using sketch
mentioned up at: 
https://docs.arduino.cc/tutorials/giga-display-shield/camera-tutorial/?_gl=1*1z0l9o0*_up*MQ..*_ga*MTI0Mjk4ODkyNy4xNzUxNTc4MTA5*_ga_NEXN8H46L5*czE3NTE1NzgxMDgkbzEkZzEkdDE3NTE1NzgxMTckajUxJGwwJGgxNzExMjg2MzEx#full-sketch

But back then if I tried the logical update the camera to output 640x480 I think I ran into similar issues.

KurtE, 2025-07-04:
> It will help to have more image sensor ported and more features in the current sensors, to compare across them.

That would be great! 

Some of the other sensors that a few of us picked up when we were adapting code similar to Arducam_DVP library by
@iabdalkader and other over to the Teensy boards (https://github.com/mjs513/Teensy_Camera)

>Camera class to support Omnivision oV2640, OV5640 OV7670 and OV7675, Himax HM01B0 and HM0360, and GalaxyCore GC2145

Which I believe I have at least one of each of these which have the pinout that matches the GIGA/H7 MID... 
I know we have had similar issues with OV7675 with erroring out. 

But maybe first a diversion to see if I can maybe get the GC2145 to see/handle setting for window...

avolmat-st, 2025-07-05:
I think I am pretty late on this topic and I couldn't follow all the discussions, apologies for that.
In one of those threads talking about HAL_DCMI_ErrorCallback I think I wrote once that it could be due to the sensor sending data too rapidly. There are two potentially limiting factors on the receiver side:

  1. the DCMI interface itself. The kernel clock of the DCMI must be at least 2.5 time faster than the pixel clock received by the sensor. So it is important to see what is the kernel clock frequency which clock the DCMI and dividing this by 2.5 will give the maximum allowed pixel clock frequency it can handle. (in the CSI context we have been discussing about having helpers to tackle those HW capabilities via usage of the link-frequencies from video-interface.yaml but that's not really in place yet)

 2. the DMA capabilities itself (and related to the downstream memory access). The DCMI IP receive data and store them into its internal FIFO. Then the DMA goes to read those data and store them somewhere else. So the performance of the DMA and memory access can also be a bottleneck.

To cop with those, it is usually possible to limit the sensor pixel clock rate.

As a matter of fact, the STM32L4 (stm32l4r9i_disco) is also facing this issue (HAL_DCMI_ErrorCallback) when used with the st_b_cam_omv shield. I haven't got time to do that cleanly and post a PR for that, however just for explanatiion / demo it is possible to make it work (and use UVC sample) by simply forcing the OV5640 PLL frequency lower:

diff --git a/drivers/video/ov5640.c b/drivers/video/ov5640.c
index 664d99b9543..3997b9a4c1e 100644
--- a/drivers/video/ov5640.c
+++ b/drivers/video/ov5640.c
@@ -428,7 +428,7 @@ static const struct ov5640_reg init_params_dvp[] = {
        {0x3814, 0x31},
        {0x3815, 0x31},
        {0x3034, 0x1a},
-       {0x3035, 0x11},
+       {0x3035, 0x21},
        {0x3036, 0x64},
        {0x3037, 0x13},
        {0x3038, 0x00},

That done, OV5640 streaming into STM32L4R9I_DISCO is working just fine.

KurtE, 2025-07-05:
Thanks @avolmat-st 

Wondering what might need to change for GC2145.
Looking at their init sequence I do see differences in the init table from the code here in zephyr, vs arducam_dvp and Teensy.

For example:  the first art of the default_regs[] on zephyr:
```
static const struct video_reg8 default_regs[] = {
  {GC2145_REG_RESET, 0xf0},
  {GC2145_REG_RESET, 0xf0},
  {GC2145_REG_RESET, 0xf0},
  {0xfc, 0x06},
  {0xf6, 0x00},
  {0xf7, 0x1d},
  {0xf8, 0x85},

  {0xfa, 0x00},
  {0xf9, 0xfe},
  
  {0xf2, 0x00},

  /* ISP settings */
```
On Teensy 
```
static const uint8_t default_regs[][2] = {
    {0xfe, 0xf0},
    {0xfe, 0xf0},
    {0xfe, 0xf0},

    {0xfc, 0x06},
    {0xf6, 0x00},
    {0xf7, 0x1d}, // 37 //17 //37 //1d//05
    $$${0xf8, 0x83}, // 87 //83 //82**
    {0xfa, 0x00},
    {0xf9, 0xfe}, // ff
    $$${0xfd, 0x00},
    $$$ {0xc2, 0x00},
    {0xf2, 0x0f},
    //////////////////////////////////////////////////////
    ////////////////////  Analog & Cisctl ////////////////
    //////////////////////////////////////////////////////
```
$$$ lines are different.  There are are other differences.  
Probably go through them all to see if anything pops out...

avolmat-st, 2025-07-06:
Hi @KurtE,

a copy of the GC2145 datasheet can be fairly easily found on the net.

I'd try to first lower down the PLL configuration, say by increasing a divider.
so that mainly registers 0xf7 / 0xf8 / 0xfa.
Another option that could work would be to increase the vertical and/or horizontal blanking in order to give more time to the DMA to transfer the data out of the FIFO if that is the blocking issue, that is registers 0x05 / 0x06 / 0x07 / 0x08 on register page 0.

KurtE, 2025-07-07:
Quick update:
> Keep wondering how the Camera examples work on MBED version of GIGA at least at the 320x240 size. I should probably
> double check that it works with Portenta H7 setup as well.

I am trying it on H7 and the currently released arducam_dvp library does not work with this setup, as several of the pin/port
numbers are different than whatever they originally tested with. Earlier I used the breakout board, and simply wired to match
the code... 

Created issue: https://github.com/ArduCAM/Arducam_dvp/issues/8
I will probably issue a PR for to try to make it at least somewhat work.
However I don't expect anything will happen with it as currently they have
issues (8 open, 0 closed) PR(1 open, 0 closed) So I believe this library has been mostly abandoned.

KurtE, 2025-07-08:
Another update: 
I now have a version that runs on MBED, in particular I have it displaying currently on an ST7796 display.
(Display code is slow, did first pass port the zephyr version I did to MBED). 
I made changes to Arducam library,   My version is up at: https://github.com/KurtE/Arducam_dvp/tree/experiments_debug
My sketch and ST7706 code are up in my github project: https://github.com/KurtE/Arduino_GIGA-stuff
I am currently configured to use SDRam for the frame buffer.  So far it has run for several minutes without failing.

Note: There are a lot of differences on how this Camera code works on MBED versus on Zephyr. 
On Zephyr you have multiple buffers defined, one is handed to the Camera, that is constantly is pounding the SDRAM with
each frame.  It then process an interrupt when each frame completes, and it then tries to grab a frame buffer from the free
list, where it memcpy the data from the camera.   The Arduino wrapper in the ArduinoCore-zephyr or my sketch then
needs to swap the bytes for each pixel and then the code writes the data out of SPI to the TFT or over the xxx to write
to the GigaDisplayShield... So lots of code hitting the SDRAM maybe at the same time...

With the MBED:  when I call:   
`int err = cam.grabFrame(fb);
`
The code does:
```
    if (HAL_DCMI_Start_DMA(&hdcmi, DCMI_MODE_SNAPSHOT,
                (uint32_t) framebuffer, framesize / 4) != HAL_OK) {
        if (_debug) {
            _debug->println("HAL_DCMI_Start_DMA FAILED!");
        }
    }

    // Wait until camera frame is ready.
    for (uint32_t start = millis(); (hdcmi.Instance->CR & DCMI_CR_CAPTURE);) {
```
So it starts the camera in SNAPSHOT mode (one frame), and then waits up to the timeout in a hard loop
until the DCMI_CR_CAPTURE bit is cleared, when the hardware detects the end of the frame and stops...

Then the application or camera library if necessary swaps the bytes and then the application does
whatever it needs to do, like send the bytes out to the display.  

So with the read an image, write to display code, only one thing is touching the SDRAM.  And by default
if your code does not set the fb memory like: fb.setBuffer(fb_buf);
The camera code will allocate the memory using malloc (not SDRAM).

Wondering if the Video/Camera code on Zephyr should have the ability for SnapShot mode?

josuah, 2025-07-08:
Avoiding the `memcpy()` operation would make most sense, and maybe support for SNAPSHOT mode is the way.
The `memcpy()` was convenient as a way to get the driver going, but for the way the video APIs work, swapping the buffer would make more sense than copying it...

I will follow-up about the `bus-type` on https://github.com/arduino/ArduinoCore-zephyr/issues/152, it is possible that it is one reason for which the OV5640 stays silent.

mjs513, 2025-07-08:
> I will follow-up about the bus-type on https://github.com/arduino/ArduinoCore-zephyr/issues/152, it is possible that it is one reason for which the OV5640 stays silent.

Had a feeling so I had also tested with `bus-type = <5>` which is VIDEO_BUS_TYPE_PARALLEL. and still didn't work.  No clue why not receiving frames.  More investigation

josuah, 2025-07-08:
> Can you point to camera Module for the OV5640 you are using - I am testing with the Adafruit OV5640.

I did not get a chance to try the OV5640 but have the Arduino Mega at hand and the Adafruit OV5640.
I'd be glad to post a picture on https://github.com/zephyrproject-rtos/zephyr/discussions/90428 as soon as I get a chance to test it!

josuah, 2025-07-08:
One frequent problem of I2C communication remaining silent with image sensors is the XCLK not being sent... I suppose "CLOCK" on the logic analyzer trace is connected to the XCLK pin of the image sensor?

It is a bit tricky as this pin is not configured in the OV5640 driver but in a separate `board.c` that simply makes function call to start the configured clock (i.e. via a PWM pin) and log the effective frequency...

https://github.com/zephyrproject-rtos/zephyr/blob/808ee1713da4cb7aa6c41e7c4cd18269fd63f7f3/boards/arduino/nicla_vision/arduino_nicla_vision_stm32h747xx_m7.dts#L269-L278

https://github.com/zephyrproject-rtos/zephyr/blob/808ee1713da4cb7aa6c41e7c4cd18269fd63f7f3/boards/arduino/nicla_vision/camera_ext_clock.c#L14-L42

KurtE, 2025-07-10:
Quick side note: for now I think I will wait a while before I play again with the OV5640 camera.

This morning, I tried instead using an OV2640, with the same sketch to read from camera and write to an ILI9341 display.
Just needed to update the overlay and config files slightly:
Under &i2c3 {
```
	ov2640: ov2640@30 {
		compatible = "ovti,ov2640";
		reg = <0x30>;
		status = "okay";

		reset-gpios = <&gpioe 3 GPIO_ACTIVE_LOW>;
		//powerdown-gpios = <&gpiod 5 GPIO_ACTIVE_HIGH>;

		port {
			ov2640_ep_out: endpoint {
				remote-endpoint-label = "dcmi_ep_in";
       			bus-type = <5>;
			};
		};
	};
```
under dcmi... port
```
			//remote-endpoint-label = "ov5640_ep_out";
			remote-endpoint-label = "ov2640_ep_out";

```

[image]

Note: I believe I purchased this waveshare camera from Robotshop.com:
https://www.robotshop.com/products/ov2640-camera-board

KurtE, 2025-07-10:
Quick update:
I hacked up my ov5640_to_ili9351 test zephyr sketch (project...) to try to build easier for OV5640 or 2640 without a lot of 
editing back and forth.  So I put both configurations in the overlay file.

Currently hack at top of overlay file:
```
#define CONFIG_VIDEO_OV2640 

/* This is copy of the Arduino setup */

```
Then later have things like:
```
&i2c3 {
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;

#ifdef CONFIG_VIDEO_OV5640 
	ov5640: ov5640@3c {
		compatible = "ovti,ov5640";
		reg = <0x3c>;
		status = "okay";

		reset-gpios = <&gpioe 3 GPIO_ACTIVE_LOW>;
		powerdown-gpios = <&gpiod 5 GPIO_ACTIVE_HIGH>;

		port {
			ov5640_ep_out: endpoint {
				remote-endpoint-label = "dcmi_ep_in";
       			bus-type = <5>;
			};
		};
	};
#endif
#ifdef CONFIG_VIDEO_OV2640 
	ov2640: ov2640@30 {
		compatible = "ovti,ov2640";
		reg = <0x30>;
		status = "okay";

		reset-gpios = <&gpioe 3 GPIO_ACTIVE_LOW>;
		//powerdown-gpios = <&gpiod 5 GPIO_ACTIVE_HIGH>;

		port {
			ov2640_ep_out: endpoint {
				remote-endpoint-label = "dcmi_ep_in";
       			bus-type = <5>;
			};
		};
	};
#endif
```
Was hoping that it would be able to know if either of those two were defined within .conf file, but that failed...
Hopefully find a cleaner way.

I tried again with 0v5640 this time asking for 24mhz, still fails with the I2C stuff at start.

Note in both cases, I hacked in some support to print out the camera registers.
You can pause the sketch by typing something in the Serial monitor.
Then when it shows paused, type in: r<cr>
It was a quick and dirty conversion of the teensy code.  For OV2640 it is showing:
```
*** Paused ***

*** Camera Registers ***
(0)  QS(44): 12(c)
(0)  HSIZE(51): 144(90)
(0)  VSIZE(52): 44(2c)
(0)  XOFFL(53): 0(0)
(0)  YOFFL(54): 0(0)
(0)  VHYX(55): 136(88)
(0)  DPRP(56): 0(0)
(0)  TEST(57): 0(0)
(0)  ZMOW(5a): 80(50)
(0)  ZMOH(5b): 60(3c)
(0)  ZMHH(5c): 0(0)
(0)  BPADDR(7c): 11(b)
(0)  BPDATA(7d): 0(0)
(0)  SIZEL(8c): 0(0)
(0)  HSIZE8(c0): 200(c8)
(0)  VSIZE8(c1): 150(96)
(0)  CTRL1(c3): 255(ff)
(0)  MS_SP(f0): 4(4)
(0)  SS_ID(f7): 96(60)
(0)  SS_CTRL(f7): 96(60)
(0)  MC_AL(fa): 0(0)
(0)  MC_AH(fb): 0(0)
(0)  MC_D(fc): 0(0)
(0)  P_CMD(fd): 0(0)
(0)  P_STATUS(fe): 79(4f)
(0)  CTRLI(50): 128(80)
(0)  CTRLI_LP_DP(80): 0(0)
(0)  CTRLI_ROUND(40): 51(33)
(0)  CTRL0(c2): 14(e)
(0)  CTRL2(86): 61(3d)
(0)  CTRL3(87): 208(d0)
(0)  R_DVP_SP(d3): 132(84)
(0)  R_DVP_SP_AUTO_MODE(80): 0(0)
(0)  R_BYPASS(5): 0(0)
(0)  R_BYPASS_DSP_EN(0): 0(0)
(0)  R_BYPASS_DSP_BYPAS(1): 0(0)
(0)  IMAGE_MODE(da): 8(8)
(0)  RESET(e0): 0(0)
(0)  MC_BIST(f9): 192(c0)
(0)  BANK_SEL(ff): 0(0)
(0)  BANK_SEL_DSP(0): 0(0)
(0)  BANK_SEL_SENSOR(1): 0(0)
(1)  GAIN(0): 63(3f)
(1)  COM1(3): 15(f)
(1)  REG_PID(a): 38(26)
(1)  REG_VER(b): 66(42)
(1)  COM4(d): 183(b7)
(1)  AEC(10): 55(37)
(1)  CLKRC(11): 135(87)
(1)  CLKRC_DOUBLE(82): 0(0)
(1)  CLKRC_DIVIDER_MASK(3f): 1(1)
(1)  COM10(15): 0(0)
(1)  HSTART(17): 17(11)
(1)  HSTOP(18): 117(75)
(1)  VSTART(19): 1(1)
(1)  VSTOP(1a): 151(97)
(1)  MIDH(1c): 127(7f)
(1)  MIDL(1d): 162(a2)
(1)  AEW(24): 64(40)
(1)  AEB(25): 56(38)
(1)  REG2A(2a): 0(0)
(1)  FRARL(2b): 0(0)
(1)  ADDVSL(2d): 0(0)
(1)  ADDVSH(2e): 0(0)
(1)  YAVG(2f): 30(1e)
(1)  HSDY(30): 8(8)
(1)  HEDY(31): 48(30)
(1)  REG32(32): 54(36)
(1)  ARCOM2(34): 160(a0)
(1)  REG45(45): 1(1)
(1)  FLL(46): 0(0)
(1)  FLH(47): 0(0)
(1)  COM19(48): 0(0)
(1)  ZOOMS(49): 0(0)
(1)  COM22(4b): 32(20)
(1)  COM25(4e): 0(0)
(1)  BD50(4f): 187(bb)
(1)  BD60(50): 156(9c)
(1)  REG5D(5d): 85(55)
(1)  REG5E(5e): 125(7d)
(1)  REG5F(5f): 125(7d)
(1)  REG60(60): 85(55)
(1)  HISTO_LOW(61): 112(70)
(1)  HISTO_HIGH(62): 128(80)
(1)  REG04(4): 43(2b)
(1)  REG08(8): 64(40)
(1)  COM2(9): 2(2)
(1)  COM2_STDBY(10): 55(37)
(1)  COM3(c): 58(3a)
(1)  COM7(12): 0(0)
(1)  COM8(13): 229(e5)
(1)  COM8_DEFAULT(c0): 0(0)
(1)  COM8_BNDF_EN(20): 128(80)
(1)  COM8_AGC_EN(4): 43(2b)
(1)  COM8_AEC_EN(1): 0(0)
(1)  COM9(14): 72(48)
(1)  CTRL1_AWB(8): 64(40)
(1)  VV(26): 130(82)
(1)  REG32(32): 54(36)
```

CharlesDias, 2025-07-12:
Hi  @KurtE and @mjs513 

Could you check this? **"The DCMI_PIXCLK and AHB clocks must respect the minimum ratio AHB / DCMI_PIXCLK of 2.5."**

This information is provided on https://www.st.com/resource/en/application_note/an5020-introduction-to-digital-camera-interface-dcmi-for-stm32-mcus-stmicroelectronics.pdf

CharlesDias, 2025-07-12:
> I also tired the OV2640 at 10mhz on the H7 which worked no issues as [@KurtE](https://github.com/KurtE) mentioned. And at 10mhz it was slow - didn't try 24mhz. Right now just wondering what is working and what not working.

Hi @KurtE  and @mjs513 , depending on the sensor configuration, the PIXCLK may not be equal to XCLK. The sensor has an internal PLL that can multiply this frequency. Could you please measure the PIXCLK and XCLK?

KurtE, 2025-09-13:
Not sure this is really stale.  

There is WIP with my PR #93797 where I have some detection and some recovery working.  Just working through
the PR feedback and a few details, like I get the DMA to start again, but may not additional work to stop the stream and
restart it, as to try to get the DMA to start at the beginning of a frame

closed 2025-11-28 by github-actions[bot]