zephyrproject-rtos/zephyr #92397: Missing fallback to a base board.dts when using a board variant without specific dts file

flo-carrere, 2025-06-30:
### Describe the bug

According to the board porting guide (here referencing v3.7.0 but surely the same on latest Zephyr),
https://docs.zephyrproject.org/3.7.0/hardware/porting/board_porting.html

```
Only one .dts file will be used, and the most specific file which exists will be used.

This means that if both plank.dts and plank_soc1_foo.dts exist, then when building for plank / plank/soc1, then plank.dts is used. When building for plank//foo / plank/soc1/foo the plank_soc1_foo.dts is used.

This allows board maintainers to write a base devicetree file for the board or write specific devicetree files for a given board’s SoC or variant.
```

What I understood from that is that conversely if **only** `plank.dts` exists (base devicetree file for the board) then it will be used for **all** `plank`, `plan/soc1`, `plank//foo` or even `plan/soc1/foo` allowing effectively board maintainers to **inherit/share elegantly** a common device tree structure for **their board variants**, adjusting it for instance with a different defconfig or with an overlay only. 

But it is not the case and building for `plank//foo` in that case actually fallback to the `common/empty/stub.dts` unexpectedly

Here is the related code fron dts.cmake
https://github.com/zephyrproject-rtos/zephyr/blob/v3.7.0-rc3/cmake/modules/dts.cmake#L126-L144

```cmake
if(NOT DEFINED DTS_SOURCE)
  zephyr_build_string(board_string SHORT shortened_board_string
                      BOARD ${BOARD} BOARD_QUALIFIERS ${BOARD_QUALIFIERS}
  )
  if(EXISTS ${BOARD_DIR}/${shortened_board_string}.dts AND NOT BOARD_${BOARD}_SINGLE_SOC)
    message(FATAL_ERROR "Board ${ZFILE_BOARD} defines multiple SoCs.\nShortened file name "
            "(${shortened_board_string}.dts) not allowed, use '<board>_<soc>.dts' naming"
    )
  elseif(EXISTS ${BOARD_DIR}/${board_string}.dts AND EXISTS ${BOARD_DIR}/${shortened_board_string}.dts)
    message(FATAL_ERROR "Conflicting file names discovered. Cannot use both "
            "${board_string}.dts and ${shortened_board_string}.dts. "
            "Please choose one naming style, ${board_string}.dts is recommended."
    )
  elseif(EXISTS ${BOARD_DIR}/${board_string}.dts)
    set(DTS_SOURCE ${BOARD_DIR}/${board_string}.dts)                    <<<<<< plank_soc1_foo.dts is accepted
  elseif(EXISTS ${BOARD_DIR}/${shortened_board_string}.dts)
    set(DTS_SOURCE ${BOARD_DIR}/${shortened_board_string}.dts)  <<<<<< plank_foo.dts is accepted
  endif()
endif()
```

I guess a last fallback (the least specific) is missing? Or what is the reason for not accepting it?

```cmake
  elseif(EXISTS ${BOARD_DIR}/${BOARD}.dts)
    set(DTS_SOURCE ${BOARD_DIR}/${BOARD}.dts)  <<<<<< plank.dts should be accepted as well?
  endif()
```

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

flo-carrere, 2025-06-30:
Similarly, detecting a <board>_<qualifier>.overlay (or a .conf) such as `plank_foo.overlay` in this example is restricted because only the <board>_<qualifier>_<revision>.overlay pattern is be accepted.

This is really limiting when thinking that BOARD_REVISION is supposed to be optional but in reality strictly expected in that case, despite the overlay is correctly detected.

```cmake
// Below in the same dts.cmake

if(EXISTS ${DTS_SOURCE})
# We found a devicetree. Append all relevant dts overlays we can find...
zephyr_file(CONF_FILES ${BOARD_DIR} DTS DTS_SOURCE)

zephyr_file(
  CONF_FILES ${BOARD_DIR}
  DTS no_rev_suffix_dts_board_overlays      <<<<<   plank_foo.overlay is detected but does not contain a <revision> suffix
  BOARD ${BOARD}
  BOARD_QUALIFIERS ${BOARD_QUALIFIERS}
  )

  # ...but remove the ones that do not include the revision suffix
  list(REMOVE_ITEM DTS_SOURCE ${no_rev_suffix_dts_board_overlays})     <<<<<< Got removed from eligible dts list
else()
  # If we don't have a devicetree, provide an empty stub
  set(DTS_SOURCE ${ZEPHYR_BASE}/boards/common/stub.dts)
endif()
```

flo-carrere, 2025-07-02:
@henrikbrixandersen I agree with your additional suggestions/thoughts but my original request is really about opening for a base .dts, potentially complemented by .overlay detection for "variant with no-revision".

For the fallback to `plank.dts`, I can understand that it would be equivalent to **another reduction** of the `shortened_board_string` and though I find it acceptable, it may have side-effects I don't master. 

But for the .overlay, the detection is **already** there but restricted on purpose. Maybe allowing it would imply some ordering/merging problems?

Do you have a view on the reasons why it implemented this way?

Do you think it is worth if I open a PR to complement this issue because, as shown on the various snippet(s) the changes to get both is really a matter of few changes?

henrikbrixandersen, 2025-07-02:
> Do you have a view on the reasons why it is implemented/restricted this way?

Perhaps @tejlmand or @nordicjm recall?

nordicjm, 2025-07-03:
> Can we please stick to the topic of the .dts source not falling back to
${BOARD_DIR}/${BOARD}.dts?
> 
> Again, as per the documentation, we can expect this identification to happen (if it is strictly the most specific available) before of falling back to the empty_stub.dts which is meant to alert the developper.

As per comment on https://github.com/zephyrproject-rtos/zephyr/issues/90160 the documentation is wrong and needs updating and using empty_stub.dts is a completely valid thing, some qemu boards purposely do this

flo-carrere, 2025-07-03:
@nordicjm About `empty_stub.dts`, I am not asking for a replacement but a last **additional** fallback BEFORE it (in line with current doc). Of course that stub is useful.

Anyway, if we prefer "fixing the docs" (to warn about such limitations explicitly) rather than fixing the code to match it better, I am OK with that. 

Thanks @nordicjm @henrikbrixandersen for the explanations.