zephyrproject-rtos/zephyr #92413: Issue with thread suspension in zephyr 3.7.1

rahulprajapati-eaton, 2025-06-30:
### Describe the bug

Previously, I was using Zephyr 3.6.0, where I was able to successfully suspend cooperative threads. However, after upgrading to Zephyr 3.7.1, the same code no longer works as expected.

In my implementation, I created 10 threads with high priority (a mix of preemptive and cooperative) and suspended each one to allow a lower-priority thread to execute. This approach worked reliably in version 3.6.0, but in 3.7.1, I’m encountering issues with thread suspension and scheduling behavior.

I’ve reviewed the autoconf.h and zephyr.dts files and did not find any significant changes that would explain this behavior.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

andyross, 2025-07-01:
That said, we shouldn't have scheduler behavior deltas like this.  Absent a few areas (e.g. time slicing is dependent on tick delivery timing, etc...) the scheduler is really supposed to be deterministic.  Apps get in trouble with timer behavior all the time by making assumptions, but scheduling in an RTOS really shouldn't have ambiguity or version-dependent semantics.

peter-mitsis, 2025-07-03:
@rahulprajapati-eaton - Thanks for the sample. I will start looking into this.

nashif, 2025-07-03:
issue introduced in 42036cdbcae270d00b5bd08e805e33bbd3b5787f.

reverted later in d929b8a9faabce092e281c20e4210552e8986be5

nashif, 2025-07-03:
and there was a backport, but it did not get merged in time and got stale and closed :(

rahulprajapati-eaton, 2025-07-04:
Thank you @nashif. Closing this ticket as the issue has been resolved.

closed 2025-07-04 by rahulprajapati-eaton