zephyrproject-rtos/zephyr #92446: STM32 platforms with TF-M fail to boot since 4.2.0-rc1

etienne-lms, 2025-07-01:
### Describe the bug

Since Zephyr bumped to TF-M 2.2.x (commit f67be667c1d3f5b196c8b65cd7bf5f0b2c2ea4a7), STM32 platforms with variant 'ns' fail to boot.
These are **nucleo_l552ze_q**, **stm32l562e_dk** and **stm32l562e_dk**.

### Regression

- [x] This is a regression.

### Steps to reproduce

Reproduced building and programming, for example, the hello_world sample on target boards 'ns' variant:

For **b_u585i_iot02a**:
```
west build -b b_u585i_iot02a/stm32u585xx/ns samples/hello_world/
./build/tfm/api_ns/regression.sh
west flash
```

For **nucleo_l552ze_q**:
```
west build -b nucleo_l552ze_q/stm32l552xx/ns samples/hello_world/
./build/tfm/api_ns/regression.sh
west flash
```

For **stm32l562e_dk**:
```
west build -b stm32l562e_dk/stm32l562xx/ns samples/hello_world/
./build/tfm/api_ns/regression.sh
west flash
```

Build sequence is fine but when the boards boot, there are no trace messages printed
whereas, for the 3 boards mentioned above, we expect build traces like the below:
```
(...)
[INF] Starting bootloader
(...)
[INF] Bootloader chainload address offset: 0x38000
[INF] Jumping to the first image slot
Booting TF-M v2.2.0+65d861552
(...)
Creating an empty ITS flash layout.
Creating an empty PS flash layout.
[INF][PS] Encryption alg: 0x5500200
[INF][Crypto] Init HW accelerator...
[INF][Crypto] Init HW accelerator... complete.
*** Booting Zephyr OS build v4.2.0-rc1 ***
Hello World! b_u585i_iot02a/stm32u585xx/ns
```

### Relevant log output

```shell

```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

- OS: Linux
- Toolchain: Zephyr SDK
- v4.2.0-rc1

wearyzen, 2025-07-01:
I tried running `mps3/corstone300/fvp/ns` and `mps3/corstone310/fvp/ns` and was able to see the hello world logs so, all commits related to BL logs/traces should already be in the 2.2.x branch.

```
[INF] Starting bootloader
[INF] Beginning BL2 provisioning
[WRN] TFM_DUMMY_PROVISIONING is not suitable for production! This device is NOT SECURE
[INF] PSA Crypto init done, sig_type: EC-P256, using builtin keys
[INF] Image index: 1, Swap type: none
[INF] Image index: 0, Swap type: none
[INF] Bootloader chainload address offset: 0x0
[INF] Image version: v0.0.0
[INF] Jumping to the first image slot
Booting TF-M v2.2.0
<NUL>[INF] Beginning TF-M provisioning
<NUL>[WRN] <NUL>TFM_DUMMY_PROVISIONING is not suitable for production! <NUL>This device is NOT SECURE<NUL>
<NUL>[WRN] <NUL>This device was provisioned with dummy keys. <NUL>This device is NOT SECURE<NUL>
<NUL>[Sec Thread] Secure image initializing!
<NUL>Creating an empty ITS flash layout.
Creating an empty PS flash layout.
[INF][PS] Encryption alg: 0x5500200
[INF][Crypto] Provision entropy seed...
[INF][Crypto] Provision entropy seed... complete.
*** Booting Zephyr OS build 4.2.00-rc1 ***
Hello World! mps3/corstone310/fvp/ns
^C
Stopping simulation...
```

Zephyr's TF-M 2.2.x branch has 52 commits added on top on the upstream TF-M's 2.2.x branch to enable different socs, maybe something for these 3 boards was missed? tagging @tomi-font @ceolin if they have any idea about this.

etienne-lms, 2025-07-08:
For info, I still can get my way out of this issue. There seem dependencies with MBEDTLS and MCUBOOT between pristine TF-M expectations and the Zephyr revisions of these modules.

The sole solution I currently see is to disable use of HW crypto accelerators for STM32 platforms' BL2 stage, until the integration issue is addressed. A bit hacky and not very satisfactory I must admit.

closed 2025-07-15 by danieldegrasse