zephyrproject-rtos/zephyr #92461: Broken suspend-to-idle for STM32F1x and probably others

Oleh-Kravchenko, 2025-07-01:
### Describe the bug

During the implementation of LPM for STM32L1x, I've faced a no-boot issue.
I checked boards/st/power_mgmt/blinky with the nucleo_f103rb board and have found that it also doesn't boot.
After a small `git bisect` trip, I've found that it is the regression added by 5a279d9309feda2b0a3f14f21f602f701d776ff1
If I revert this commit, the boards/st/power_mgmt/blinky sample, with nucleo_f103rb, works just fine.

### Regression

- [x] This is a regression.

### Steps to reproduce

Steps to reproduce the bug:
1. west build --board nucleo_f103rb ./zephyr/samples/boards/st/power_mgmt/blinky
2. west flash

### Relevant log output

```shell
With OpenOCD - I see system halted with reason 20.
UART doesn't work to print proper error message.
```

### Impact

Showstopper – Prevents release or major functionality; system unusable.

### Environment

- OS: MacOS
- Toolchain zephyr-sdk-0.16.8
- Commit 5a279d9309feda2b0a3f14f21f602f701d776ff1

etienne-lms, 2025-07-03:
Sorry, I meant "could you check P-R https://github.com/zephyrproject-rtos/zephyr/pull/92525?" (not issue https://github.com/zephyrproject-rtos/zephyr/issues/92511).

Oleh-Kravchenko, 2025-07-03:
@etienne-lms,
I've checked blinky and wkup_pins on my nucleo_f103rb - fixed!

closed 2025-07-04 by danieldegrasse