zephyrproject-rtos/zephyr #92488: tests: tests/drivers/spi/spi_loopback: mimxrt700_evk/mimxrt798s/cm33_cpu0: failed: Failed harness:'Testsuite failed'

hakehuang, 2025-07-02:
**Describe the bug**
below case are failed
drivers.spi.loopback.spi_loopback.spi_rx_every_4
drivers.spi.loopback.spi_loopback.spi_async_call
drivers.spi.loopback.spi_loopback.spi_complete_loop_mode_0
drivers.spi.loopback.spi_loopback.spi_write_back
drivers.spi.loopback.spi_loopback.spi_complete_multiple_timed
drivers.spi.loopback.spi_loopback.spi_word_size_7
drivers.spi.loopback.spi_loopback.spi_word_size_16
drivers.spi.loopback.spi_loopback.spi_complete_multiple
drivers.spi.loopback.spi_loopback.spi_concurrent_transfer_same_spec
drivers.spi.loopback.spi_loopback.spi_word_size_32
drivers.spi.loopback.spi_loopback.spi_complete_loop_mode_1
drivers.spi.loopback.spi_loopback.spi_rx_bigger_than_tx
drivers.spi.loopback.spi_loopback.spi_rx_half_end
drivers.spi.loopback.spi_loopback.spi_complete_loop_mode_3
drivers.spi.loopback.spi_loopback.spi_complete_loop_mode_2
drivers.spi.loopback.spi_loopback.spi_rx_half_start
drivers.spi.loopback.spi_loopback.spi_word_size_24
drivers.spi.loopback.spi_loopback.spi_word_size_9
drivers.spi.loopback.spi_loopback.spi_complete_large_transfers
drivers.spi.loopback.spi_loopback.spi_concurrent_transfer_different_spec

test is failed on v4.2.0-rc1 on mimxrt700_evk/mimxrt798s/cm33_cpu0
testcase path is tests/drivers/spi/spi_loopback

see logs for details

**To Reproduce**
1. 
```
scripts/twister -p mimxrt700_evk/mimxrt798s/cm33_cpu0  -T tests/drivers/spi/spi_loopback -s drivers.spi.loopback
```
or
```
# west twister -p mimxrt700_evk/mimxrt798s/cm33_cpu0 -T tests/drivers/spi/spi_loopback -s drivers.spi.loopback
```
2. See error

```
Failed harness:'Testsuite failed'
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
START - test_spi_rx_every_4
Buffer contents are different:
0x34,0x35,0x36,0x37,
vs:
0xbe,0x3f,0xbf,0x3f,
FAIL - test_spi_rx_every_4 in 0.008 seconds

START - test_spi_complete_loop_mode_0
Buffer contents are different:
0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
0x38,0x39,0x3b,0x3b,0x3e,0x3f,0x3f,0x3f,0x3c,0x3d,0x71,0x73,0x73,0x76,0x77,0x77,0x3f,0x00,
FAIL - test_spi_complete_loop_mode_0 in 0.021 seconds

START - test_spi_write_back
Buffer contents are different:
0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
0xb8,0x39,0xbb,0x3b,0xbe,0x3f,0xbf,0x3f,0xbc,0x3d,0xf1,0xf3,0x73,0xf6,0x77,0xf7,0x3f,0x80,
FAIL - test_spi_write_back in 0.021 seconds

START - test_spi_complete_multiple_timed
Transfer took 319 us vs theoretical minimum 27 us
Latency measurement: 292 us
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:359: spi_loopback_test_spi_complete_multiple_timed: (time_spent_us <= expected_transfer_time_us * CONFIG_SPI_IDEAL_TRANSFER_DURATION_SCALING is false)
Very high latency
FAIL - test_spi_complete_multiple_timed in 0.031 seconds

START - test_spi_word_size_16
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:661: spi_loopback_test_word_size: (memcmp(compare_data, rx_buffer, buffer_size) is true)
16-bit word buffer contents are different
FAIL - test_spi_word_size_16 in 0.020 seconds

START - test_spi_complete_multiple
Buffer contents are different:
0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
0xb8,0x39,0xbb,0x3b,0xbe,0x3f,0xbf,0x3f,0xbc,0x3d,0xf1,0xf3,0x73,0xf6,0x77,0xf7,0x3f,0x80,
FAIL - test_spi_complete_multiple in 0.021 seconds

START - test_spi_concurrent_transfer_same_spec
SPI concurrent transfer data mismatch, spec 0x30184f00
SPI concurrent transfer data mismatch, spec 0x30184f00
SPI concurrent transfer data mismatch, spec 0x30184f00
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:805: test_spi_concurrent_transfer_helper: atomic_get(&thread_test_fails) not equal to 0
FAIL - test_spi_concurrent_transfer_same_spec in 0.032 seconds

START - test_spi_word_size_32
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:661: spi_loopback_test_word_size: (memcmp(compare_data, rx_buffer, buffer_size) is true)
32-bit word buffer contents are different
FAIL - test_spi_word_size_32 in 0.020 seconds

START - test_spi_complete_loop_mode_1
Buffer contents are different:
0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
0xb8,0x39,0xbb,0x3b,0xbe,0x3f,0xbf,0x3f,0xbc,0x3d,0xf1,0xf3,0x73,0xf6,0x77,0xf7,0x3f,0x80,
FAIL - test_spi_complete_loop_mode_1 in 0.021 seconds

START - test_spi_rx_bigger_than_tx
Buffer contents are different:
0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,
vs:
0x38,0x39,0xbb,0x3b,0xbe,0x3f,0xbf,0x3f,
FAIL - test_spi_rx_bigger_than_tx in 0.012 seconds

START - test_spi_rx_half_end
Buffer contents are different:
0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,
vs:
0xbc,0x3d,0xf1,0xf3,0x73,0xf6,0x77,0xf7,
FAIL - test_spi_rx_half_end in 0.012 seconds

START - test_spi_complete_loop_mode_3
Buffer contents are different:
0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
0xb8,0x39,0xbb,0x3b,0xbe,0x3f,0xbf,0x3f,0xbc,0x3d,0xf1,0xf3,0x73,0xf6,0x77,0xf7,0x3f,0x80,
FAIL - test_spi_complete_loop_mode_3 in 0.021 seconds

START - test_spi_complete_loop_mode_2
Buffer contents are different:
0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,0x38,0x39,0x61,0x62,0x63,0x64,0x65,0x66,0x2d,0x00,
vs:
0x38,0x39,0x3b,0x3b,0x3e,0x3f,0x3f,0x3f,0x3c,0x3d,0x71,0x73,0x73,0x76,0x77,0x77,0x3f,0x00,
FAIL - test_spi_complete_loop_mode_2 in 0.021 seconds

START - test_spi_rx_half_start
Buffer contents are different:
0x30,0x31,0x32,0x33,0x34,0x35,0x36,0x37,
vs:
0x38,0x39,0xbb,0x3b,0xbe,0x3f,0xbf,0x3f,
FAIL - test_spi_rx_half_start in 0.012 seconds

START - test_spi_complete_large_transfers
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:556: spi_loopback_test_spi_complete_large_transfers: (memcmp(large_buffer_tx, large_buffer_rx, BUF3_SIZE) is true)
Large Buffer contents are different
FAIL - test_spi_complete_large_transfers in 0.059 seconds

START - test_spi_concurrent_transfer_different_spec
SPI concurrent transfer data mismatch, spec 0x30185aa4
SPI concurrent transfer data mismatch, spec 0x30185a74
SPI concurrent transfer data mismatch, spec 0x30185a8c
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/spi/spi_loopback/src/spi.c:805: test_spi_concurrent_transfer_helper: atomic_get(&thread_test_fails) not equal to 0
FAIL - test_spi_concurrent_transfer_different_spec in 0.032 seconds

    
```

```

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.2.0-rc1

hakehuang, 2025-07-08:
> [@hakehuang](https://github.com/hakehuang) is this a regression?

@dleach02 no, those are new added cases.

decsny, 2025-07-08:
> > [@hakehuang](https://github.com/hakehuang) is this a regression?
> 
> [@dleach02](https://github.com/dleach02) no, those are new added cases.

Actually it looks like every single case is failing, did any of the cases ever pass on this board

hakehuang, 2025-07-08:
> Actually it looks like every single case is failing, did any of the cases ever pass on this board

there are some cases pass before more cases added, such as 

[RT700_spi.csv](https://github.com/user-attachments/files/21115947/RT700_spi.csv)

hakehuang, 2025-07-11:
> What SPI fixes would fix this? I don't think there is any PRs to this driver right now

@decsny , you are right. I just want to wait all other fixes merged and then you can focus on this platform.