zephyrproject-rtos/zephyr #92510: samples: samples/subsys/ipc/ipc_service/static_vrings: mimxrt1180_evk/mimxrt1189/cm33: failed: Timeout

hakehuang, 2025-07-02:
**Describe the bug**
below case are failed
sample.ipc.static_vrings

test is failed on v4.2.0-rc1 on mimxrt1180_evk/mimxrt1189/cm33
testcase path is samples/subsys/ipc/ipc_service/static_vrings

see logs for details

**To Reproduce**
1. 
```
scripts/twister -p mimxrt1180_evk/mimxrt1189/cm33  -T samples/subsys/ipc/ipc_service/static_vrings -s sample.ipc.static_vrings
```
or
```
# west twister -p mimxrt1180_evk/mimxrt1189/cm33 -T samples/subsys/ipc/ipc_service/static_vrings -s sample.ipc.static_vrings
```

git bisect found
175da6bdb0da7c6b626c4e2d62520764998ebc07 is the first bad commit
```
commit 175da6bdb0da7c6b626c4e2d62520764998ebc07
Author: Tomasz Bursztyka <tobu@bang-olufsen.dk>
Date:   Wed May 14 15:59:08 2025 +0200

    init: Make entry init-function less and introduce service objects

    Since the addition of deinit operation in device, init and deinit have
    been within each device, rendering their init entry's init function
    useless.

    In order to save ROM space, let's remove the init function from
    init entry altogether, and introduce a new object called "service"
    which owns an init function to go along with SYS_INIT/SYS_INIT_NAMED.

    Signed-off-by: Tomasz Bursztyka <tobu@bang-olufsen.dk>

 include/zephyr/device.h                            |  1 -
 include/zephyr/init.h                              | 22 +++---
 .../linker/common-rom/common-rom-kernel-devices.ld |  2 +
 include/zephyr/service.h                           | 92 ++++++++++++++++++++++
 kernel/init.c                                      | 26 +++++-
 subsys/net/lib/sockets/sockets_service.c           |  2 +-
 6 files changed, 130 insertions(+), 15 deletions(-)
 create mode 100644 include/zephyr/service.h
```

2. See error

```
Timeout
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build v4.1.0-7103-g175da6bdb0da ***
IPC-service HOST [INST 0 - ENDP A] demo started
IPC-service HOST [INST 0 - ENDP B] demo started
IPC-service HOST [INST 1] demo started

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.2.0-rc1

this also impacts NXP frdm_mcxn947 platforms

dleach02, 2025-07-07:
@tbursztyka Any thoughts on what could be going on here?

hakehuang, 2025-09-30:
> [@hakehuang](https://github.com/hakehuang) , can you please check if this issue still exists.

yes, still fails on v4.2.0-4553-g7fc8051b0f4b