zephyrproject-rtos/zephyr #92521: tests: tests/arch/arm/arm_irq_vector_table: mimxrt1180_evk/mimxrt1189/cm33: failed: Timeout

hakehuang, 2025-07-02:
**Describe the bug**
below case are failed
arch.arm.irq_vector_table.vector_table.arm_irq_vector_table

test is failed on v4.2.0-rc1 on mimxrt1180_evk/mimxrt1189/cm33
testcase path is tests/arch/arm/arm_irq_vector_table

see logs for details

**To Reproduce**
1. 
```
scripts/twister -p mimxrt1180_evk/mimxrt1189/cm33  -T tests/arch/arm/arm_irq_vector_table -s arch.arm.irq_vector_table
```
or
```
# west twister -p mimxrt1180_evk/mimxrt1189/cm33 -T tests/arch/arm/arm_irq_vector_table -s arch.arm.irq_vector_table
```
2. See error

```
Timeout
```

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```

    
```

```

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux ): posix
 - Toolchain (e.g Zephyr SDK): zephyr
 - Commit SHA or Version used: v4.2.0-rc1

dleach02, 2025-07-07:
@hakehuang is this a regression? Or a sporadic failure?

hakehuang, 2025-07-08:
> [@hakehuang](https://github.com/hakehuang) is this a regression? Or a sporadic failure?

@dleach02, this is test case issue, need update the test setting for this platform.

closed 2025-07-22 by hakehuang