zephyrproject-rtos/zephyr #92528: RT1061 I2C stuck in non-stop interrupts after debug halt

ofirshe, 2025-07-02:
### Describe the bug

Hi,

We have an RT1061 on a custom board, and one of the I2C buses has many devices on it which are sampled periodically. I’ve noticed that if I load my application through Segger and halt the CPU in the debugger for a "long" time (2–3 seconds) after periodic sampling has started, then upon resuming, the CPU gets interrupted endlessly by the I2C peripheral.
It continuously enters LPI2C_MasterTransferHandleIRQ, and if I dig into the call to LPI2C_RunTransferStateMachine, I see that stateParams.status is always either:
* 769 - with MSR[TDF], MSR[EPF], and MSR[SDF] set, and the SDA line is high/idle:
[image]
* 50332417 — with MSR[MBF] and MSR[BBF] also set, and the SDA line is low:
[image]
In both cases, txCount is always 4.
Once this state occurs, the I2C bus remains idle (no activity), and the RT system becomes non-functional, as it gets stuck handling continuous interrupts. I can confirm this by halting the CPU again and seeing it's always inside LPI2C_MasterTransferHandleIRQ.
This behavior happens on both versions 4.1 and 4.2-rc1, so it doesn’t appear to be a regression - more likely a core issue. I suspect halting the CPU mid-transaction may leave the I2C peripheral in an unrecoverable state. The only way to recover is to power cycle the board.
To clarify, this has never happened during normal execution without a debugger, but it makes debugging difficult because halting the CPU can lead to this hang.
Any ideas on how to recover from this state or prevent the IRQ flood after resuming from debug halt?

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Additional Context

I've never seen this on normal run without debugger but when i debug from the start of the main and i hold the the cpu halted for long time it occurs.

ofirshe, 2025-07-24:
Hi @dpiskula-nxp, any update ?

dpiskula-nxp, 2025-07-25:
Hello @ofirshe the CONFIG_BOARD_EARLY_INIT_HOOK is calling the function too soon and the configuration gets overwritten during the system init process. You can either configure it in main.c or if you'd prefer to have everything ready before main, I suggest this approach:

create a new file, e.g. src/board_setup.c, add it to CMakeLists.txt: _target_sources(app PRIVATE src/board_setup.c)_

In the file, add this code:

```
#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/init.h>

#define I2C_NODE DT_NODELABEL(lpi2c1)

static int lpi2c1_mcr_dbgen_enable_init(const struct device *arg)
{
    ARG_UNUSED(arg);

    LPI2C_Type *base = (LPI2C_Type *) DT_REG_ADDR(I2C_NODE);
    base->MCR |= LPI2C_MCR_DBGEN_MASK;

    return 0;
}

SYS_INIT(lpi2c1_mcr_dbgen_enable_init, POST_KERNEL, 99);
```

If you look inside **zephyr\drivers\i2c\i2c_mcux_lpi2c.c**, you'll see that the LPI2C peripheral is initialized during the POST_KERNEL phase with priority 50:
```
I2C_DEVICE_DT_INST_DEFINE(n, mcux_lpi2c_init, NULL,		\
	&mcux_lpi2c_data_##n,			\
	&mcux_lpi2c_config_##n, POST_KERNEL,	\
	CONFIG_I2C_INIT_PRIORITY,		\
	&mcux_lpi2c_driver_api);
```

Thus, using SYS_INIT() with POST_KERNEL level and priority of 99 ensures the function is called late enough.

To verify the change, you can call this in main.c:

```
LPI2C_Type *base = (LPI2C_Type *) DT_REG_ADDR(I2C_NODE);
printf("%d\n", (base->MCR & LPI2C_MCR_DBGEN_MASK) >> LPI2C_MCR_DBGEN_SHIFT);
```

Please, let me know if this helps you with the debug experience.

Best Regards,
David