zephyrproject-rtos/zephyr #92537: STM32L5 async UART is not working

SeppoTakalo, 2025-07-02:
### Describe the bug

I have Nucleo L552ZE Q board to test.
It seem like [Async UART API](https://docs.zephyrproject.org/apidoc/latest/group__uart__async.html) is not working on this board, SOC or platform at all.

When running `tests/drivers/uart/uart_async_api/` all tests fail. It looks like all RX data comes back as full of zeroes. This test case provides DMA configuration on the `boards/` directory for this board, so I assume that is not causing the issue.

Basic UART tests in `tests/drivers/uart/uart_basic_api` seem to work fine.

### Regression

- [ ] This is a regression.

### Steps to reproduce

Build and flash
```
cd tests/drivers/uart/uart_async_api/
west build -b nucleo_l552ze_q/stm32l552xx
west flash --erase
```

### Relevant log output

```shell
------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [uart_async_chain_read]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.094 seconds
 - FAIL - [uart_async_chain_read.test_chained_read] duration = 0.094 seconds

SUITE FAIL -   0.00% [uart_async_chain_write]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.023 seconds
 - FAIL - [uart_async_chain_write.test_chained_write] duration = 0.023 seconds

SUITE FAIL -   0.00% [uart_async_double_buf]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.046 seconds
 - FAIL - [uart_async_double_buf.test_double_buffer] duration = 0.046 seconds

SUITE FAIL -   0.00% [uart_async_long_buf]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.075 seconds
 - FAIL - [uart_async_long_buf.test_long_buffers] duration = 0.075 seconds

SUITE FAIL -   0.00% [uart_async_multi_rx]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.524 seconds
 - FAIL - [uart_async_multi_rx.test_multiple_rx_enable] duration = 0.524 seconds

SUITE FAIL -   0.00% [uart_async_read_abort]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.069 seconds
 - FAIL - [uart_async_read_abort.test_read_abort] duration = 0.069 seconds

SUITE FAIL -   0.00% [uart_async_single_read]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.274 seconds
 - FAIL - [uart_async_single_read.test_single_read] duration = 0.274 seconds

SUITE FAIL -   0.00% [uart_async_timeout]: pass = 0, fail = 1, skip = 0, total = 1 duration = 3.019 seconds
 - FAIL - [uart_async_timeout.test_forever_timeout] duration = 3.019 seconds

SUITE FAIL -   0.00% [uart_async_var_buf_length]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.032 seconds
 - FAIL - [uart_async_var_buf_length.test_var_buf_length] duration = 0.032 seconds

SUITE FAIL -   0.00% [uart_async_write_abort]: pass = 0, fail = 1, skip = 0, total = 1 duration = 0.070 seconds
 - FAIL - [uart_async_write_abort.test_write_abort] duration = 0.070 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

- OS: Linux
- Zephyr: 4bd1d39c67d
- Toolchain: Zephyr SDK 0.16.8

gautierg-st, 2025-07-03:
Hi @SeppoTakalo,
I've just tested on a Nucleo L552ZE Q and it works fine on my end. Did you put the loopback in place between Arduino pin D0/D1?

SeppoTakalo, 2025-07-03:
Also, without the loopback, the test results are different.

### Without loopback, no data RX
```
Running TESTSUITE uart_async_chain_read
===================================================================
UART instance:serial@40004800
START - test_chained_read

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:434: uart_async_chain_read_test_chained_read: (rx_data_idx not equal to sizeof(tx_buf))
Unexpected amount of data received 0 exp:10
 FAIL - test_chained_read in 0.092 seconds
===================================================================
TESTSUITE uart_async_chain_read failed.
Running TESTSUITE uart_async_chain_write
===================================================================
UART instance:serial@40004800
START - test_chained_write

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:890: uart_async_chain_write_test_chained_write: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_chained_write in 0.121 seconds
===================================================================
```

### With loopback, content mismatch
```
Running TESTSUITE uart_async_chain_read
===================================================================
UART instance:serial@40004800
START - test_chained_read

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:437: uart_async_chain_read_test_chained_read: (memcmp(tx_buf, chained_cpy_buf, sizeof(tx_buf)) not equal to 0)
Buffers not equal exp Message 0, real 
 FAIL - test_chained_read in 0.094 seconds
===================================================================
TESTSUITE uart_async_chain_read failed.
Running TESTSUITE uart_async_chain_write
===================================================================
UART instance:serial@40004800
START - test_chained_write

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:891: uart_async_chain_write_test_chained_write: (memcmp(chained_write_tx_bufs[0], rx_buf, 10) not equal to 0)
Buffers not equal
 FAIL - test_chained_write in 0.023 seconds
===================================================================
```

So it receives correct amount of bytes, but buffers are all empty.

SeppoTakalo, 2025-07-14:
@gautierg-st I found the issue.

As you indicated, only difference was the use of STM32Cubeprogrammer. Well, that was not the root cause, but after lots of googling and trying it out, I found out that my TZEN bit was set. And the `./build/tfm/api_ns/regression.sh` script, as documented in the board page, did not really unset it.

After a while, I found a workaround explained here: https://wiki.st.com/stm32mcu/wiki/Security:How_to_disable_TrustZone_in_STM32L5xx_devices_during_development_phase and now after disabling the TZEN bit, my board is receiving the data from DMA.

```
------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [uart_async_chain_read]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.428 seconds
 - PASS - [uart_async_chain_read.test_chained_read] duration = 0.428 seconds

SUITE PASS - 100.00% [uart_async_chain_write]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.003 seconds
 - PASS - [uart_async_chain_write.test_chained_write] duration = 0.003 seconds

SUITE PASS - 100.00% [uart_async_double_buf]: pass = 1, fail = 0, skip = 0, total = 1 duration = 1.733 seconds
 - PASS - [uart_async_double_buf.test_double_buffer] duration = 1.733 seconds

SUITE PASS - 100.00% [uart_async_long_buf]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.164 seconds
 - PASS - [uart_async_long_buf.test_long_buffers] duration = 0.164 seconds

SUITE PASS - 100.00% [uart_async_multi_rx]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.703 seconds
 - PASS - [uart_async_multi_rx.test_multiple_rx_enable] duration = 0.703 seconds

SUITE PASS - 100.00% [uart_async_read_abort]: pass = 1, fail = 0, skip = 0, total = 1 duration = 1.120 seconds
 - PASS - [uart_async_read_abort.test_read_abort] duration = 1.120 seconds

SUITE PASS - 100.00% [uart_async_single_read]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.352 seconds
 - PASS - [uart_async_single_read.test_single_read] duration = 0.352 seconds

SUITE PASS - 100.00% [uart_async_timeout]: pass = 1, fail = 0, skip = 0, total = 1 duration = 3.001 seconds
 - PASS - [uart_async_timeout.test_forever_timeout] duration = 3.001 seconds

SUITE PASS - 100.00% [uart_async_var_buf_length]: pass = 1, fail = 0, skip = 0, total = 1 duration = 1.011 seconds
 - PASS - [uart_async_var_buf_length.test_var_buf_length] duration = 1.011 seconds

SUITE PASS - 100.00% [uart_async_write_abort]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.132 seconds
 - PASS - [uart_async_write_abort.test_write_abort] duration = 0.132 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

@nashif this issue can be closed as it was not a Zephyr issue.

closed 2025-07-14 by SeppoTakalo