zephyrproject-rtos/zephyr #92542: NXP: RW61x: Flexcomm SPI CS

matt-wood-ct, 2025-07-02:
### Describe the bug

Configuring software (gpio) chip selects appears to not work, the HW CS0 is stuck enabled and activates with transactions to other devices on the bus.

If you attempt to configure the original CS0 as a gpio based CS, the entire spi pin muxing breaks and clock, miso and mosi stop functioning.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Define a board or overlay which uses a flexcomm spi with multiple slaves (`cs-gpio` not overlapping CS0)
2. capture CS0 pin state with a logic analyser, you will see it goes low with every transaction to any slave

Test 2

1. Define a board or overlay which uses a flexcomm spi with one or more slaves, overlap the CS0 with its associated CS
2. Capture sck with a logic analyser, you will see no activity.

### Relevant log output

```shell

```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

- Ubuntu 24.04 LTS
- SHA: 4bd1d39c67df
- Board: custom RW610 based.

matt-wood-ct, 2025-07-02:
The IO are grouped on the RW610 FCx mux and the FCx_SPI muxing is higher priority then GPIO so it is not possible to remux the cs into a gpio, the pinctrl knows this and unmuxes the whole SPI if you even try in DT. The actual internal transactions to send over SPI allow for specifying if a HW CS is used on a transaction so it should be possible to identify CS pins in the cs-gpio and simply set the relevant flag instead, that would be a clean solution

decsny, 2025-07-15:
I am confused at what you are describing, are you saying you are trying to use the native CS0 and other gpios CS? Or vice versa? Or not trying to use the native CS at all? Can you provide an overlay to illustrate what you mean?

DerekSnell, 2025-08-19:
Hi @matt-wood-ct ,
I believe the issue you are describing is a hardware limitation of the RW61x SOC.  As you said, pinmuxing of signals is done by groups of signals.  The snippets below are from the User Manual and the Register Manual, and shows that if the bit `SEL_FC0_SPI` is set, then the 4 pins tied to FC0 become SPI signals, including the HW chip select.

If you want to use software GPIO chip selects, you need to use pins that are not grouped in the same `SEL_FCx_SPI` group.  It seems to me, you already see that option with "multiple slaves (cs-gpio not overlapping CS0)", right?  

Can you clarify the software issue here in the Zephyr repo that you hope to see resolved?  Or if you agree this must be dealt with in hardware, then I think we should close this issue.  Thank you

[image]

[image]

[image]

matt-wood-ct, 2025-08-20:
> Hi [@matt-wood-ct](https://github.com/matt-wood-ct) , I believe the issue you are describing is a hardware limitation of the RW61x SOC. 

Yes, but from a driver perspective, the flex comm module is capable of specifying which CS is used for a transaction, this chip only physically has one CS, so it should be possible to mix soft and hard chip selects by selecting a "dead" HW CS channel when using soft CS pins for transactions. There is no precedent for this in zephyr as far as I can tell, but frankly with this peripheral as it is cant really be used with multiple slaves.

_personal remarks_  
I am vocal on this issue because we got caught out by this poor muxing situation, priority based muxing is a very inconvenient strategy. The RW61X IC has one of the worst pin muxing setups of any SOC I've encountered in my career so far, it feels very out of pace with other chip vendors designs, many vendors are leaning into universal muxing on low speed peripherals like this.

matt-wood-ct, 2025-08-21:
I've realized this issue thread has become mixed between a feature request and a bug, there is still a bug which applies, the HW chip select is still active when you use fully gpio based CS pins, it operates will all bus transactions.

That issue is really very distinct from the ideal outcome for me which is the ability to mix soft and hard CS

decsny, 2025-10-29:
@matt-wood-ct The pinctrl on RW6xx is indeed a black sheep type of situation.

If I am understanding you correctly, you just want to be able to have multiple slaves on one SPI bus, and don't care what is the method of controlling the CS.

And issue you are running into is that the hardware of the RW61x only supports one HW based CS due to minimal pinctrl scheme. But also does not support software CS being used **on the same pin** as the HW CS (also due to all-or-nothing pinctrl scheme).

Here is how I have analyzed this issue:
- It is **not** possible to use two HW chip selects from same controller is due to SOC limitation of only allowing a specific one of the CS to be muxed to a pin and useable for each a flexcomm (apparently, CS 0).
- It is **not** possible to use a software CS on the same pin as the HW CS would be if using that flexcomm controller, due to pinctrl cannot be separated.
- It **is** possible to use multiple GPIO/software CS with the flexcomm controller, just not with the pin that is forced to be muxed to HW CS. So that pin can be used as CS0 and all other CS can be driven by software GPIO. Or you can use all software CS and not connect or use that HW CS pin.
- It **is** possible to use SPI bitbang driver using all GPIOs instead of flexcomm peripheral, if the CPU utilization and bandwidth is not an issue. (last resort, but I think it's not better than previous option)
- As far as mixing by having the one HW CS useable with other CS being software, this should be "theoretically possible". But I found that the current problem is that the DT description of the CS-GPIOS implicitly is specifying the CS starting with CS0, and not possible to explicitly say which CS each GPIO corresponds to. The ability to do this is I guess a feature request, which I will make a separate ticket for and probably propose it sometime after the release window for 4.3, as I am also a SPI collaborator in Zephyr and I think this is a fair request to be able to do this, and I have some idea of what I would propose (will put more info in the ticket I make).

For this ticket conclusion, as it is marked as bug currently, I would say, there is not really a bug here in software in regards to the ability to control the CS. The problem of not being able to mux the HW CS pin to a GPIO while using the rest of the FC pins as SPI, is an SOC hardware limitation of RW, and cannot be "fixed" by software. The problem of not being able to "mix" HW and SW CS, is IMO a feature request of zephyr generally, and not a bug.

But, I noticed there is a bug as far as the slave number validation in the driver, and also an expectation mismanagement about what slave number is valid. So I will link a PR to this issue which fixes some issues around validating the slave select number passed to the driver, and will close this ticket when that merges, and move the mixing feature request to a different ticket.