zephyrproject-rtos/zephyr #92551: ZMS: Big values: Size must be multiple of 4?

clamattia, 2025-07-02:
### Describe the bug

- We are attempting to write values into zms
- We have observed, that objects larger then x (255) need to have a multiple of 4 or the write will fail

### Regression

- [ ] This is a regression.

### Steps to reproduce

- large blocks (65K) -> zms_mount
- zms_write
  - if size > 255
  - and size % 4 != 0
  - FAILS

### Relevant log output

```shell
TBD
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

TBD

### Additional Context

TBD

rghaddab, 2025-07-03:
Hi @clamattia please describe the bug more in details :
provide your device tree node for the storage
provide a snippet of the failing sample
provide the error code that you get when this happens

rghaddab, 2025-07-09:
@erwango do you maintain these drivers ?

mathieuchopstm, 2025-07-09:
@FRASTM PTAL when possible

erwango, 2025-07-09:
> zephyr 4.1 with cherry-picks

Can you test on main branch ?
We need to be sure we're investigating main branch issues, not out of tree issues.

clamattia, 2025-07-14:
> > zephyr 4.1 with cherry-picks
> 
> Can you test on main branch ? We need to be sure we're investigating main branch issues, not out of tree issues.

@erwango attached sample reproduces on `v4.2.0-rc3-0-g322da1d1a5a`

closed 2025-09-15 by rghaddab

tpambor, 2025-09-16:
This issue is still present on main as of 5525941d839099309a44a0be868151fa894a5425. I was able to further slim down the example to reproduce the bug to:

```
#include <zephyr/fs/zms.h>
#include <zephyr/storage/flash_map.h>
#include <string.h>

#define ZMS_BLOCK_SIZE 65536

static struct zms_fs fs;

int main(void)
{
    int rc;
    const struct device *flash_dev;
    static char buffer0[4000] = {0};
    static char buffer1[4000] = {0};

    for (size_t i = 0; i < sizeof(buffer0); ++i) {
        buffer0[i] = (char)(i % 200);
    }

    flash_dev = FIXED_PARTITION_DEVICE(scratch_partition);
    __ASSERT(device_is_ready(flash_dev), "device must be ready");

    fs.flash_device = flash_dev;
    fs.offset = FIXED_PARTITION_OFFSET(scratch_partition);
    size_t partition_size = FIXED_PARTITION_SIZE(scratch_partition);

    fs.sector_size = ZMS_BLOCK_SIZE;
    fs.sector_count = partition_size / fs.sector_size;

    rc = zms_mount(&fs);
    __ASSERT(rc == 0, "mount failed");
    __ASSERT(fs.ready, "fs not ready");

    for (size_t i = 1; i < sizeof(buffer0); ++i) {
        printk("testing %zu\n", i);
        k_msleep(1000);
        rc = zms_delete(&fs, 1);
        __ASSERT(rc == 0, "delete error");
        rc = zms_write(&fs, 1, buffer0, i);
        __ASSERT(rc > 0 && (size_t)rc == i, "write error");
        int rc = zms_read(&fs, 1, &buffer1, i);
        __ASSERT(rc > 0 && (size_t)rc == i, "read error");
        __ASSERT(memcmp(buffer0, buffer1, i) == 0, "Did not read back ok");
    }

    return 0;
}
```

with config:
```
CONFIG_LOG=y
CONFIG_LOG_MODE_DEFERRED=y
CONFIG_CBPRINTF_FP_SUPPORT=y
CONFIG_LOG_FUNC_NAME_PREFIX_DBG=y
CONFIG_LOG_FUNC_NAME_PREFIX_ERR=y
CONFIG_LOG_FUNC_NAME_PREFIX_INF=y
CONFIG_LOG_FUNC_NAME_PREFIX_WRN=y
CONFIG_LOG_TIMESTAMP_64BIT=y

CONFIG_MAIN_STACK_SIZE=4512
CONFIG_ISR_STACK_SIZE=4512

CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_ZMS=y

CONFIG_ASSERT=y
CONFIG_ASSERT_LEVEL=2
```

tpambor, 2025-09-16:
@rghaddab As suggested I tried to increase `HAL_XSPI_TIMEOUT_DEFAULT_VALUE` from 5s to 30s. It hangs then 30 seconds before it fails with
```
[00:00:00.140,000] <inf> flash_stm32_xspi: stm32_xspi_config_mem: XSPI flash config is OCTO / DTR
[00:00:00.140,000] <inf> flash_stm32_xspi: xspi_read_sfdp: Read SFDP from externalFlash
[00:00:00.140,000] <inf> flash_stm32_xspi: xspi_read_sfdp: Read SFDP from externalFlash
[00:00:00.140,000] <inf> flash_stm32_xspi: flash_stm32_xspi_init: NOR external-flash at 0x90000000 (0x4000000 bytes)
*** Booting Zephyr OS build v4.2.0-3746-g5525941d8390 ***
[00:00:00.224,000] <inf> fs_zms: zms_mount: 384 Sectors of 65536 bytes
[00:00:00.224,000] <inf> fs_zms: zms_mount: alloc wra: 2, 7670
[00:00:00.224,000] <inf> fs_zms: zms_mount: data wra: 2, 158a
testing 1
testing 2
testing 3
testing 4
testing 5
testing 6
testing 7
testing 8
testing 9
tess[00:00:40.228,000] <err> flash_stm32_xspi: stm32_xspi_mem_ready: XSPI AutoPoll command failed
[00:00:40.228,000] <err> flash_stm32_xspi: flash_stm32_xspi_write: XSPI: write not ready
```