zephyrproject-rtos/zephyr #92565: Please add Raymond022 to the Contributor list

dleach02, 2025-07-02:
### Full Name

Raymond Lei 

### GitHub Username

Raymond022

### Organization

NXP

### Supporting Documents

Raymond is one of the NXP Drivers collaborators. 

- https://github.com/zephyrproject-rtos/zephyr/pulls?q=sort%3Aupdated-desc+is%3Apr+author%3ARaymond0225+is%3Amerged
- https://github.com/zephyrproject-rtos/zephyr/pull/88877
- https://github.com/zephyrproject-rtos/zephyr/pull/91853
- https://github.com/zephyrproject-rtos/zephyr/pull/86588

stephanosio, 2025-07-17:
@dleach02 Is it @Raymond022 or @Raymond0225? Because @Raymond0225 is already a collaborator.

nashif, 2025-07-17:
i think this is a typo, it should be Raymond0225, and he is already in the system.

closed 2025-07-17 by nashif