zephyrproject-rtos/zephyr #92613: [Coverity CID: 434743] Out-of-bounds access in drivers/lora/rylrxxx.c

zephyrbot, 2025-07-03:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/fd6b05c81b86a1ffe465d56ded7b73a9fdb7f14b/drivers/lora/rylrxxx.c

Category: Memory - corruptions
Function: `rylr_init`
Component: Drivers
CID: [434743](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434743)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/fd6b05c81b86a1ffe465d56ded7b73a9fdb7f14b/drivers/lora/rylrxxx.c#L620

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434743

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

Also see the related [Coverity documentation](https://docs.zephyrproject.org/latest/contribute/guidelines.html#static-code-analysis) for more information.

JordanYates, 2025-07-04:
This looks like a false positive @ljd42, I can't see a way in which a `memset` call will overrun the variable.
The input variable is definitely of the intended type (`struct modem_chat`).

[image]

ljd42, 2025-07-05:
Yes, it's a known false positive with Coverity. Coverity seemed confused when a structure has conditional fields, the structure is used in two different compile units and the size is computed through sizeof(*ptr). @bjarki-andreasen has similar issue with GNSS driver, like #92614.

Marked in Coverity as false positive.

ljd42, 2025-07-05:
@JordanYates  or @kartben : Issue can be closed.

closed 2025-07-06 by JordanYates