zephyrproject-rtos/zephyr #92625: [Coverity CID: 434574] Unchecked return value in drivers/sensor/apds9306/apds9306.c

zephyrbot, 2025-07-03:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/fd6b05c81b86a1ffe465d56ded7b73a9fdb7f14b/drivers/sensor/apds9306/apds9306.c

Category: Error handling issues
Function: `apds9306_sensor_setup`
Component: Drivers
CID: [434574](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434574)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/fd6b05c81b86a1ffe465d56ded7b73a9fdb7f14b/drivers/sensor/apds9306/apds9306.c#L310

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=434574

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_CHECKED_RETURN). ([CWE-252](http://cwe.mitre.org/data/definitions/252.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

Also see the related [Coverity documentation](https://docs.zephyrproject.org/latest/contribute/guidelines.html#static-code-analysis) for more information.

ljd42, 2025-07-09:
Analysis
------------
As commented, it is a dummy read operation is a work-around for I2C error after a reset of the chip, see #89850
This is best effort, and it makes here no sense to have an error handling at that stage. 

@Kampi  : is my understanding correct?

ljd42, 2025-07-09:
Marked in Coverity as False Positive and  issue labelled as such. 

@danieldegrasse or @dkalowsk: ticket can be closed.

closed 2025-07-15 by ljd42