zephyrproject-rtos/zephyr #92670: TFM_Integration: Using own custom key for image signing does not work

arnebdr, 2025-07-04:
### Describe the bug

**Describe the bug**
Image signing with custom keys for TFM applications does not work as described here: [https://docs.zephyrproject.org/latest/services/tfm/build.html#signing-images](url)

Output of MCUBoot when custom keys are used is:

> [INF] Starting bootloader
> [WRN] This device was provisioned with dummy keys. This device is NOT SECURE
> [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
> [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
> [INF] Boot source: primary slot
> [INF] Image index: 1, Swap type: none
> [ERR] Image in the primary slot is not valid!
> [ERR] Image in the primary slot is not valid!
> [ERR] Unable to find bootable image

**Steps to reproduce**

1.  Create two new keys with the commands 
```
imgtool keygen -k root-rsa-3072_custom_s.pem -t rsa-3072
imgtool keygen -k root-rsa-3072_custom_ns.pem -t rsa-3072
```
2. Edit samples/tfm_integration/tfm_ipc/prj.conf and add the lines:
```
CONFIG_TFM_BL2=y
CONFIG_TFM_MCUBOOT_SIGNATURE_TYPE="RSA-3072"
CONFIG_TFM_KEY_FILE_S="C:/Users/Benutzer/zephyrproject/zephyr/samples/tfm_integration/tfm_ipc/root-rsa-3072_custom_s.pem"
CONFIG_TFM_KEY_FILE_NS="C:/Users/Benutzer/zephyrproject/zephyr/samples/tfm_integration/tfm_ipc/root-rsa-3072_custom_ns.pem"
```
3. Build the sample and flash:
```
west build -p -b b_u585i_iot02a/stm32u585xx/ns samples/tfm_integration/tfm_ipc/
sh build/tfm/api_ns/regression.sh
west flash
```
4. Open shell, reset the board. Shell Output is as above.

**Relevant log output**
> [INF] Starting bootloader
> [WRN] This device was provisioned with dummy keys. This device is NOT SECURE
> [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
> [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
> [INF] Boot source: primary slot
> [INF] Image index: 1, Swap type: none
> [ERR] Image in the primary slot is not valid!
> [ERR] Image in the primary slot is not valid!
> [ERR] Unable to find bootable image

**Impact**
Show stopper, secure boot only working with default key.

**Environment**
Zephyr 4.1 running on STM32U585 Board b_u585i_iot02a

**Additional Context**
When the key paths are set to the default keys booting is working as normal:
```
CONFIG_TFM_KEY_FILE_S="C:/Users/Benutzer/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-RSA-3072.pem"
CONFIG_TFM_KEY_FILE_NS="C:/Users/Benutzer/zephyrproject/modules/tee/tf-m/trusted-firmware-m/bl2/ext/mcuboot/root-RSA-3072_1.pem"
```

> [INF] Starting bootloader
> [WRN] This device was provisioned with dummy keys. This device is NOT SECURE
> [INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
> [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
> [INF] Boot source: primary slot
> [INF] Image index: 1, Swap type: none
> [INF] Primary image: magic=bad, swap_type=0x1, copy_done=0x3, image_ok=0x3
> [INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
> [INF] Boot source: none
> [INF] Image index: 0, Swap type: none
> [INF] Bootloader chainload address offset: 0x38000
> [INF] Jumping to the first image slot
> Booting TF-M v2.1.2
> [WRN] This device was provisioned with dummy keys. This device is NOT SECURE
> [Sec Thread] Secure image initializing!
> [INF][PS] Encryption alg: 0x5500200
> [INF][Crypto] Init HW accelerator...
> [INF][Crypto] Init HW accelerator... complete.
> *** Booting Zephyr OS build v4.1.0-34-g7d39a3245667 ***
> TF-M IPC on b_u585i_iot02a
> The version of the PSA Framework API is 257.
> The PSA Crypto service minor version is 1.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

arnebdr, 2025-07-04:
Verification of the signatures with imagetool is successfull, so the reason must be that MCUBoot does not have the correct public key

```
imgtool verify --key root-rsa-3072_custom_s.pem build/tfm/bin/tfm_s_signed.bin
imgtool verify --key root-rsa-3072_custom_ns.pem build/zephyr/zephyr_ns_signed.hex
```

arnebdr, 2025-07-04:
Thanks for sharing your result. 
I've tested "CONFIG_TFM_DUMMY_PROVISIONING=n" but it seems to have no effect, the log still says:
> [WRN] This device was provisioned with dummy keys. This device is NOT SECURE
And the image validation still fails.

joerchan, 2025-07-05:
@Grolleau-Benjamin For the nRF5340DK you should use https://docs.nordicsemi.com/bundle/ncs-3.0.0/page/nrf/samples/tfm/tfm_psa_template/README.html.
You need to use the nRF Connect SDK, as you won't get a proper PSA compliant TF-M build without it.

That does not apply for the STM32 board in the original bug description though.
I would check the `build/tfm/CMakeCache.txt` (path my be different with sysbuild) file to see if:
MCUBOOT_KEY_NS MCUBOOT_KEY_S matches your key paths.

arnebdr, 2025-07-05:
Thanks, I already checked MCUBOOT_KEY_NS and MCUBOOT_KEY_S in tfm/CMakeCache.txt, they are set to the correct path as defined with CONFIG_TFM_KEY_FILE_S and CONFIG_TFM_KEY_FILE_NS.
I also used imgtool to check if tfm_s_signed.bin and zephyr_ns_signed.hex are correctly signed with the given key. And they are correct.
So the only explanation is that MCUBoot got the wrong public key build in.

arnebdr, 2025-07-07:
Yes TFM does use MCUBoot.
I have found out that MCUBOOT_HW_KEY is ON by default for my board.
This means that the Headers of the Secure and Non-Secure Images contain the Public Key and the MCUBoot Image should contain the corresponding Hashes of the Public keys.
In the build binaries of the S/NS-Images I can find the correct public key that was specified with CONFIG_TFM_KEY_FILE_S/CONFIG_TFM_KEY_FILE_NS.
But the binary of BL2 (MCUBoot) only contains the Hashes of the default keys.

Grolleau-Benjamin, 2026-09-07:
Hi @tomi-font, after the merge of those issues, I think we can close this issue as completed! 
- https://github.com/zephyrproject-rtos/zephyr/pull/114040
- https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/205

closed 2026-09-07 by tomi-font