zephyrproject-rtos/zephyr #92711: dead link to pico RP2020 datasheet

doomdagadiggiedahdah, 2025-07-04:
### Summary

"They are equipped with an RP2040 SoC, an on-board LED, a USB connector, and an SWD interface." -[source](https://docs.zephyrproject.org/latest/boards/raspberrypi/rpi_pico/doc/index.html)

following link under RP2020 is a deadend

### Describe the solution you'd like

link to datasheet

moonlight83340, 2025-07-05:
Hi,
I don't think the documentation should have a direct link to the datasheet. (It's not done with any other raspberry pi board.)
You can have access on raspberry pi website.
I have made a PR to remove the dead/wrong link to the datasheet.
https://www.raspberrypi.com/products/rp2040/

closed 2025-07-16 by danieldegrasse

doomdagadiggiedahdah, 2025-07-21:
Thanks for following up!