zephyrproject-rtos/zephyr #93017: SOF compilation broken with recent Zephyr

lyakh, 2025-07-11:
### Describe the bug

Trying to upgrade SOF to a recent Zephyr fails compilation with:
```
clang: error: unsupported option '--config=/home/runner/work/sof/sof/workspace/zephyr/cmake/toolchain/llvm/clang_libgcc.cfg'
```

### Regression

- [x] This is a regression.

### Steps to reproduce

- Update SOF to use the latest Zephyr like in https://github.com/thesofproject/sof/pull/10113
- Run CI tests

### Relevant log output

```shell
[10/315] Building C object zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj
FAILED: [code=1] zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj 
/usr/bin/clang -DKERNEL -DK_HEAP_MEM_POOL_SIZE=2048 -D__ZEPHYR__=1 -I/home/runner/work/sof/sof/workspace/zephyr/kernel/include -I/home/runner/work/sof/sof/workspace/zephyr/arch/posix/include -I/home/runner/work/sof/sof/workspace/build-fuzz/zephyr/include/generated/zephyr -I/home/runner/work/sof/sof/workspace/zephyr/include -I/home/runner/work/sof/sof/workspace/build-fuzz/zephyr/include/generated -I/home/runner/work/sof/sof/workspace/zephyr/soc/native/inf_clock -I/home/runner/work/sof/sof/workspace/zephyr/boards/native/native_sim -I/home/runner/work/sof/sof/workspace/zephyr/scripts/native_simulator/common/src/include -I/home/runner/work/sof/sof/workspace/zephyr/scripts/native_simulator/native/src/include -I/home/runner/work/sof/sof/workspace/zephyr/drivers -I/home/runner/work/sof/sof/workspace/sof/zephyr/include -I/home/runner/work/sof/sof/workspace/zephyr/arch/host/include -I/home/runner/work/sof/sof/workspace/sof/src/platform/posix/include -isystem /home/runner/work/sof/sof/workspace/zephyr/lib/libc/minimal/include -isystem /home/runner/work/sof/sof/workspace/zephyr/lib/libc/common/include -fno-strict-aliasing -O2 -imacros /home/runner/work/sof/sof/workspace/build-fuzz/zephyr/include/generated/zephyr/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fcolor-diagnostics --config=/home/runner/work/sof/sof/workspace/zephyr/cmake/toolchain/llvm/clang_libgcc.cfg -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-unused-but-set-variable -Wno-typedef-redefinition -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wno-self-assign -Wno-initializer-overrides -Wno-section -Wno-gnu -Werror=implicit-int -fno-pic -fno-pie -Werror -fno-asynchronous-unwind-tables -fstrict-overflow -Wno-vla -fmacro-prefix-map=/home/runner/work/sof/sof/workspace/sof/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/runner/work/sof/sof/workspace/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/runner/work/sof/sof/workspace=WEST_TOPDIR -ffunction-sections -fdata-sections -m32 -msse2 -mfpmath=sse -fvisibility=hidden -nostdinc -isystem /usr/lib/llvm-14/lib/clang/14.0.0/include -include /home/runner/work/sof/sof/workspace/zephyr/arch/posix/include/undef_system_defines.h -fno-builtin -fsanitize-recover=all -fPIC -fsanitize=address,fuzzer -std=c11 -fno-lto -MD -MT zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -c /home/runner/work/sof/sof/workspace/zephyr/arch/posix/core/offsets/offsets.c
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/runner/work/sof/sof/workspace/build-fuzz
clang: error: unsupported option '--config=/home/runner/work/sof/sof/workspace/zephyr/cmake/toolchain/llvm/clang_libgcc.cfg'
ninja: build stopped: subcommand failed.
Error: Process completed with exit code 1.
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

Zephyr commit: e22ca6b132840efb23b4a00ff7718225a0dd9f29

henrikbrixandersen, 2025-07-11:
How come this is reported against Zephyr and not SoF?

lyakh, 2025-07-11:
> How come this is reported against Zephyr and not SoF?

@henrikbrixandersen because similar to #93020 this is toolchain-specific, not application specific. Looks like Zephyr introduced some changes, that made it break with older toolchains, without raising toolchain requirements

lyakh, 2025-07-21:
> [@lyakh](https://github.com/lyakh) does [#93158](https://github.com/zephyrproject-rtos/zephyr/pull/93158) fix this for you? It would also be helpful to know which platform you reproduced this on

@danieldegrasse no, unfortunately it doesn't https://github.com/thesofproject/sof/actions/runs/16411680054/job/46367953484?pr=10127 . It's building for the "nativ_sim" x86-64 platform