zephyrproject-rtos/zephyr #93045: hdlc_rcp_if: hdlc_rcp_if_uart.c contains an infinite loop due to improper handling of a full ring buffer

MichaelFeistETC, 2025-07-11:
### Describe the bug

Using a lpuart on an IMXRT1064 as my ot-uart, I run into an issue where the hdlc_rcp_if_uart driver gets stuck within the callback (`uart_callback`). The `while (uart_irq_update(dev) && uart_irq_is_pending(dev))` constantly evaluates to true due to the RDRF flag being set to 1 indicating data on the UART. However, in order to deassert this flag, the `uart_fifo_read` call within `uart_rx_handle` must be called. This call is prevented due to a lack of space in the ring buffer.

```
static void uart_rx_handle(const struct device *dev)
{
	uint32_t rd_len = 0, len;
	uint8_t *data;

	len = ring_buf_put_claim(
		ot_uart.rx_ringbuf, &data,
		ot_uart.rx_ringbuf->size);
	if (len > 0) {
		rd_len = uart_fifo_read(dev, data, len);

		int err = ring_buf_put_finish(ot_uart.rx_ringbuf, rd_len);
		(void)err;
		__ASSERT_NO_MSG(err == 0);
	}
}
```

If the ring buffer is full, len evaluates to 0 and `uart_fifo_read` is never called. This leads to the infinite loop. Since this is in an IRQ, this completely stalls the processor.

It seems like this case should be handled a bit more gracefully. Either an assert if we expect this to never happen or probably more likely, drop some bytes and notify the user somehow.

### Regression

- [ ] This is a regression.

### Steps to reproduce

I don't have an exact repro, but I believe this should be fairly easy to replicate by some combination of setting CONFIG_OPENTHREAD_HDLC_RCP_IF_UART_RX_RING_BUFFER_SIZE to a low value and/or blocking the uart irq via a higher priority irq long enough for the data in the fifo to exceed the ring buffer size. This may also be dependent on the specific uart implementation (uart_mcux_lpuart.c)? Part of this relies on how `uart_irq_is_pending` is implemented. I'm not sure if other platforms' implementations somehow get around this.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- Present on the latest commit. Currently: 2c097a4c0052ab2496957b9b0aefbb2ef2e5bf1e
- Custom IMXRT1064 based board

lmajewski, 2025-07-14:
Thanks for pointing this out.

What value of `CONFIG_OPENTHREAD_HDLC_RCP_IF_UART_RX_RING_BUFFER_SIZE` have you setup?

MichaelFeistETC, 2025-07-14:
It does seem like my main problem is even getting to that 4096 in the first place though. Seems like under normal operation, it's fairly reasonable to assume you never should hit that boundary. Regardless, it still seems like a bug to hit an infinite loop in that case. 

I added some logging, and it seems like my `ot_uart_rx_cb` sometimes hits a case where it never gets called or at least isn't called for a very long time. Looking into that.