zephyrproject-rtos/zephyr #93055: samples: subsys: usb: dfu-next: unable to get DFU working

henrikbrixandersen, 2025-07-12:
### Describe the bug

I am trying to get USB DFU working with the `samples/subsys/usb/dfu-next/` on the `frdm_k64f` board based on the instructions from https://docs.zephyrproject.org/latest/samples/subsys/usb/dfu-next/README.html#building-with-flash-backend-enabled, but the sample results in UDC error events printed to the Zephyr console while uploading, and the uploaded image is invalid.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. `west build -b frdm_k64f/mk64f12  --sysbuild samples/subsys/usb/dfu-next/ -- -DCONFIG_APP_USB_DFU_USE_FLASH_BACKEND=y -DSB_CONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_MPU_ALLOW_FLASH_WRITE=y -DSB_CONFIG_BOOT_SIGNATURE_TYPE_NONE=y`
2. `west flash -r pyocd`
3. `sudo dfu-util -l`
```
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found Runtime: [2fe3:0005] ver=0402, devnum=75, cfg=1, intf=0, path="2-3.2", alt=0, name="UNKNOWN", serial="FFFFFFFF4E454832200E0019"
```
5. `sudo dfu-util -e`
```
...
Opening DFU capable USB device...
Device ID 2fe3:0005
Device DFU version 0110
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
Device will detach and reattach...
```
7. `sudo dfu-util -l`
```
...
Found DFU: [2fe3:ffff] ver=0402, devnum=76, cfg=1, intf=0, path="2-3.2", alt=1, name="slot1_image", serial="UNKNOWN"
Found DFU: [2fe3:ffff] ver=0402, devnum=76, cfg=1, intf=0, path="2-3.2", alt=0, name="ramdisk0", serial="UNKNOWN"
```
9. `sudo dfu-util -a 1 -D build/dfu-next/zephyr/zephyr.signed.bin`
```
...
dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 2fe3:ffff
Device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Interface #1 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 512
Copying data from PC to DFU device
Download	[=========================] 100%        72600 bytes
Download done.
DFU state(6) = dfuMANIFEST-SYNC, status(0) = No error condition is present
DFU state(2) = dfuIDLE, status(0) = No error condition is present
Done!
```

### Relevant log output

```shell
*** Booting MCUboot v2.1.0-rc1-389-g4eba8087fa60 ***
*** Using Zephyr OS build v4.2.0-rc2-107-ge22ca6b13284 ***
I: Starting bootloader
I: Primary image: magic=good, swap_type=0x2, copy_done=0x1, image_ok=0x1
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Image index: 0, Swap type: none
I: Bootloader chainload address offset: 0x10000
I: Image version: v0.0.0
I: Jumping to the first image slot
*** Booting Zephyr OS build v4.2.0-rc2-107-ge22ca6b13284 ***
[00:00:00.002,000] <inf> main: USB DFU sample is initialized
[00:00:00.005,000] <inf> main: USBD message: Device suspended
[00:00:00.247,000] <inf> main: USBD message: Bus reset
[00:00:00.317,000] <inf> main: USBD message: New device configuration
[00:00:00.317,000] <inf> main: 	Configuration value 1
[00:00:15.519,000] <inf> main: USBD message: DFU detach request
[00:00:15.519,000] <inf> main: Detach USB device
[00:00:15.521,000] <inf> main: USBD message: Bus reset
[00:00:15.524,000] <wrn> udc: Spurious suspend/resume event
[00:00:15.524,000] <inf> main: USBD message: Device suspended
[00:00:15.863,000] <inf> main: USBD message: Bus reset
[00:00:15.920,000] <inf> main: USBD message: New device configuration
[00:00:15.920,000] <inf> main: 	Configuration value 1
[00:00:38.003,000] <err> usbd_core: UDC error event
[00:00:38.004,000] <err> usbd_core: UDC error event
[00:00:38.004,000] <err> usbd_core: UDC error event
[00:00:38.004,000] <err> usbd_core: UDC error event
[00:00:38.004,000] <err> usbd_core: UDC error event
[00:00:38.004,000] <err> usbd_core: UDC error event
[00:00:38.004,000] <err> usbd_core: UDC error event
[00:00:38.004,000] <err> usbd_core: UDC error event
[00:00:38.004,000] <err> udc: Cannot determine the next stage
[00:00:38.004,000] <inf> main: USBD message: Controller error
[00:00:38.004,000] <inf> main: USBD message: Controller error
[00:00:38.004,000] <inf> main: USBD message: Controller error
[00:00:38.004,000] <inf> main: USBD message: Controller error
[00:00:38.004,000] <inf> main: USBD message: Controller error
[00:00:38.004,000] <inf> main: USBD message: Controller error
[00:00:38.004,000] <inf> main: USBD message: Controller error
[00:00:38.004,000] <inf> main: USBD message: Controller error
[00:00:38.004,000] <err> udc: Cannot determine the next stage
[00:00:38.004,000] <wrn> udc: State not changed!
[00:00:38.019,000] <inf> main: USBD message: DFU download completed

*** Booting MCUboot v2.1.0-rc1-389-g4eba8087fa60 ***
*** Using Zephyr OS build v4.2.0-rc2-107-ge22ca6b13284 ***
I: Starting bootloader
I: Primary image: magic=good, swap_type=0x2, copy_done=0x1, image_ok=0x1
I: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Image index: 0, Swap type: test
E: Image in the secondary slot is not valid!
I: Bootloader chainload address offset: 0x10000
I: Image version: v0.0.0
I: Jumping to the first image slot
*** Booting Zephyr OS build v4.2.0-rc2-107-ge22ca6b13284 ***
[00:00:00.002,000] <inf> main: USB DFU sample is initialized
[00:00:00.005,000] <inf> main: USBD message: Device suspended
[00:00:00.288,000] <inf> main: USBD message: Bus reset
[00:00:00.358,000] <inf> main: USBD message: New device configuration
[00:00:00.358,000] <inf> main: 	Configuration value 1
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

- OS: Ubuntu Linux 24.04 LTS, kernel 6.11.0
- Zephyr SDK: v0.17.2
- Zephyr revision: e22ca6b132840efb23b4a00ff7718225a0dd9f29

henrikbrixandersen, 2025-07-12:
I have just tested the same sample, same configuration on the `lpcxpresso55s16` board. There it also logs a couple of UDC error events, but otherwise works:

```
[00:00:26.450,000] <err> usbd_core: UDC error event
[00:00:26.450,000] <err> usbd_core: UDC error event
```

PeeJay, 2025-11-10:
Not sure if it's relevant but I have been having trouble with `<wrn> udc: Spurious resume event` on STM32G4. Enabling `CONFIG_UDC_DRIVER_LOG_LEVEL_ERR=y` remarkably fixes the error.