zephyrproject-rtos/zephyr #93056: west on macOS: stm32cubeprogrammer has incorrect default path and does not respect `PATH`

glward, 2025-07-12:
### Describe the bug

While attempting to set up a development environment for an [STM32 Nucleo U575ZI Q](https://docs.zephyrproject.org/latest/boards/st/nucleo_u575zi_q/doc/index.html) on macOS arm64, I was unable to get the sample to flash:

```
-- west flash: using runner stm32cubeprogrammer
FATAL ERROR: required program /Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/MacOs/bin/STM32_Programmer_CLI not found; install it or add its location to PATH
```

At least for SetupSTM32CubeProgrammer 2.20.0, the path now seems to be:
```
$ ls /Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/Resources/bin/STM32_Programmer_CLI
/Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/Resources/bin/STM32_Programmer_CLI
```
(Note "Contents/Resources" instead of "Contents/MacOs")

Additionally, even with the correct path added to `PATH`, `west flash` still fails:

```
(.venv) $ which STM32_Programmer_CLI
/Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/Resources/bin/STM32_Programmer_CLI
(.venv) $ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner stm32cubeprogrammer
FATAL ERROR: required program /Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/MacOs/bin/STM32_Programmer_CLI not found; install it or add its location to PATH
```

This seems to be due to the `_get_stm32cubeprogrammer_path` method in `scripts/west_commands/runners/stm32cubeprogrammer.py` not including the same fallback to `shutil.which` that Linux and Windows do. Either adding that fallback or altering the "default" path resolve the issue and allow `west flash` to complete successfully. 

Happy to open a PR to add either fix. I think adding the `shutil.which` fallback should be done, as the error message indicates that adding `STM32_Programmer_CLI` to `PATH` should work and it does not on macOS. I'm more hesitant to change the default path as the current path may well be correct on Intel, and presumably was correct for older versions of the STM32CubeProgrammer app. 

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Install Zephyr on a macOS arm64 device
2. Install the latest STM32CubeProgrammer for macOS ARM
3. Attempt to build and flash an STM32 board

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- OS: macOS Sequoia
- Zephyr: 4.2.0-rc3
- Toolchain: Zephyr

daniel-ojeyemi, 2025-07-28:
I'm experiencing the same issue and I just installed zephyr on macos today

jamesy0ung, 2025-08-21:
Yeah same issue for me! I manually added `/Applications/STMicroelectronics/STM32Cube/STM32CubeProgrammer/STM32CubeProgrammer.app/Contents/Resources/bin` which contains `STM32_Programmer_CLI` and I still get the error

closed 2025-08-22 by kartben