zephyrproject-rtos/zephyr #93081: Cache for "west update" in CI

thorsten-klein, 2025-07-14:
### Summary

Hello,

we are currently setting up our first cloud-based CI build for our zephyr project.

In our initial runs, we’ve observed the following timings:

- Cloning our repository: **1 second**
- Setting up the initial pip3 environment (`pip3 install -r requirements.txt`): **15 seconds**
- `west update`: **90 seconds**
- `west packages pip --install`: **15 seconds**
- `west build`: **30 seconds**

As you can see, the `west update` step is by far the most time-consuming and generates significant network traffic. We would like to optimize this by introducing caching.

Do you already have a recommended solution for caching `west update`?

### Describe the solution you'd like

One idea we’re considering is to cache the Git repositories that west clones. For example:

- Keep a centralized mirror of the modules in a local directory (e.g., `~/.west/git`)
- Configure west to clone from these local mirrors instead of remote URLs
  - This could be done using `git worktree`, or
  - By cloning from a local `--mirror` repository using file paths

This would allow the `west update` step to pull from the local system, significantly reducing clone time and external traffic.

### Alternatives

Are there already alternative solutions existing?

nordicjm, 2025-07-14:
> Keep a centralized mirror of the modules in a local directory (e.g., ~/.west/git)

This is just a freestanding application which is already supported https://docs.zephyrproject.org/latest/develop/west/workspaces.html#t3-forest-topology

> west update: 90 seconds

Have you looked at the `west init` help and tried what it suggests?
```
  -o, --clone-opt CLONE_OPT
                        additional option to pass to 'git clone' (e.g. '-o=--depth=1'); may be given more than once; cannot be
                        combined with -l
```

thorsten-klein, 2025-07-14:
Ticket is closed here and re-opened in west project: https://github.com/zephyrproject-rtos/west/issues/828

closed 2025-07-14 by thorsten-klein