zephyrproject-rtos/zephyr #93109: WiFi roaming on the ESP32

cc-zelp, 2025-07-14:
### Is your feature request related to a problem? Please describe.

I am currently trying to do roaming at the application level, manually, by scanning and picking the best BSSID every so often, but this seems to cause issues. Namely, upon the next connection, sometimes, I do not receive an L4 event for IP address received, whereas I don't ever get this issue upon a fresh WiFi connection. 

### Describe the solution you'd like

It would be great if WiFi roaming was handled under-the-hood, where it's well tested, and not on the application layer. 

I know the ESP-IDF provides this via the wifi_config_t struct (btm_enabled). There's a chance it's a straight forward change (not sure if it requires any idf blob changes).

Is it possible to add this config option, along with a KCONFIG choice? @marekmatej (I have been told you are maintaining the ESP port for Zephyr, is this something you could shed some light on, please?)

sylvioalves, 2026-02-20:
@VenetiaFurtado It is not only adding `Kconfig`, there are extra calls to be added in HAL if I am not mistaken.
However, @marekmatej, do you plan to submit anything as you may have already started adding this?

sylvioalves, 2026-10-19:
landed in https://github.com/zephyrproject-rtos/zephyr/pull/118844

closed 2026-10-19 by sylvioalves