zephyrproject-rtos/zephyr #93129: [NXP HAL] MKV56F24 header incorrectly defines FLEXCAN2, causing FlexCAN build failure

tbitcs, 2025-07-14:
### Describe the bug

When building for the **NXP MKV56F24 SoC** (e.g., `nxp_kv56f1m0vlx24`), the MCUX FlexCAN driver (`fsl_flexcan.c`) fails to compile because the **device header (`MKV56F24.h`) incorrectly defines `FLEXCAN2` as `CAN2`**, even though this SoC only has **two FlexCAN instances** (`CAN0`, `CAN1`).

The incorrect define in `MKV56F24.h`:

```c
#define FLEXCAN0 CAN0
#define FLEXCAN1 CAN1
#define FLEXCAN2 CAN2  /* <-- Should NOT exist for MKV56F24 */
```

Error:

```
error: 'CAN2' undeclared (first use in this function); did you mean 'CAN0'?
   s_flexcanIsr(FLEXCAN2, s_flexcanHandle[2]);
```

The SoC feature macros are otherwise correct (`FSL_FEATURE_SOC_FLEXCAN_COUNT = 2`).

### **Expected behavior**

* `FLEXCAN2` should **not be defined** in `MKV56F24.h` for a SoC with only 2 FlexCAN instances.
* CAN2 IRQ handlers in `fsl_flexcan.c` should compile only when `FSL_FEATURE_SOC_FLEXCAN_COUNT > 2`.

---

### **Root Cause**

* The **MCUXpresso device header (`MKV56F24.h`) incorrectly defines FLEXCAN2** as `CAN2`, even though the peripheral does not exist.
* Because `FLEXCAN2` is defined, the FlexCAN driver assumes CAN2 is present and compiles the extra IRQ handler (`CAN_FD2_DriverIRQHandler`), which fails.

Correct behavior:

```c
#define FLEXCAN0 CAN0
#define FLEXCAN1 CAN1
/* FLEXCAN2 should NOT be defined */
```

---

### **Workaround**

Add this in your board’s `CMakeLists.txt`:

```cmake
# Workaround: prevent build failure by assigning a dummy value to CAN2.
# This compiles a dead IRQ handler but avoids modifying the SDK.
zephyr_compile_definitions(CAN2=0)
```

---

### **Proposed Fix**

The correct fix must be made upstream in the NXP MCUX SDK:

* Remove or conditionally guard the `#define FLEXCAN2 CAN2` in `MKV56F24.h`.

Optionally, Zephyr could add guards in `fsl_flexcan.c` as a temporary mitigation:

```c
#if (FSL_FEATURE_SOC_FLEXCAN_COUNT > 2)
void CAN_FD2_DriverIRQHandler(void)
{
    s_flexcanIsr(FLEXCAN2, s_flexcanHandle[2]);
}
#endif
```

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Build any CAN-enabled application for KV56F24:

   ```
   west build -b nxp_kv56f1m0vlx24
   ```
2. Enable CAN:

   ```
   CONFIG_CAN=y
   ```
3. Build fails during compilation of `fsl_flexcan.c`.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

* **Zephyr version**: 4.1.0 (`v4.1.0`)
* **NXP HAL**: `hal_nxp` as shipped with Zephyr (mirrored MCUX SDK)
* **Toolchain**: Zephyr SDK 0.17.0 (arm-zephyr-eabi-gcc 12.2.0)
* **SoC**: MKV56F1M0VLQ24 (KV56F24 series)
* **Board**: Custom KV56F24 board

### Additional Context

This issue originates from the **NXP MCUXpresso SDK**:

* The same incorrect define exists in **upstream NXP repo**:
  [https://github.com/NXPmicro/mcux-sdk](https://github.com/NXPmicro/mcux-sdk) (latest KV56F24 header as of 7-14-2025)

An upstream issue will be filed to correct the MCUX header.
Tracking here for Zephyr users since it blocks KV56F24 CAN builds.

mmahadevan108, 2025-07-22:
@tbitcs , please give the PR a try.

ElectroInnovator, 2025-08-04:
@mmahadevan108 PR is good. Thank you!

mmahadevan108, 2025-08-08:
Fixed by https://github.com/zephyrproject-rtos/hal_nxp/commit/f63c98595f094f6374e4fab8528d3e5cebfb1d90

closed 2025-08-08 by mmahadevan108