zephyrproject-rtos/zephyr #93287: STM32 - the DMA channel number passed to the callback does not match the channel number passed in to dma_start

KurtE, 2025-07-17:
### Describe the bug

A few of us have been experimenting with using DMA on Arduino Giga and Portenta H7 boards 
STM32H77....  If interested more details are up in several places including:
https://github.com/zephyrproject-rtos/zephyr/discussions/93058#discussioncomment-13795375

I am currently experimenting with trying to emulate DMA chaining (or scatter/gather on these boards, which it appears like the DMA1 and DMA2 objects do not support.  So trying to emulate it by starting up the next
chunk in the callback function.  But they kept failing.  

Then I noticed that the channel number that was passed back to me in the callback did not match the 
channel number I passed in to the APIS. 

For example I passed in Channel 2 into the APIS and the channel number that was passed back to me on the
callback was 1. 

My guess is that: the channel numbers passed into the APIs are 1 biased and the internal code converts to
0 biased. 

A quick and dirty extract of a Arduino sketch based off a Zephyr test I was running shows me:
```
DMA Channel allocated 2
dma_start(80665F8,2)
DCB: 80665F8 1
```
Where the DCB line is the pointer to DMA object and channel number.

Not sure of Impact.  My current work around is to stash the channel number into the user data I am
passing to the DMA operation and then use it within the callback.  

If it is simply an off by one and consistent across, different boards, then I could simply increment it...

### Regression

- [ ] This is a regression.

### Steps to reproduce

Currently I have it setup as an Arduino sketch. 

The sketch is using the latest ArduinoCore-zephyr code (github).  I am currently running it
on an Arduino GIGA.

```

#include <zephyr/device.h>
#include <zephyr/kernel.h>
#include <zephyr/sys/ring_buffer.h>
#include <zephyr/logging/log.h>

LOG_MODULE_REGISTER(main, LOG_LEVEL_INF);
#include <zephyr/devicetree.h>
#include <zephyr/multi_heap/shared_multi_heap.h>
#include <zephyr/drivers/dma.h>
#include <zephyr/timing/timing.h>

#define BUFFER_SIZE 128
__aligned(32) uint8_t tx_buffer[BUFFER_SIZE];
__aligned(32) uint8_t rx_buffer[BUFFER_SIZE];
volatile bool dma_completed;

void dma_callback(const struct device *dma_dev, void *user_data, uint32_t channel, int status) {
  Serial.print("DCB: ");
  Serial.print((uint32_t)dma_dev, HEX);
  Serial.print(" ");
  Serial.println(channel);
  if (status == 0) {
    return;
  } else {
    LOG_ERR("DMA chunk failed with status %d on channel %d", status, channel);
    dma_completed = true;
  }
  dma_completed = true;
}

int configure_dma_transfer(const struct device *dma_dev, uint8_t *src, uint8_t *dest, size_t size, int channel) {
  dma_completed = false;

  SCB_CleanDCache_by_Addr((uint32_t *)src, size);
  SCB_InvalidateDCache_by_Addr((uint32_t *)dest, size);

  struct dma_block_config block_cfg = {
    .source_address = (uint32_t)(src),
    .dest_address = (uint32_t)(dest),
    .block_size = size,
  };

  struct dma_config dma_cfg = {
    .dma_slot = 0,
    .channel_direction = MEMORY_TO_MEMORY,
    .complete_callback_en = true,
    .source_data_size = 1,
    .dest_data_size = 1,
    .source_burst_length = 1,
    .dest_burst_length = 1,
    .block_count = 1,
    .head_block = &block_cfg,
    .user_data = nullptr,
    .dma_callback = dma_callback,
  };

  dma_stop(dma_dev, channel);  // Reset before each chunk
  int ret = dma_config(dma_dev, channel, &dma_cfg);
  if (ret) {
    LOG_ERR("DMA config failed: %d", ret);
    return ret;
  }

  printk("DMAStart");
  Serial.print("dma_start(");
  Serial.print((uint32_t)dma_dev, HEX);
  Serial.print(",");
  Serial.print(channel, DEC);
  Serial.println(")");
  ret = dma_start(dma_dev, channel);
  if (ret) {
    LOG_ERR("DMA start failed: %d", ret);
    return ret;
  }

  return 0;
}

#define DMA_NODE DT_NODELABEL(dma1)

void setup(void) {
  while (!Serial && millis() < 5000)
    ;

  Serial.begin(115200);

  const struct device *dma_dev = DEVICE_DT_GET(DMA_NODE);
  //struct dma_status dma_status = {0};
  int ret = 0;
  int tx_dma_channel = 0;
  // Start up the Serial
  //SerialX.begin();
  k_sleep(K_MSEC(250));
  printk("SDRAM test...\n");

  for (uint8_t channel = 1; channel <= 7; channel++) {
    dma_status dmastat;
    memset(&dmastat, 0, sizeof(dmastat));
    ret = dma_get_status(dma_dev, channel, &dmastat);
    printk("ch:%u stat:%d - %u %u %u %u %u %u %u\n", channel, ret,
           dmastat.busy, dmastat.dir, dmastat.pending_length,
           dmastat.free, dmastat.write_position, dmastat.read_position, (uint32_t)dmastat.total_copied);
    if ((tx_dma_channel == 0) && (ret == 0) && (!dmastat.busy)) tx_dma_channel = channel;
  }

  //tx_dma_channel = dma_request_channel(dma_dev, 0);
  printk("DMA Channel allocated %d\n", tx_dma_channel);
  Serial.print("DMA Channel allocated ");
  Serial.println(tx_dma_channel);
  memset(tx_buffer, 0x42, BUFFER_SIZE);
  ret = configure_dma_transfer(dma_dev, tx_buffer, rx_buffer, BUFFER_SIZE, tx_dma_channel);
  if (ret != 0) {
    LOG_ERR("DMA configuration or transfer failed\n");
  }
  int wait_count = 0;
  while (!dma_completed && wait_count++ < 1000) {
    k_sleep(K_USEC(100));
  }
  if (!dma_completed) {
    LOG_ERR("DMA transfer timeout");
  }
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
  delay(500);
}
```
I also have a WIP zephyr test app that I run directly on zephyr currently for portenta H7.
https://github.com/KurtE/zephyr_test_sketches/tree/master/SDRAM_test

This is a WIP, that is currently only setup for the Portenta H7 and probably uses several 
of my other directories... I currently have it setup on windows
d:\zephyrproject 
Under zephyrproject I did the setup for zephyr. 
I then have my testing apps cloned into d:\zephyrproject\zephyr_test_sketches
I cd into there to: (.venv) D:\zephyrproject\zephyr_test_sketches\SDRAM_test>
build: west build -p -b arduino_portenta_h7//m7
flash: west flash

Some kernel monitor output:
```
loop: 49(31) uart:~$

SDRAM test...
DMA Channel allocated 1
Buffer: 0 ADDR:0xc0000080 size:153600
Buffer: 1 ADDR:0xc00258a0 size:153600
Buffer: 2 ADDR:0xc004b0c0 size:153600
loop: 0(0)
dma_callback(0x80525d8 0xc00708c0 0 0)
 DMA:  (65504 0 aa) (65505 0 aa) (65506 0 aa) (65507 0 aa) 88096
loop: 1(1)  DMA:  (65504 1 aa) (65505 1 aa) (65506 1 aa) (65507 1 aa) 88096
```

### Relevant log output

```shell
loop: 49(31) uart:~$

SDRAM test...
DMA Channel allocated 1
Buffer: 0 ADDR:0xc0000080 size:153600
Buffer: 1 ADDR:0xc00258a0 size:153600
Buffer: 2 ADDR:0xc004b0c0 size:153600
loop: 0(0)
dma_callback(0x80525d8 0xc00708c0 0 0)
 DMA:  (65504 0 aa) (65505 0 aa) (65506 0 aa) (65507 0 aa) 88096
loop: 1(1)  DMA:  (65504 1 aa) (65505 1 aa) (65506 1 aa) (65507 1 aa) 88096
```

### Impact

Not sure

### Environment

Two different ones:  Arduino
Windows 11**: ArduinoCore-zephyr** currently off of my own fork which is currently in sync with the changes
as of this morning.   The zephyr it is based on is a snapshot 20250714 on Arduino github.
This morning I rebuild the all of the Arduino "Firmwares" this morning using WSL, then copied the data out
of Ubuntu file system into Windows Filesystem and did the Arduino builds from there...

Zephyr: Currently running on Windows, 
```
kurte@KurtXPS8950:~/git/zephyr$ git branch
* (HEAD detached at 8f6974472e9)
  c33_wifi_sf_rebased

```

erwango, 2025-07-18:
@FRASTM Please have a check on dma api usage in sample above. This off by 1 issue sounds familiar to me.
@KurtE It would be much easier if you convert the reproduction code as a Zephyr application. For investigations, it's much easier to explore directly Zephyr applications than having to set up various related environments (Arduino, Rust, ..)

erwango, 2025-07-21:
@KurtE Did you try to run dedicated dma tests (tests/drivers/dma) ?

KurtE, 2025-07-21:
> [@KurtE](https://github.com/KurtE) Did you try to run dedicated dma tests (tests/drivers/dma) ?

I think I tried building it for the arduino_portenta_h7//m7 or giga  board, but nothing would work.  I don't believe either of
these boards are defined in the boards contained in the boards directories of these tests.  I believe neither of these have their
DMA objects enabled within their overlay/dts files.   Probably should be added.

Sorry, I am still trying to better understand, how some of these sub-systems work, so a lot of fumbling around. 
Things like:  Does a DMA controller support a  block_count > 1 Looks like the STM32H7... Does not.  
Does it support scatter/gather, ...  Maximum block_size.  
How to transfer > 65535 bytes in a DMA operation, etc,

So was doing my own tests, like can I restart a DMA operation within the callback.  So far only tried it memory to memory,
but curious about trying SPI to memory or memory to SPI, to receive an image from Camera or send it to display.
Where image may be 320x480x2 (minimum), would be nice to go up to Giga Display Shield size or ST7796 (48x320x2)

So @mjs513 and I have been hacking up our own, including ones that run directly on zephyr, mentioned above, plus
ones we wrapped up in Arduino, to test through the ArduinoCore-zephyr side.  Like a more complex callback
https://github.com/KurtE/Arduino_GIGA-stuff/blob/main/sketches/zephyr/kurts_dma_test_chained/kurts_dma_test_chained.ino#L39-L103

Where we found this issue.  Currently playing around with this chained sketch to see why: if the buffer size I am
trying to send is: 320x240x2 bytes and block size is (65536-32) Trying to keep all transfers starting on 32 byte boundary and
size  is 0 to 65535.  That this works on the callback using:  
`ret = dma_reload(dma_dev, puser_transfer_data->channel, (uint32_t)puser_transfer_data->tx_buffer, (uint32_t)puser_transfer_data->rx_buffer, chunk_size);
`
So for 153600 bytes we transfer; 65504, 65504, 22592 bytes per the transfer works.
But with: total size of: (128 * 1024) = 65504, 65504, 64, 
The dma_reload version fails on the last 64 byte reload.  

```
loop: 10(a)
  call conf DMAStart memcpy Check BUffers via memcpy
No errors
DCB (0x8084960 0x24033800 1 0)
DMA CB: reload 0xc0010060 0xc00500a0 65504
 start DCB (0x8084960 0x24033800 1 0)
DMA CB: reload 0xc0020040 0xc0060080 64
 start [00:00:08.666,000] <err> dma_stm32: Transfer Error.
[00:00:08.674,000] <inf> dma_stm32_v1: tc: 1, ht: 1, te: 0, dme: 0, fe: 0
DCB (0x8084960 0x24033800 1 -5)
Check BUffers via DMA
No errors
```

But if instead I setup and call dma_config with block and config stucture, both work...

But that is probably a different subject.

closed 2025-10-05 by github-actions[bot]