zephyrproject-rtos/zephyr #93321: Conflicting definitions of LOG_ERR in logging/log.h and posix/syslog.h

besmarsh, 2025-07-18:
### Describe the bug

`LOG_ERR` is a constant defined by POSIX and defined in `include/zephyr/posix/syslog.h`. Zephyr's logging subsystem also defines `LOG_ERR` as a macro to write an error level message to the log.

These definitions are conflicting. This causes a compile-time warning. This also causes a compile-time error if the version of `LOG_ERR` that is included first is used by an application. The order of includes ends up determining whether an application using `LOG_ERR` and including both `logging/log.h` and `syslog.h` (directly or indirectly) will compile.

Applications using the POSIX API that would otherwise be compatible with Zephyr can fail to compile due to this issue.

I noticed in the XSI system logging test this issue being worked around by undefining `LOG_ERR` and manually defining `_LOG_ERR`. This isn't really a generally acceptable solution when compiling POSIX-based code for Zephyr. 

Since `LOG_ERR` is specified by POSIX, the Zephyr logging macro would probably need to be renamed to fix this issue.

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. Add the following to `samples/hello_world/prj.conf`:
```
CONFIG_POSIX_API=y
CONFIG_LOG=y
CONFIG_XSI_SYSTEM_LOGGING=y
```
2. Add the following to the includes section of `samples/hello_world/src/main.c`:
```
#include <syslog.h>
#include <zephyr/logging/log.h>
```
3. Add the following to `samples/hello_world/src/main.c:main()`:
```
syslog(LOG_ERR, "Error");
```
4. `west build -b <board> samples/hello_world`

### Relevant log output

```shell
In file included from <...>/zephyrproject/zephyr/samples/hello_world/src/main.c:10:
<...>/zephyrproject/zephyr/include/zephyr/logging/log.h:48: warning: "LOG_ERR" redefined
   48 | #define LOG_ERR(...)    Z_LOG(LOG_LEVEL_ERR, __VA_ARGS__)
      |
In file included from <...>/zephyrproject/zephyr/samples/hello_world/src/main.c:9:
<...>/zephyrproject/zephyr/include/zephyr/posix/syslog.h:42: note: this is the location of the previous definition
   42 | #define LOG_ERR     3
      |
<...>/zephyrproject/zephyr/samples/hello_world/src/main.c: In function 'main':
<...>/zephyrproject/zephyr/samples/hello_world/src/main.c:15:16: error: 'LOG_ERR' undeclared (first use in this function)
   15 |         syslog(LOG_ERR, "Error");
      |                ^~~~~~~
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- OS: Windows
- Toolchain: Zephyr SDK (arm-zephyr-eabi)
- Commit SHA: 31ef45e2f11543a07773fd0e802efe2409da9638

cfriedt, 2025-07-21:
> As such there can not be, and should not be, a "fix" to POSIX. This could be fixed in Zephyr's log subsystem though, by renaming the macro.

Yeah, it's a small enough change, actually.

> Clearly this would be a breaking change for many applications/subsystems, but it should be an easy change. Is this is an unacceptable solution?

It is, and we have a process for changes like that. I don't have the link handy, but it effectively spans 2 releases (at least). It should follow, approximately,

1. Add new symbol
2. Change all upstream modules and in-tree uses tothe new symbol 
3. Mark the old symbol as deprecated
4. Cut a release
5. Remove the old symbol
6. Cut another release

carlescufi, 2025-10-17:
> [@cfriedt](https://github.com/cfriedt) Do you have any thoughts on my previous comment on this issue?
> 
> * Renaming just `LOG_ERR` or prefixing all the log macros (`Z_LOG_ERR` or `K_LOG_ERR`, maybe)?
> * For either solution, having a temporary Kconfig option to not define the deprecated symbols until they are removed completely?

This would be a massive breaking change. Instead, can't we add some sort of check in the `syslog.h` POSIX header so it detects a redefinition and issues a warning @cfriedt ?

nashif, 2025-10-17:
> * Renaming just `LOG_ERR` or prefixing all the log macros (`Z_LOG_ERR` or `K_LOG_ERR`, maybe)?

please no prefix, or you will have to prefix all LOG_ macros.

just create a new macro, call it LOG_ERROR, deprecate the old one, no?