zephyrproject-rtos/zephyr #93397: west flash fails for all STM32 boards in clean install of Zephyr 4.2.99

quantum-leaps, 2025-07-21:
### Describe the bug

I have cleanly installed Zephyr 4.2.99 on Ubuntu 24.04.2 LTS. The examples build cleanly for STM32 boards. However, the subsequent "west flash" fails for all STM32 boards with the following error message:

west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner stm32cubeprogrammer
FATAL ERROR: required program /home/me/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin/STM32_Programmer_CLI not found; install it or add its location to PATH

This stuff used to work in the past without extra steps or installations.

Thank you for your help.

### Regression

- [x] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

quantum-leaps, 2025-07-22:
I installed STM32_CubeProgrammer on my Ubuntu 24.04.2 LTS. Now, `west flash` fails as follows:

```
$ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner stm32cubeprogrammer
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.20.0                  
      -------------------------------------------------------------------

libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/005, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/005, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
. . .
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/005, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
ST-LINK error (DEV_CONNECT_ERR)
FATAL ERROR: command exited with status 1: /home/me/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin/STM32_Programmer_CLI --connect 'port=swd reset=HWrst' --download /home/me/qp/qpc/example
```

I also tried to use `west flash --runner openocd`. This fails as follows:

```
$ west flash --runner openocd
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/me/qp/qpc/examples/zephyr/dpp/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.12.0-01018-gb89d626c6 (2025-06-23-11:53)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
init
Info : clock speed 1000 kHz
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: open failed
in procedure 'init'

FATAL ERROR: command exited with status 1: /home/me/zephyr-sdk-0.17.2/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/me/zephyrproject/zephyr/boards/st/nucleo_f103rb/support -s /home/me/zephyr-sdk-0.17.2/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/me/zephyrproject/zephyr/boards/st/nucleo_f103rb/support/openocd.cfg '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase /home/me/qp/qpc/examples/zephyr/dpp/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

Since this is STM32 NUCLEO board, I ended up programming the board manually by simply copying the `zephyr.bin` image to the board (which enumerates as a flash drive). Obviously, this is not a general solution.

alexapostolu, 2025-07-22:
`west flash --runner openocd` might have failed due to a permission error.

If you're on linux, try `lsusb` then `sudo chmod 666 /dev/bus/usb/<id>/<id>`, replacing `id` with the three digit numbers like `003/004` for example.