zephyrproject-rtos/zephyr #93617: STM32 ADC DMA Callback off by 1

rjackson159, 2025-07-23:
### Describe the bug

Zephyr Version: 4.1.0
Platform STM32G0B1RE

Utilizing DMA to handle ADC conversion transfers (peripheral to memory)

Sequence of events: ADC conversions complete->DMA is flagged->DMA transfers succesfully (TCI=1)->Transfer complete interrupt->DMA callback (in adc_stm32.c)

What happens:
the DMA channel is passed into the function and compared to the local adc DMA configuration (data->dma.channel) if the channels match ADC conversion completes normally if the channels do not match the function returns taking no action

The channel ID passed into the DMA callback is off by the zephyr offset (+1) when compared to the local adc DMA config (data->dma.channel) resulting in the ADC never finishing a conversion and the calling thread hanging.

Conditional check currently in drivers (adc_stm32.c 790): if (channel == data->dma.channel) this comparison is incorrectly never true resulting in hanging thread

By changing to: if (channel+1 == data->dma.channel)
ADC conversion completes succesfully and thread resumes normal operation

### Regression

- [ ] This is a regression.

### Steps to reproduce

Setup STM32G0 board to utilize DMA and ADC, perform ADC conversion from thread.

### Relevant log output

```shell
Bad log output prior to fixing off by 1:
[00:00:01.130,000] <dbg> adc: adc_entry: ADC Task started
[00:00:01.130,000] <dbg> adc: adc_entry: Alias: adc_5v, Channel ID: 10
[00:00:01.130,000] <dbg> adc: adc_entry: Alias: adc_ntc, Channel ID: 8
[00:00:01.130,000] <dbg> adc: adc_entry: Alias: adc_ambient, Channel ID: 9
[00:00:01.130,000] <dbg> adc: adc_entry: Alias: adc_24v, Channel ID: 11
[00:00:01.130,000] <dbg> adc: adc_entry: Alias: adc_current_fb, Channel ID: 0
[00:00:01.130,000] <dbg> adc_stm32: adc_stm32_channel_setup: Channel setup succeeded!
[00:00:01.130,000] <dbg> adc_stm32: adc_stm32_channel_setup: Channel setup succeeded!
[00:00:01.130,000] <dbg> adc_stm32: adc_stm32_channel_setup: Channel setup succeeded!
[00:00:01.130,000] <dbg> adc_stm32: adc_stm32_channel_setup: Channel setup succeeded!
[00:00:01.130,000] <dbg> adc_stm32: adc_stm32_channel_setup: Channel setup succeeded!
[00:00:01.130,000] <dbg> adc: adc_entry: ADC controller device ready: adc@40012400
[00:00:01.130,000] <dbg> adc: adc_entry: ADC task loop start
[00:00:01.131,000] <dbg> dma_stm32: dma_stm32_configure: Channel (0) memory inc (80).
[00:00:01.131,000] <dbg> dma_stm32: dma_stm32_configure: Channel (0) peripheral inc (0).
[00:00:01.131,000] <dbg> adc_stm32: adc_stm32_dma_start: DMA started
[00:00:01.131,000] <dbg> adc_stm32: adc_stm32_start_conversion: Starting conversion
[00:00:01.131,000] <dbg> adc_stm32: dma_callback: dma callback

Corrected log output after fixing off by 1:
[00:00:22.148,000] <dbg> dma_stm32: dma_stm32_configure: Channel (0) memory inc (80).
[00:00:22.148,000] <dbg> dma_stm32: dma_stm32_configure: Channel (0) peripheral inc (0).
[00:00:22.148,000] <dbg> adc_stm32: adc_stm32_dma_start: DMA started
[00:00:22.148,000] <dbg> adc_stm32: adc_stm32_start_conversion: Starting conversion
[00:00:22.148,000] <dbg> adc: adc_entry: ADC sequencing
[00:00:22.148,000] <dbg> adc_stm32: dma_callback: dma callback
[00:00:22.148,000] <dbg> adc: adc_callback: ADC Callback: ADC sampling index 0
[00:00:22.148,000] <dbg> adc: adc_callback: ADC Callback: ADC sampling index complete
[00:00:22.148,000] <dbg> adc: adc_entry: ADC poll event signaled
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

OS - Linux (Ubuntu)
Toolchain - Zephyr 4.1.0+0, Zephyr SDK 0.16.
      
remote: zephyrproject
     revision: v4.1.0

erwango, 2025-07-24:
Probably missing use of `STM32_DMA_STREAM_OFFSET`.

See dcmipp code for instance:
```
	ret = dma_config(config->dma.dma_dev,
			config->dma.channel + STM32_DMA_STREAM_OFFSET, &dma_cfg);
```

Can you have a try with the following change in ADC driver:
```
	ret = dma_config(data->dma.dma_dev, data->dma.channel + STM32_DMA_STREAM_OFFSET,
			 &dma->dma_cfg);
```

erwango, 2025-07-24:
Can you have a check on https://github.com/zephyrproject-rtos/zephyr/pull/93673 and report back ?

erwango, 2025-08-22:
Some more insights on this:
- `STM32_DMA_STREAM_OFFSET` is actually required today only for stm32 dma client drivers that use HAL as in this case only `dma_config()` is called but remaining part of the DMA handling is done internally to the HAL w/o trouble.
- `STM32_DMA_STREAM_OFFSET` should not be used in LL based dma client drivers which make use of the zephyr dma driver and hence, we should keep id arg consistent across all api calls (start, stop, config, reload).

Which means that:
- the adc driver implementation is fine towards dma api usage and the fix should be done in the DMA driver. 
- and then the issue is not ADC specific but for some reason only seen in your configuration so far.

rjackson159, 2025-08-22:
Alright I've made the change with a small exception this part has 2 dmamux so I had to specify which to use:
```c
	dmas = <&dmamux1 1 5 (STM32_DMA_PERIPH_TO_MEMORY | STM32_DMA_MEM_INC | STM32_DMA_PRIORITY_HIGH | STM32_DMA_MEM_16BITS | STM32_DMA_PERIPH_16BITS)>;
	dma-names = "dmamux1";
```

And with that the callback is performing as expected:

> [00:00:44.536,000] <dbg> dma_stm32: dma_stm32_configure: Channel (1) memory inc (80).
> [00:00:44.536,000] <dbg> dma_stm32: dma_stm32_configure: Channel (1) peripheral inc (0).
> [00:00:44.536,000] <dbg> adc_stm32: adc_stm32_dma_start: DMA started
> [00:00:44.536,000] <dbg> adc_stm32: adc_stm32_start_conversion: Starting conversion
> [00:00:44.536,000] <dbg> adc: adc_entry: ADC sequencing
> [00:00:44.536,000] <dbg> adc_stm32: dma_callback: dma callback
> [00:00:44.536,000] <dbg> adc: adc_callback: ADC Callback: ADC sampling index complete

Nice find, is it not possible to use the DMA directly without using the DMAMUX per this driver? Or am I misunderstanding the relationship between the two?

erwango, 2025-08-22:
> Nice find, is it not possible to use the DMA directly without using the DMAMUX per this driver? Or am I misunderstanding the relationship between the two?

As a mater of fact, it is working if you disable `dmamux1` node. But we're not sure why though. At the very least it isn't safe to do so.