zephyrproject-rtos/zephyr #93620: DesignWare SPI driver crashes on emsdp_em11d when reading SSI_COMP_VERSION register with aux-reg enabled

M-Moawad, 2025-07-23:
### Describe the bug

The DesignWare SPI driver causes instruction errors on the **emsdp_em11d** board when attempting to read the **SSI_COMP_VERSION** register during initialization. This occurs specifically for SPI controllers configured with auxiliary register access (**aux-reg** property).

- SSI_COMP_VERSION register definition: https://github.com/zephyrproject-rtos/zephyr/tree/4867d4564714e42c08b4769f579dc927b8f82270/drivers/spi/spi_dw_regs.h#L40
- Auxiliary register access functions: https://github.com/zephyrproject-rtos/zephyr/tree/4867d4564714e42c08b4769f579dc927b8f82270/drivers/spi/spi_dw.h#L67-L94
- Board device tree configuration:https://github.com/zephyrproject-rtos/zephyr/tree/4867d4564714e42c08b4769f579dc927b8f82270/dts/arc/synopsys/emsdp.dtsi#L131-L158
- Register read helper macro:https://github.com/zephyrproject-rtos/zephyr/tree/4867d4564714e42c08b4769f579dc927b8f82270/drivers/spi/spi_dw_regs.h#L53

### Regression

- [x] This is a regression.

### Steps to reproduce

Use the Zephyr SDK

```bash
west twister -p emsdp/emsdp_em11d -W --scenario sample.basic.helloworld
```

### Relevant log output

```shell
/verification_workspace/jenkins-jobs/zephyr-sanity-check/../../twister-out/emsdp_emsdp_em11d/zephyr/samples/hello_world/sample.basic.helloworld/device.log
--------------------
Open On-Chip Debugger 0.12.0-01018-gb89d626c6 (2025-05-21-22:01)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
25163300316B
DEPRECATED! use 'adapter serial' not 'ftdi_serial'
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: arc-em.cpu tap/device found: 0x200044b1 (mfg: 0x258 (ARC International), part: 0x0004, ver: 0x2)
Info : starting gdb server for arc-em.cpu on 3333
Error: couldn't bind gdb to socket on port 3333: Address already in use
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* arc-em.cpu         arcv2      little arc-em.cpu         running

Info : JTAG tap: arc-em.cpu tap/device found: 0x200044b1 (mfg: 0x258 (ARC International), part: 0x0004, ver: 0x2)
18706 bytes written at address 0x60000000
416 bytes written at address 0x80000000
downloaded 19122 bytes in 0.022556s (827.887 KiB/s)

shutdown command invoked

==== Another type of errors ======

/verification_workspace/jenkins-jobs/zephyr-sanity-check/../../twister-out/emsdp_emsdp_em11d/zephyr/samples/userspace/hello_world_user/sample.helloworld/handler.log
--------------------
***** Exception vector: 0x2, cause code: 0x0, parameter 0x0
Address 0x600046e4
Instruction Error
 r0: 0x20  r1: 0x80010000  r2: 0x80010017  r3: 0x600046e0
 r4: 0x8000117c  r5: 0x8000117c  r6: 0x0  r7: 0xf
 r8: 0x1  r9: 0x0 r10: 0x10 r11: 0x80808080
r12: 0x600046d4 r13: 0x6000b4a8  pc: 0x600046e4
 blink: 0x60004d36 status32: 0x80102806
lp_end: 0x0 lp_start: 0x0 lp_count: 0x0
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

OS: Linux
Toolchain (e.g Zephyr SDK, ...): Zephyr SDK
Commit SHA or Version used: Main 4867d45

### Additional Context

The issue was introduced in commit [db168539](https://github.com/zephyrproject-rtos/zephyr/commit/db168539cf15dcd8a747cb22a0fde46af117e011) from PR https://github.com/zephyrproject-rtos/zephyr/pull/87716, which added SSI component version reading functionality.

The issue is in the code that reads the SSI component version register during SPI initialization: https://github.com/zephyrproject-rtos/zephyr/blob/4867d4564714e42c08b4769f579dc927b8f82270/drivers/spi/spi_dw.c#L569-L573

The emsdp_em11d board has SPI controllers configured with aux-reg property , which uses different register access functions .

For auxiliary register access, the addressing calculation uses `addr + off/4` where the **SSI_COMP_VERSION** register offset is `0x5c` . This results in accessing an invalid memory location on certain DesignWare SPI controller variants that don't implement this register through the auxiliary register interface.

M-Moawad, 2025-07-27:
@cujomalainey Could you check this issue?

ruuddw, 2025-10-24:
Clear. The problem is that the DW SPI controller on the ARC Data Fusion/Sensor&Control has been optimized for low-power/area (among others closely coupling the peripheral with AUX registers, instead of using a bus). It doesn't have this SSI_COMP_VERSION register. If you need it, it should be conditionally included. @M-Moawad can you propose a PR with that?

cujomalainey, 2025-10-24:
Ack, I will gate my features further behind that check.