zephyrproject-rtos/zephyr #93695: NRF54L: New data alignment requirements

JordanYates, 2025-07-25:
### Describe the bug

The nRF54L SoC series has new data alignment requirements (relative to nRF52/53/91/etc) that do not appear to be validated in software.
The peripherals that look to have this new requirement (basic on OPS v0.9, searching for "word aligned")
 * I2S
 * PWM
 * SPIM
 * SPIS
 * TWIM
 * TWIS
 * UARTE

Example datasheet descriptions:
[image]
[image]

Problems have been observed using the SPIS peripheral with `samples/bluetooth/hci_spi.c` due to these alignment requirements, with corrupt data being observed on the Bluetooth controller side.

A brief inspection of the SPIM, UARTE and TWIM drivers don't reveal any validation of this requirement.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

- Zephyr v4.2

JordanYates, 2026-03-31:
@carlescufi @masz-nordic any action on this? I am still unsure whether the datasheet is correct in its listed requirements

carlescufi, 2026-11-12:
@JordanYates approach for the fix currently being discussed internally