zephyrproject-rtos/zephyr #93723: bluetooth: GAP Device Name and Apperance write callbacks

rkel, 2025-07-25:
### Is your feature request related to a problem? Please describe.

Zephyr Bluetooth stack supports GAP Name and Appearance to be writable. Currently this parameters can be changed without any possibility for the application to know about the change. Application may want to update advertising data when this happen or do anything else (like update the displayed name).

### Describe the solution you'd like

Implementation similar to bt_conn_cb with its BT_CONN_CB_DEFINE macro. Lets add BT_CONN_GAP_CB_DEFINE that would hold name_updated_cb and appearance_updated_cb.

Thalley, 2025-07-31:
> is if such device might fail the certification process.

I have seen PTS fail tests if characteristics has other properties than what they expect, so this is quite likely. The "proper" approach would be to raise an erratum to optionally allow notifications or indications for the characteristic, but that will take quite a while to get through, if at all. 

> I was testing a few approaches to allow the update of the device name displayed by the Android system:

Does Android even read the Device Name characteristic, or does it only use the advertised values? 
i.e. if you advertise with "DEVICE_NAME_1" and the GATT characteristic is "DEVICE_NAME_2", what is shown by Android? 

> Nevertheless I still need notify the App about the name change. 

Yub, this make sense regardless :)

rkel, 2025-07-31:
> Does Android even read the Device Name characteristic, or does it only use the advertised values?

It uses the advertised one, and the name is updated only on "Pairing" screen. Once the device is paired, even if new device name is advertised the name is not updated. On our side this would be probably the job for the application.