zephyrproject-rtos/zephyr #93834: Fuel gauge cmake library structure incorrect

matt-wood-ct, 2025-07-29:
### Describe the bug

Fuel gauge drivers are creating a library per driver rather then one library for the fuel_gauge group with all sources included. This is not inline with the rest of the drivers in zephyr, this was a minor detail missed during the reviews on the original introduction.

The issue with this structure is that its is not properly extensible out of tree, normal out of tree drivers utilize zephyr_library_amend(), this cannot be used on fuel_gauges as there is no library called that to amend.

### Regression

- [ ] This is a regression.

### Steps to reproduce

N/A structural issue, reproduction not relevant

### Relevant log output

```shell
N/A
```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

N/A

### Additional Context

Discussed in Discord in advance of raising issue, @bjarki-andreasen confirms this is a mistake likely missed during initial reviews.

matt-wood-ct, 2025-07-29:
Drafted in https://github.com/zephyrproject-rtos/zephyr/pull/93836

teburd, 2025-07-29:
only the build system maintainer would be able to really say one way or another on this one, @tejlmand or maybe @nordicjm would likely be the best folks to discuss with

bjarki-andreasen, 2025-07-30:
The reason to declare one common library for drivers is the common CMake parts, like including private headers and sources needed by the subsystem. GNSS for example creates one library for drivers and common parts
https://github.com/zephyrproject-rtos/zephyr/blob/5d42408efa41e459baf3d352ab3592d362dbaf60/drivers/gnss/CMakeLists.txt#L4-L8
I2C has a common context
https://github.com/zephyrproject-rtos/zephyr/blob/5d42408efa41e459baf3d352ab3592d362dbaf60/drivers/i2c/CMakeLists.txt#L3-L7
If every driver creates its own library, the common parts will likely be duplicated.

Neither approach is "wrong", but common library definitely has its perks

closed 2025-07-31 by matt-wood-ct