zephyrproject-rtos/zephyr #93881: driver: stm32-i2c-v2: negative return from i2c_target_read_processed causes lockup.

iabdalkader, 2025-07-30:
### Describe the bug

According to the [docs](https://docs.zephyrproject.org/apidoc/latest/group__i2c__interface.html#ga4cb0ae2cf41fc2105d1baa5e496e5dae), a negative return from `i2c_target_read_requested_cb_t` or `i2c_target_read_processed_cb_t` should:

> cause the controller to ignore bus operations until a new start condition is received.

However, what actually happens is that the I2C target stops responding, not quite sure what happens, voltage turns off and break points don't get hit anymore.

Here's a normal test with these functions returning 0:
```
I2C> scan
I2C address search:
0x60 (0xC0 W) (0xC1 R)

I2C> [0xC1 r:1]

I2C START
TX: 0xC1 ACK 
RX: 0x42 NACK 
I2C STOP
I2C> [0xC1 r:1]

I2C START
TX: 0xC1 ACK 
RX: 0x42 NACK 
I2C STOP
I2C> [0xC1 r:1]
```

Here's a test with `sample_target_read_processed_cb` returning `-1`:
```
I2C> [0xC1 r:1]

I2C START
TX: 0xC1 ACK 
RX: 0x42 NACK 
I2C STOP
I2C> [0xC1 r:1]

I2C START
TX: 0xC1 NACK 
RX: 0xFF NACK 
I2C STOP
```

FWIW I'm testing with an Arduino Giga R1.

Also, this might be related: https://github.com/zephyrproject-rtos/zephyr/pull/80994

### Regression

- [ ] This is a regression.

### Steps to reproduce

1. In `samples/drivers/i2c/custom_target/src/main.c` change `sample_target_read_processed_cb` to return `-1`.
2. Read from the I2C target.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

erwango, 2025-08-27:
`sample_target_read_processed_cb` is implemented only in `samples/drivers/i2c/custom_target` or a user application so I'm not clear on where the bug would be.
@pillo79 Maybe this would be clearer to you in Arduino context ?

pillo79, 2025-08-28:
@erwango it might indeed be an issue in the ST I2C driver for the "target" mode. I don't have the hardware to reproduce, but looking at the code I _think_ it might be the I2C clock stretching in action. 
I don't see it being explicitly disabled by the driver, and from the H7 RM0433 one of the stretch situations is:

> In transmission, if the previous data transmission is completed and no new data is written in `I2C_TXDR` register, or [...]. This stretch is released when the data is written to the `I2C_TXDR` register.

Before #80994, the code in `stm32_i2c_slave_event` always called `LL_I2C_TransmitData8` (ignoring the callback return value). Now, when -1 is returned by the `read_processed` callback, nothing else is done, therefore the bus locks "forever". 
The RM talks about using "slave byte control mode" to allow code to choose whether to ACK or NACK the current transfer; AFAICS we don't do that yet?

pillo79, 2025-09-02:
The issue is with a Zephyr I2C target sample that uses no Arduino APIs, and the bug can be verified on all STM32H747 devices (or any other that share the I2C driver and HAL).

erwango, 2025-10-08:
^^@juickar

iabdalkader, 2025-11-21:
> so I'm okay to remove the log and maybe we should change the return value to be transmitted to 0xFF in case of an error to be more explicit

That sounds good. FWIW, we could also revert it and just fix the docs to say that if there's no more data (i.e., cb returns `-1`), the target stretches the clock. I haven't actually tested that it can recover from this state though, but it should.