zephyrproject-rtos/zephyr #93898: Enable/Enhance PSCI Support in Zephyr

reach2shaunak, 2025-07-30:
### Problem Description

Background:
-------------------------
PSCI specifications are defined here: https://developer.arm.com/documentation/den0022/latest/

Power management aware operating systems dynamically change the power states of cores, balancing the available compute capacity to match the current workload, while striving to use the minimum amount of power. Some of these techniques dynamically switch cores on and off or place them in quiescent states, where they no longer perform computation. This means they consume very little power. The main examples of these techniques are:

- Idle Management: When the kernel in an OS has no threads to schedule on a core, it places that core into a clock-gated, retention, or even fully power-gated state. However, the core remains available to the OS.

- Hotplug: Cores are physically switched off when compute demand is low, and then brought back online when it increases. The OS migrates all interrupts and threads away from the cores that are taken offline and re balances the load when they are brought back online.

This is a request for comments (RFC) on enabling limited client-mode PSCI support from ARM Cortex-M cores in Zephyr, where appropriate. Some SoCs with heterogeneous CPU clusters (e.g., Cortex-A + Cortex-M) expose PSCI services to M-core firmware or runtime code, enabling system control such as SYSTEM_RESET or SYSTEM_OFF from the M-core.

Zephyr currently assumes PSCI is relevant only to Cortex-A targets and includes a PSCI (Power State Coordination Interface) implementation under drivers/pm_cpu_ops/pm_cpu_ops_psci.c, targeted at ARM Cortex-A cores. This implementation provides:

pm_cpu_on()
pm_cpu_off()
pm_system_reset()
Basic SMC/HVC transport support

However, this code is only compiled for PM_CPU_OPS_PSCI, and it calls "arm_smccc_smc" which depends on CONFIG_HAS_ARM_SMCCC which is defined under arch/arm64 (A-profile targets, ARM Cortex-A). Not proper support for 32 bit architecture.

PSCI is excluded entirely for M-profile cores (e.g., Cortex-M4, M7, M33, M85), even in heterogeneous SoC environments where Cortex-M may need to access PSCI services.
This RFC proposes an optional mechanism for enabling PSCI access (via SMC or mailbox interface) even on M-profile platforms where firmware or hardware permits.

Motivation
----------------------------
Some SoCs (e.g., STM32MP1, i.MX8M) include both Cortex-A and Cortex-M cores. In these configurations, the Cortex-M core (often running Zephyr) may be expected to:

1. Request a system reboot or shutdown via PSCI (e.g., SYSTEM_RESET, SYSTEM_OFF)
2. Notify the secure firmware for power transitions
3. Invoke custom PSCI services or vendor extensions

Currently, Zephyr cannot use the existing PSCI support from M-cores, due to build and architecture constraints.

### Proposed Change (Summary)

This RFC proposes to extend Zephyr's PSCI support to Cortex-M platforms, enabling Zephyr (running on an M-core) to act as a PSCI client in systems where:

PSCI-like services are available (via secure firmware such as TF-M or TF-A),

Zephyr needs to request system control actions (e.g., reset, shutdown),

The system is either:

Dual-core (Cortex-A + Cortex-M) with secure firmware managing power/reset via PSCI, or

M-core-only, but with firmware that implements a lightweight PSCI interface for consistency.

### Proposed Change (Detailed)

Extend the existing PSCI implementation in Zephyr to optionally support Cortex-M profiles for a limited subset of PSCI calls.
	                 +-------------------------------------------------------------+
                         |        Secure Firmware (TF-A or TF-M)           |
                         |-------------------------------------------------------------|
                         | - Runs on Secure world (A-core or M-core)|
                         | - Implements PSCI or PSCI-like services     |
                         |   (e.g., SYSTEM_RESET, SYSTEM_OFF)           |
                         | - Interfaces: SMC, IPC, Mailbox, etc.            |
                         +------------------^------------------------------------------+
                                                                       |
            Secure communication (SMC or IPC/Mailbox calls)
                                                                         |
+---------------------------------------------+             |           +---------------------------------------+
|     Cortex-A Core(s)                      |             |           |      Cortex-M Core                |
|  (Linux, U-Boot, etc.)                   |             |           |     (Zephyr RTOS)                  |
+---------------------------------------------+             |           +---------------------------------------+
| - OS uses PSCI for:                      |             |           | - Zephyr optionally uses     |
|   - CPU suspend/resume            |             |           |   light PSCI client to call:     |
[... 119 lines cut ...]
3. **Zephyr alreday have the DTS Binding Recognition for PSCI through smc,hvc**
Add support for M-core builds, allow /psci nodes in the devicetree using mailbox or vendor-ipc:

psci {
    compatible = "arm,psci-1.1/0.2";
    method = "smc" / "mailbox"/ "vendor-ipc";
};

4. **Refactor Existing PSCI Code**
Split existing A-profile PSCI code to allow reuse of:
PSCI call IDs
Return codes
Shared SMC/HVC call helpers (if applicable)

Into a new location, e.g., arch/arm/core/common/psci_common.c.

This avoids code duplication and keeps Cortex-A logic isolated from Cortex-M logic where needed.

### Concerns and Unresolved Questions

Important limitation: 

- Cortex-M cores do not support the SMC instruction.
- If PSCI is implemented only via SMC in the secure world (like in TF-A for Cortex-A), the M-core cannot directly use it.
- Requires an alternate interface, e.g., mailbox or TF-M service.

reach2shaunak, 2025-08-01:
yes.

reach2shaunak, 2025-08-05:
@bjarki-andreasen, @henrikbrixandersen  Do you think we should discuss this for architectural review. Please let me know.

henrikbrixandersen, 2025-08-05:
> @bjarki-andreasen, @henrikbrixandersen  Do you think we should discuss this for architectural review. Please let me know. 

Yes, I think presenting this for the Architecture WG is a great idea.

reach2shaunak, 2025-08-05:
Thanks @henrikbrixandersen

carlescufi, 2025-08-11:
@reach2shaunak I've scheduled this for tomorrow's Arch WG meeting:
https://lists.zephyrproject.org/g/devel/message/9889

reach2shaunak, 2025-08-11:
Thanks @carlescufi . I will attend and present.

closed 2026-09-09 by github-actions[bot]