zephyrproject-rtos/zephyr #93929: `CONFIG_DCACHE=y` breaks `spi_nxp_lpspi_dma.c`on RT1060_evk

JaagupAverin, 2025-07-31:
### Describe the bug

Using NXP RT1060_evk, attempting to use lpspi1 for SPI communication.

### Disabling EDMA only:
```
&lpspi1 {
        /delete-property/ dmas;
        /delete-property/ dma-names;
	status = "okay";
};
```
Then SPI works properly.
### Disabling cache only:
```
CONFIG_DCACHE=n
```
Then SPI works properly.
### Using both (evaluation kit default):
```
&lpspi1 {
        /* EDMA is enabled in evaluation kit .dtsi files by default */
	status = "okay";
};
```
```
CONFIG_DCACHE=y
```
Then no errors are logged but transferred SPI data is garbage.

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

Tested on both 4.1, 4.2 and main.

JaagupAverin, 2025-08-01:
Hello, thanks for this information.
For more context, SPI is actually used by the nRF7002 shield stack and not by user code (https://docs.zephyrproject.org/latest/boards/shields/nrf7002ek/doc/index.html)
Does this imply that the NRF stack does not use nocache attributes in relevant places?

JaagupAverin, 2025-08-03:
For reference: /home/jaagup/krakul/harju-elekter/zephyr/modules/nrf_wifi/bus/spi_if.c I think this is the relevant part.

JaagupAverin, 2025-08-03:
For reference, I think this is the relevant code: `zephyr/modules/nrf_wifi/bus/spi_if.c`

decsny, 2025-08-04:
duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/93929

closed 2025-08-04 by decsny